RROrg/rr #940: BUG - Plex use cpu when hardware transcode

vuminhtuanhvtc, 2024-05-07:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N100
 * NIC:  i226

RR version: (You can find it in the update menu.)  
 * RR:  24.4.8
 * addons:  reboottoloader, cpuinfo, localrss, bootwait, hdddb, mountloader, acpid, sensors, notify, revert, eudev, rndis, misc, disks, wol
 * modules:  as default
 * lkms:  prod

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1-69057 Update 5

Issue:  I am experiencing an issue with hardware transcoding on my server running Plex. Despite enabling hardware transcoding and having an iGPU available, the transcoding process seems to utilize the CPU instead, leading to high CPU usage (around 80-90%) and causing buffering during playback.

Configuration:
*Plex installed in Package Center, version 1.40.2.8395-7000
Hardware transcoding enabled in Plex settings. 
iGPU available for transcoding

This is my setting in Plex transcoder

[image]

I've ask some of my friends who use N100/N5105 with SA6400 and they have same issue with me

I suspect that this issue may be related to the driver for the SA6400 on the N100 CPU and the hardware transcoding process is not utilizing the available iGPU as expected.

I'd really appreciate it if the dev team could take a look into this. Let me know if you need any more info or if there's anything I can do to help troubleshoot.

Thank you so much

wjz304, 2024-05-07:
@MoetaYuko

MoetaYuko, 2024-05-07:
We don't have plex pass so cannot test, use emby/jellyfin.

closed 2024-05-07 by MoetaYuko