RROrg/rr #947: Multiple bootloader disks found, please insert only one bootloader disk

WUDDC, 2024-05-08:
[image]

[image]
RR：24.5.1
新u盘安装

WUDDC, 2024-05-08:
> [image]
>
> 请填写以下信息. Please fill in the following information.
>
> Install ENV: (You can find it in the boot interface.)
>
> * DMI:
> * CPU:
> * NIC:
>
> RR version: (You can find it in the update menu.)
>
> * RR:
> * [...]

没办法通过#173的方法获取到日志，同时/mnt目录也是空的，因为就是直接写盘后插上就显示这样

wjz304, 2024-05-09:
好吧，是我不严谨了，你竟然 nas名字 包含 RR2

closed 2024-06-23 by wjz304