RROrg/rr #949: SA6400 CPU N100 VMM无法启动虚拟机

milesking, 2024-05-10:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N100
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  latest
 * addons:  n/a
 * modules: n/a  
 * lkms:  n/a

DSM: 
 * model:  SA6400
 * version:  7.2

Issue:  
虚拟机无法启动，不知道有没有碰到类似情况的大佬，能给点解决思路

milesking, 2024-05-17:
> 你是不是用esxi安装的黑群晖？

没有，倍控N100直接安装的

andy094, 2024-05-26:
> > 你是不是用esxi安装的黑群晖？
> 
> 没有，倍控N100直接安装的

可以检查一下BIOS设置该开的设置是否开启，这个问题没法回答，因为有的人大部分人都可以，少部分人不行。基本都是BIOS或者主板问题

milesking, 2024-05-27:
> > > 你是不是用esxi安装的黑群晖？
> > 
> > 
> > 没有，倍控N100直接安装的
> 
> 可以检查一下BIOS设置该开的设置是否开启，这个问题没法回答，因为有的人大部分人都可以，少部分人不行。基本都是BIOS或者主板问题

虚拟化是打开的，vt-d也是打开的虽然应该和这个问题没啥关系。我看[chiphell](https://www.chiphell.com/forum.php?mod=viewthread&tid=2538343&extra=&ordertype=1&page=1)上也有相关的讨论，有些推测是内核原因。

wjz304, 2024-06-23:
无法复现，无法调查。

Lynnhs1n, 2024-07-13:
Install ENV: (You can find it in the boot interface.)

DMI:DELL R730XD
CPU: 2630V4 *1
RAM:32GB
NIC:Intel(R) 2P X540/2P I350 rNDC
Emvironment: PVE
SCSI:H330 MINI 
RR version: RR: rr-24.7.0
addons: n/a
modules: n/a
lkms: n/a
DSM:
model: SA6400
version: 7.2

我是在PVE虚拟机下安装的黑裙，做了SCSI卡直通，也是无法打开VMM虚拟机，提示的信息与你一样。

DannyChen0623, 2024-08-10:
[image]
[image]
原來不是只有我一個人寂寞，同樣的NAS型號、處理器，只是我可以正常安裝但是安裝後系統很不穩定不定時藍屏
使用的硬體是：T-bao天钡WTR R1英特尔N100小主机 3台都有同樣的問題
[image]

alwin0516, 2024-08-19:
> ![2024-08-08_171031](https://private-user-images.githubusercontent.com/130951276/356782282-0c9360dd-65fb-458e-a32e-a4760ac20557.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQwMzc1NzUsIm5iZiI6MTcyNDAzNzI3NSwicGF0aCI6Ii8xMzA5NTEyNzYvMzU2NzgyMjgyLTBjOTM2MGRkLTY1ZmItNDU4ZS1hMzJlLWE0NzYwYWMyMDU1Ny5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxOVQwMzE0MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lOWNhYzdjMzAwM2RhODQ0YjRlMzcxZmQ0M2YxZjJlMDdiYTFhNGM4Y2NmYTFkMDEwNjkyYmQ1YzM5YWMyM2U5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iYiwPEQFXt7Ga40nkvdI2M4eLzYW0V9WJFsUfHjufK8) ![2024-08-08_171647](https://private-user-images.githubusercontent.com/130951276/356782283-a63de27d-3fea-4b5b-9cb4-5058ddc22a70.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQwMzc1NzUsIm5iZiI6MTcyNDAzNzI3NSwicGF0aCI6Ii8xMzA5NTEyNzYvMzU2NzgyMjgzLWE2M2RlMjdkLTNmZWEtNGI1Yi05Y2I0LTUwNThkZGMyMmE3MC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxOVQwMzE0MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMzZmYzY5NDFmZDA3NzI1ZDdmOTA1NGUyN2Y1YTgzYjgxYjI1M2U5MzkzZWUwODFlMTg5Yzk1ZjVlMDAyNDJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.atSSYSeBXaP5TneV10xAkbRpjsdvrhTno5JSuz-o_RA) 原來不是只有我一個人寂寞，同樣的NAS型號、顯示卡，只是我可以正常安裝但是安裝後系統很不穩定不定時藍屏 使用的硬體是：T-bao天VATR R1 IntelN100小主機3台都有同樣的問題 ![2024-08-10_095744](https://private-user-images.githubusercontent.com/130951276/356782452-6c470234-a2ab-49af-86f2-16f07e704cd6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQwMzc1NzUsIm5iZiI6MTcyNDAzNzI3NSwicGF0aCI6Ii8xMzA5NTEyNzYvMzU2NzgyNDUyLTZjNDcwMjM0LWEyYWItNDlhZi04NmYyLTE2ZjA3ZTcwNGNkNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxOVQwMzE0MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jODAwY2ZiN2M3OGViZGE3ZTQ1YzhhMGRjZTA4ZmM3YWQ1MzY5MDMzMWU5YTBlNjJhYTU2NmRhOGZhMGQzOWY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DbEPQgL9oLrjVroBNnygbJdA8hH5U4yANBk7QbfkYR0)

我也是天鋇N100，一樣也有VMM藍屏的問題

DannyChen0623, 2024-08-19:
T-bao 天钡 WTR PRO 也是一樣的情況，但這台機器目前Windows的系統只有XP系統穩定運作。
T-bao 天钡 WTR R1 運行XP也是會當機
以上兩台機器建議安裝Linuxmint系統， 官方網址：https://linuxmint.com/ 

另外最近安裝的迈云达M1 mini N100，運行Windows虛擬機都很正常。

alwin0516, 2024-08-19:
> T-bao 天钡 WTR PRO 也是一樣的情況，但這台機器目前Windows的系統只有XP系統穩定運作。 T-bao 天钡 WTR R1 運行XP也是會當機 以上兩台機器建議安裝Linuxmint系統， 官方網址：https://linuxmint.com/
> 
> 另外最近安裝的迈云达M1 mini N100，運行Windows虛擬機都很正常。

我還有一台天鋇R1 5700U裝不了DSM，機型選DS923 kernel會直接崩潰重啟，選sa6400裝DSM重啟後就抓不到IP

DannyChen0623, 2024-08-19:
天鋇先前的BIOS似乎裝不了黑群暉，後來有更改BIOS後才能正常安裝。
可以查看看是否有新版BIOS

wjz304, 2024-08-19:
VMM win11 蓝屏 的问题 尝试 将引导改为 直接启动，

DannyChen0623, 2024-08-19:
> VMM win11 蓝屏 的问题 尝试 将引导改为 直接启动，

沒有看過這個選項，可以指導一下在哪邊設定嗎，感恩。

Gitawake, 2024-08-20:
一样的天呗r1 n100。黑群sa6400 vmm装windows不定时蓝屏重启。不知什么原因

DannyChen0623, 2024-08-20:
> 一样的天呗r1 n100。黑群sa6400 vmm装windows不定时蓝屏重启。不知什么原因
https://github.com/RROrg/rr/issues/949#issuecomment-2295796072 
此時裝Linux會是一個好選擇，除非原廠有修改BIOS。
或者在BIOS裡面某些參數要調整，機器都沒在身邊無法測試。

DannyChen0623, 2024-08-29:
> 常安裝但是安裝後系統很不穩定不定時藍屏
> 使用的硬件是：T-bao天VATR R1 IntelN100小主機3台都有同樣的問題

我想回報一下：原本可以正常運行後來不行? 這會是啥原因?
原本的用廠商送的無印刷白牌8G TF卡引導，但覺得開機太慢了重新製作一張Sandisk金牌32G TF卡，
由原本rr-24.8.4換為rr-24.8.5後VM就藍屏了.......

DannyChen0623, 2024-08-29:
再分享N100執行VM不藍屏的Windows系統：Windows7 精簡版
下載網址：https://github.com/WhatTheBlock/WindowsSimplify/blob/master/7/README.md

loveoranges, 2024-09-06:
不要用群晖的virtio驱动和guest-agent  我换官方virtio驱动后没有遇到过蓝屏问题 https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/stable-virtio/

yanchiyun, 2024-09-17:
> > VMM win11 蓝屏 的问题 尝试 将引导改为 直接启动，
> 
> 沒有看過這個選項，可以指導一下在哪邊設定嗎，感恩。

你找到办法了吗？

yanchiyun, 2024-09-17:
> VMM win11 蓝屏 的问题 尝试 将引导改为 直接启动，

大佬如何改成直接启动？

orange-ho, 2024-10-11:
> > 常安裝但是安裝後系統很不穩定不定時藍屏
> > 使用的硬件是：T-bao天VATR R1 IntelN100小主機3台都有同樣的問題
> 
> 我想回報一下：原本可以正常運行後來不行?這會是啥原因? 原本的用廠商送的無印刷白牌8G TF卡引導，但覺得開機太慢了重新製作一張Sandisk金牌32G TF卡， 由原本rr-24.8.4換為rr-24.8.5後VM就藍屏了.......

问题解决了？

loveoranges, 2024-10-12:
> > > 常安裝但是安裝後系統很不穩定不定時藍屏
> 
> > > 使用的硬件是：T-bao天VATR R1 IntelN100小主機3台都有同樣的問題
> 
> > 
> 
> > 我想回報一下：原本可以正常運行後來不行?這會是啥原因? 原本的用廠商送的無印刷白牌8G TF卡引導，但覺得開機太慢了重新製作一張Sandisk金牌32G TF卡， 由原本rr-24.8.4換為rr-24.8.5後VM就藍屏了.......
> 
> 
> 
> 问题解决了？

引导改为 直接启动 ，关闭cpu选项的hyper-v指令集就没发生过蓝屏

kimyxl-yang, 2024-10-17:
> > > > 常安裝但是安裝後系統很不穩定不定時藍屏
> > 
> > 
> > > > 使用的硬件是：T-bao天VATR R1 IntelN100小主機3台都有同樣的問題
> > 
> > 
> > > 
> > 
> > 
> > > 我想回報一下：原本可以正常運行後來不行?這會是啥原因? 原本的用廠商送的無印刷白牌8G TF卡引導，但覺得開機太慢了重新製作一張Sandisk金牌32G TF卡， 由原本rr-24.8.4換為rr-24.8.5後VM就藍屏了.......
> > 
> > 
> > 问题解决了？
> 
> 引导改为 直接启动 ，关闭cpu选项的hyper-v指令集就没发生过蓝屏

大佬，怎么改直接启动啊？

wjz304, 2024-10-29:
https://github.com/RROrg/rr/releases/tag/24.10.3

closed 2024-10-29 by wjz304