RROrg/rr #953: BUG nvmecache插件反而导致ssd不显示

lnc1995, 2024-05-10:
RR 24.5.1 插件也是最新。
全新编译918+，全新安装，安装完成首次启动能看到SSD。
重启后在存储管理器中看不到。
这时候把libsynonvme.so.1.bak还原回来，一切就正常了。
也就是nvmecache这个脚本在逻辑上好像有点问题，他成功改了so.1文件，但是重启一次以后把这个改好的文件变成bak了

lnc1995, 2024-05-10:
更进一步测试，首次启动成功后，重启到RR，把这个插件删除了再启动，SSD仍旧会不显示。这个应该可以更详细一点的帮助定位脚本在哪一步运行的时候出了问题

wjz304, 2024-05-10:
ls -ld /sys/block/nvme*

lnc1995, 2024-05-12:
> ls -ld /sys/block/nvme*

lrwxrwxrwx 1 root root 0 May 11 19:37 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:01.0/0000:01:00.0/nvme/nvme0/nvme0n1

wjz304, 2024-05-14:
原因不详。

lnc1995, 2024-05-14:
> 原因不详。

好吧，我还以为另一个哥们的问题跟我类似，已经修复了呢。
或者说我这边还可以做什么帮助排查这个问题

wjz304, 2024-05-14:
怀疑你在添加插件前做过修改

wjz304, 2024-05-18:
> > ls -ld /sys/block/nvme*
> 
> lrwxrwxrwx 1 root root 0 May 11 19:37 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:01.0/0000:01:00.0/nvme/nvme0/nvme0n1

这是 RR 下的结果还是DSM下的结果

lnc1995, 2024-05-18:
> > > ls -ld /sys/block/nvme*
> > 
> > 
> > lrwxrwxrwx 1 root root 0 May 11 19:37 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:01.0/0000:01:00.0/nvme/nvme0/nvme0n1
> 
> 这是 RR 下的结果还是DSM下的结果

上面的是DSM的
这个是RR的
lrwxrwxrwx 1 root root 0 May 18 16:04 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:01.0/0000:01:00.0/nvme/nvme0/nvme0n1

wjz304, 2024-05-18:
/usr/lib/libsynonvme.so.1  /usr/lib/libsynonvme.so.1.bak 取出来看看吧

lnc1995, 2024-05-18:
> /usr/lib/libsynonvme.so.1  /usr/lib/libsynonvme.so.1.bak 取出来看看吧

已通过邮箱发送，麻烦了

wjz304, 2024-05-19:
了解了，

wjz304, 2024-05-19:
根本问题：
因为你的 ssd 的 0000:00:01.0 本身为 官方第二条SSD默认的路径之一，这导致 再第二次执行插件时被替换为为第二条ssd 的路径。
但是只有一条所以再第二次重启后路径错乱。
解决方案：
插件每次修改均使用 bak 的原始文件。

附加问题：
你的 bak文件为修改后的问价，并非原始文件，需要先还原备份文件。

wjz304, 2024-05-19:
https://github.com/RROrg/rr/releases/tag/24.5.5

此外
在DSM 下：
```
# 还原为官方的 libsynonvme.so.1 （取自 DS918+ 69057）
curl -kL https://github.com/RROrg/public/releases/download/public/libsynonvme.so.1 -o /root/libsynonvme.so.1
mv -f /root/libsynonvme.so.1 /usr/lib/libsynonvme.so.1
```

lnc1995, 2024-05-19:
> https://github.com/RROrg/rr/releases/tag/24.5.5
> 
> 此外
> 在DSM 下：
> ```
> # 还原为官方的 libsynonvme.so.1 （取自 DS918+ 69057）
> curl -kL https://github.com/RROrg/public/releases/download/public/libsynonvme.so.1 -o /root/libsynonvme.so.1
> mv -f /root/libsynonvme.so.1 /usr/lib/libsynonvme.so.1.bak
> ```
> 

👌 感谢

closed 2024-05-26 by wjz304