RROrg/rr #963: after updated to 24.5.3 can't connect.

eragon4k, 2024-05-12:
It shows and detected e1000e (intel I219-LM) but cant connect.

请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  24.5.3
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  ds3622
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-05-13:
check other model.

eragon4k, 2024-05-13:
used RR Manager 2.0.19 from 24.5.1 to 24.5.3.

24.5.1 was working and stable.

I'll give it a try "use of official drivers" method.

eragon4k, 2024-05-14:
Looks like it's not the driver issue.
I believe there is conflict when you setup static ip with in the DSM.

When the loader give and detects lan and it give initial IP address to connect but when you have static ip already set up inside DSM. It not responding from static ip and the loaders initial ip address.

And when it boots, I ping the loader's initial IP address, and it responds, and then it stops.

Now I'm going back to 24.5.1, removing the static IP from the DSM, and trying to update.

wjz304, 2024-05-14:
The e1000e simulated by pve will continue to go up/down using the driver I compiled. Just use the official driver.
Of course, there are many models of e1000e, and it may not be suitable for all.
Just that the e1000e driver has not been updated recently.

eragon4k, 2024-05-14:
i did tried "Priority use of official drivers: true" with 24.5.4 but the result was the same thing.

eragon4k, 2024-05-14:
So I did a few tests.

1. I freshly built the loader and the "Priority use of official drivers: true" method with 24.5.4.
Out comes with no success.

2. With version 24.5.1 removed DSM Static ip and updated to 24.5.4 with "Priority use of official drivers: true" and out come was no success.

For now going back to 24.5.1.

wjz304, 2024-05-15:
取一下  pdivid

wjz304, 2024-05-15:
intel I219-LM

wjz304, 2024-05-15:
https://github.com/RROrg/rr/issues/173#issuecomment-1715763449

eragon4k, 2024-05-15:
Can't find it.

The log is scrolling far too quickly.

There may be more logs on the top part of the screen.

wjz304, 2024-05-15:
lspci -nn | grep 0200

eragon4k, 2024-05-15:
I219-LM [8086:15b7]

snailium, 2024-05-16:
My loader lost model/architecture information after upgrade to 24.5.4. I had to rebuild the loader from scratch.

After that, DSM seems corrupted. It gave me the first setup wizard. I had to setup everything as new, and recover the configuration from Synology account.

There are too much hassle upgrade cross 24.5.2 version, which seems heavily restructuring in the loader.

wjz304, 2024-05-16:
> 升级到 24.5.4 后，我的加载器丢失了模型/架构信息。我不得不从头开始重建装载机。
> 
> 在那之后，DSM似乎已损坏。它给了我第一个设置向导。我必须将所有内容都设置为新的，并从 Synology 帐户中恢复配置。
> 
> 跨 24.5.2 版本升级有太多麻烦，这似乎在加载器中进行了大量重组。

> My loader lost model/architecture information after upgrade to 24.5.4. I had to rebuild the loader from scratch.
>
> After that, DSM seems corrupted. It gave me the first setup wizard. I had to setup [...]

Yes, the logic adjustment in 5.2 is significant, but it does not involve driver adjustment,
At that time, I was hesitant about whether to revise it to restrict updates from being allowed,

wjz304, 2024-05-18:
> I219-LM [8086:15b7]
> 
> Thank you

还好，官方的驱动也支持这个 pid&vid

eragon4k, 2024-05-18:
> > I219-LM [8086:15b7]
> > Thank you
> 
> Fortunately, the official driver also supports this pid&vid

So, my understanding is the driver was removed after 24.5.1. Right?

wjz304, 2024-05-18:
> > > I219-LM [8086:15b7]
> > > Thank you
> > 
> > 
> > Fortunately, the official driver also supports this pid&vid
> 
> So, my understanding is the driver was removed after 24.5.1. Right?

no, I don’t know why at the moment. I’ll try updating the driver version.

eragon4k, 2024-05-18:
> > > > I219-LM [8086:15b7]
> > > > Thank you
> > > 
> > > 
> > > Fortunately, the official driver also supports this pid&vid
> > 
> > 
> > So, my understanding is the driver was removed after 24.5.1. Right?
> 
> no, I don’t know why at the moment. I’ll try updating the driver version.

I did try with Priority use of official drivers: true as you mentioned  before

lyp49472060, 2024-05-18:
24.4.6升级到24.5.4一样找不到

wjz304, 2024-05-18:
try add `e1000e.KumeranLockLoss=1` to cmdline
[image]

eragon4k, 2024-05-18:
> try add `e1000e.KumeranLockLoss=1` to cmdline [...]

Is this reply for me or @lyp49472060?

wjz304, 2024-05-18:
you

> Is this reply for me or @lyp49472060?

eragon4k, 2024-05-18:
> you

Understood.  I'll try this later today and update you.

Thank you

eragon4k, 2024-05-18:
> you

so i tried with "e1000e.KumeranLockLoss=1". the result was the same and I just removed the internet cable and re-plugged it and it was getting a ping again.

therefore i removed "e1000e.KumeranLockLoss=1" and test it out to see if I could get a ping just by removing the ethernet cable and re-plugging it. Surprisingly I was getting a ping.

But after the recovery process it’s not responding.

wjz304, 2024-05-19:
How many NIC are there in total?

eragon4k, 2024-05-19:
> How many NIC are there in total?

Just one.
I am using a white label SN but it got banned  after upgrade to 5.4.

wjz304, 2024-05-19:
e1000e has another parameter `e1000e.SmartPowerDownEnable=1` , You can also try it

wjz304, 2024-05-19:
I use PVE to simulate e1000e, with a total of 5 simulated network cards. After adding `e1000e.KumeranLockLoss=1` and netifsort addon, DHCP to IP works normally

eragon4k, 2024-05-19:
> I use PVE to simulate e1000e, with a total of 5 simulated network cards. After adding `e1000e.KumeranLockLoss=1` and netifsort addon, DHCP to IP works normally

I'll give it a try with the netifsort addon

ZMDLBX, 2024-05-20:
24.5.5更新后，我的DSM依旧失联。网卡是Intel I219LM， cpu:Intel 10500，ds918+

nillebor, 2024-05-23:
Same here 24.5.x with intel i5 8600 DS1621xs+ with DSM7.2
Old version works fine. Clean install and online update does the same problem > no connection.

eragon4k, 2024-05-28:
> netifsort

I was busy and didn't have time to test it out e1000e. KumeranLockLoss=1 and netifsort addon. 
Today I tested out with 24.5.6, and the result was negative.

I wonder what has been changed from 24.5.1 to 24.5.6.

wjz304, 2024-05-29:
try
`sed -i 's|/etc/init.d/S41dhcpcd|#/etc/init.d/S41dhcpcd|g' /opt/rr/boot.sh /opt/rr/init.sh`

eragon4k, 2024-05-29:
> try RR shell `sed -i 's|/etc/init.d/S41dhcpcd|#/etc/init.d/S41dhcpcd|g' /opt/rr/boot.sh /opt/rr/init.sh`

I had an expi9301ctblk and just added 9301 NIC and it works for now with 5.6.
Do you know  if the intel 1226v or rtl8125b is supported? 

What is your recommendation for  a 2.5g nic?

wjz304, 2024-05-30:
> > try RR shell `sed -i 's|/etc/init.d/S41dhcpcd|#/etc/init.d/S41dhcpcd|g' /opt/rr/boot.sh /opt/rr/init.sh`
> 
> I had an expi9301ctblk and just added 9301 NIC and it works for now with 5.6. Do you know if the intel 1226v or rtl8125b is supported?
> 
> What is your recommendation for a 2.5g nic?

i226 OK, (No one is currently not available),
R8125* OK, (someone feeds that certain suffixes are unavailable, but no details are provided.

eragon4k, 2024-05-31:
> @eragon4k e1000e try https://github.com/RROrg/rr/releases/tag/24.6.0

It didn't work.
But one wierd behavior.

When it boots, it's not responding from the e1000e, so I plug it into expi9301 and it works, and then while it's turned on, I plug it back into the e1000e and it works.

so it's just not responding when it boots.

nillebor, 2024-06-03:
i'm still with the old version. also the current release does not work

My device: HP Elitedesk G4 800 with Intel I219-LM LAN.
The error occurs only with DSM boot, reconnecting also does not work anything. Everything works as it should in the loader. Of course, clean installation.

Please help and fix it

wjz304, 2024-06-04:
[e1000e.zip](https://github.com/user-attachments/files/15550358/e1000e.zip)

wjz304, 2024-06-04:
To be honest, I don't know why. Since 24.5.1, I haven't modified the driver-related functions. Some issues related to DHCP have also been rolled back in 24.6.0. This problem is a bit difficult to solve.
I don't have a physical e1000e device. I use PVE virtual e1000e. There are no problems in various tests.
I modified the driver again and tested it.

> [e1000e.zip](https://github.com/user-attachments/files/15550358/e1000e.zip) [...]

nillebor, 2024-06-04:
Unfortunately, I have not had any success. I tried 2 DS (gemenilake & broadwellnk) in both versions. The connection works up to the loader. After that, a connection (ping) is no longer possible. Even a reboot does not help.

wjz304, 2024-06-04:
dmesg | grep e1000e
lsmod | grep e1000e

snailium, 2024-06-04:
Just some idea. Is it related to the SN/MAC settings? How is the MAC used and how does DSM use the supplied MAC value?

Maybe try to set mac1 and mac2 to be the real MAC of the NIC (instead of 00:11:32 ones) and see if that works.

wjz304, 2024-06-04:
> Unfortunately, I have not had any success. I tried 2 DS (gemenilake & broadwellnk) in both versions. The connection works up to the loader. After that, a connection (ping) is no longer possible. Even a reboot does not help.

get pid vid (Used to confirm whether it only occurs in a certain model)

nillebor, 2024-06-05:
dmesg | grep e1000e:

```
[   13.660450] e1000e: Intel(R) PRO/1000 Network Driver
[   13.660451] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
[   13.661506] e1000e 0000:00:1f.6: Interrupt Throttling Rate (ints/sec) set to dynamic conservative mode
[   13.810673] e1000e 0000:00:1f.6 0000:00:1f.6 (uninitialized): registered PHC clock
[   13.876500] e1000e 0000:00:1f.6 eth0: (PCI Express:2.5GT/s:Width x1) xx:xx:xx:xx:xx:xx
[   13.876503] e1000e 0000:00:1f.6 eth0: Intel(R) PRO/1000 Network Connection
[   13.876644] e1000e 0000:00:1f.6 eth0: MAC: 13, PHY: 12, PBA No: FFFFFF-0FF
[   18.547463] e1000e 0000:00:1f.6 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
```
Duplicate of #The correct Mac address is displayed, I just anonymized it.

lsmod | grep e1000e:
`e1000e                282624  0`

the same result with true and false option (p):
[image]

nillebor, 2024-06-05:
I select the DS, DSM and add-ons. Then I build the loader and start it. Under 24.4.8 it works without problems. But I need the current modules because of NVMe. An update of the modules is not possible. The network connection is disconnected. Also in the router, the device is no longer displayed as connected.

[image]

snailium, 2024-06-05:
> I select the DS, DSM and add-ons. Then I build the loader and start it. Under 24.4.8 it works without problems. But I need the current modules because of NVMe. An update of the modules is not [...]

What is skip_vendor_mac_interface?

nillebor, 2024-06-05:
The last working version for me is the 24.5.1. (prerelease not tested).

@snailium,
the setting is default. i have also changed nothing about the model, version and NVMe add-on. the same procedure as in the other versions as well. I just need the NVMe fix.

snailium, 2024-06-05:
> The last working version for me is the 24.5.1. (prerelease not tested).
> 
> @snailium,
> the setting is default. i have also changed nothing about the model, version and NVMe add-on. the same procedure as in the other versions as well. I just need the NVMe fix.

I'm asking because I built my loader from scratch and I don't have that skip vendor parameter. It seems it is network related. Maybe try to get rid of that parameter and see if any improvements?

nillebor, 2024-06-05:
this option is under cmdline. There is nothing to be found under cmdline, modules and addons.

[image]

snailium, 2024-06-06:
it seems the skip vendor parameter is added in 24.6.0. I don't see this option in older loader.

nillebor, 2024-06-08:
After the new update 24.6.1, the connection does not work. :(  
How can I help to fix the error?

wjz304, 2024-06-08:
[image]
Change this option to true and try

lyp49472060, 2024-06-08:
24.6.1更新后，我的DSM依旧失联。24.4.6版本正常
网卡是intel 82599 10g，ds920+

wjz304, 2024-06-08:
如果方便就远程（anydesk，todesk，向日葵）吧，QQ: 304403268，TG: https://t.me/wjz304

lyp49472060, 2024-06-08:
> 24.6.1更新后，我的DSM依旧失联。24.4.6版本正常 网卡是intel 82599 10g，固定ip地址，ds920+,

ESXi-8.0U2c 确认和下图一致，
[image]
[image]

删除vmw_pvscsi，正常启动

nillebor, 2024-06-08:
@wjz304,

Thanks, the booting and connecting works. :)
Can you just briefly say what the problem is and why it doesn't work with the default values?

wjz304, 2024-06-08:
> @wjz304,
> 
> Thanks, the booting and connecting works. :) Can you just briefly say what the problem is and why it doesn't work with the default values?

I just guess that it is related to the power management or memory of the network card.
Direct boot will skip the RR system, and the network card will be started under DSM
Indirect boot means that the network card is started under RR, and then kexec to the DSM system to start again. This process may cause power/memory abnormalities of the network card

wjz304, 2024-06-08:
We still need to investigate the details, but it is not easy to investigate the existing logs for problems like this. I don't have the relevant hardware, so this is difficult.

nillebor, 2024-06-08:
> @wjz304,
This process may cause power/memory abnormalities of the network card

wouldn't it be better to make direct boot as standard?

wjz304, 2024-06-08:
> > @wjz304,
> > This process may cause power/memory abnormalities of the network card
> 
> wouldn't it be better to make direct boot as standard?

There is no good or bad difference, the non-direct startup UI just displays more information

[16 more replies]

closed 2024-06-24 by wjz304