RROrg/rr #972: 使用Nvme做引导盘时，DSM内不显示其他非引导的m2盘

gasment, 2024-05-13:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  G4560
 * NIC:  8215b

RR version: (You can find it in the update menu.)  
 * RR:  24.5.3
 * addons:  HDDdb、nvmecache、nvmevolume
 * modules:  
 * lkms:  

DSM: 
 * model:  918+/920+/SA6400
 * version:  7.2.1

Issue:  
使用一个16G傲腾m10作为RR引导盘，系统正常安装，SATA盘全部正常，除引导外的m.2盘则不显示在管理页面，查询ssh是有驱动到并识别，m.2相关插件也已安装
手动使用脚本创建卷也提示失败，尝试编译918+/920+/SA6400均相同问题，使用USB引导盘则M.2正常识别使用
logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

lnc1995, 2024-05-13:
感觉跟我遇到的情况有点像，我是还原lib64/libsynonvme.so.1.bak以后正常。

wjz304, 2024-05-14:
https://github.com/RROrg/rr/releases/tag/24.5.4

closed 2024-05-14 by wjz304