RROrg/rr #987: BUGRRv24.5.3安装DS918+完成重启后显示找不到页面

a117171, 2024-05-14:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  J3355
 * NIC:  igb

RR version: (You can find it in the update menu.)  
 * RR:  24.5.3
 * addons:  latest
 * modules:  latest
 * lkms:  latest

DSM: 
 * model:  SA6400
 * version:  7.2.1up5

Issue:  尝试迁移至DS918+时安装完成重启后显示找不到页面

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-05-18:
https://github.com/RROrg/rr/issues/175

closed 2024-05-18 by wjz304

a117171, 2024-05-19:
ip访问是正常的，安装也正常，就是安装完成重启后无法进入管理UI，显示“抱歉，您所访问的页面不存在”（DSM提示，非浏览器提示）

#175<https://github.com/RROrg/rr/issues/175>

wjz304, 2024-05-19:
http://<ip>:5000