RROrg/rr #1014: hpe Gen10 plus的引导程序从24.4.5升级到24.5.X（最新）后，直通硬盘控制器，有原来组成RAID1的两个硬盘无法识别。

Erignik, 2024-05-18:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  E2136
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  [24.5.4](https://github.com/RROrg/rr/releases/tag/24.5.4)
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DAV3221
 * version:  7.2

Issue:  
 机器是hpe Gen10 plus。引导程序从24.4.5升级到24.5.X（最新）后，直通硬盘控制器有原来组成RAID1的两个硬盘无法识别
我试过升级到24.4.1，也是无法发现原来组成RAID1的两个硬盘，回退到24.4.5就可以识别。
隔离了下，和DSM的类型应该无关，DS3617XS在引导程序为24.4.5也可以正常识别硬盘。
logs:  
需要什么日志请回复我下，我来取
(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

Erignik, 2024-05-19:
> DSM shell
> 
> ```
> ls -d /sys/block/sd*
> grep -E "portcfg|maxdisks" /etc/synoinfo.conf
> ```

这个是重新试了下DS918+在24.5.4引导下的截图，无法发现两个组RAID1的硬盘
[image]

这个是DAV3221在24.4.5引导下的截图，可以发现两个组RAID1的硬盘
[image]

Erignik, 2024-05-19:
> 这个是重新试了下DS918+在24.5.4引导下的截图，无法发现两个组RAID1的硬盘 [...]

当我在引导程序24.5.4中将这几个参数修改成和24.4.5保持一致时，如
[image]
则在安装群晖的时候会爆出无法安装DSM
[image]

wjz304, 2024-05-19:
在24.5.4 中
```
ls -d /sys/block/sd*
grep -E "portcfg|maxdisks" /etc/synoinfo.conf
cat /sys/block/sd*/uevent
```

wjz304, 2024-05-19:
在 卡8% 的时候 http://<ip>:5000/webman/get_logs.cgi

wjz304, 2024-05-19:
https://github.com/RROrg/rr/releases/tag/24.5.5

closed 2024-05-20 by wjz304