RROrg/rr #1015: BUG

Vivants, 2024-05-19:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  AMD 锐龙3 2200G 处理器 (r3) 4核4线程 搭载Radeon Vega8 Graphic 3.5GHz AM4接口
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  [24.5.1](https://github.com/RROrg/rr/releases/tag/24.5.1)
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS3622xs+
 * version:  DSM 7.2.1-69057 Update 5

Issue:  
内存已保留过大，虚拟机无法运行
[image]

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

Vivants, 2024-05-19:
[image]

wjz304, 2024-05-19:
[image]

Vivants, 2024-05-19:
感谢

closed 2024-05-19 by wjz304