RROrg/rr #1031: RR开启后无法正常启动DSM

ShCby, 2024-05-20:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC （Q35 + ICH9, 2009) , BIOS 3.20230228-4 06/06/2023
 * CPU:  INTEL Xeon CPU E5-2690 V4
 * NIC:  X520+virtio-net

RR version: (You can find it in the update menu.)  
 * RR:  24.5.5
 * addons:  待补充
 * modules:  默认全选
 * lkms:  prod（dev也尝试过）

DSM: 
 * model:  DS3622XS+
 * version:  69057 Update 1

Issue:  
RR启动DSM之后，启动日志正常（提示进入linux），但是无法访问管理后台，提示ERR_EMPTY_RESPONSE（共享也不可用），已经尝试过#175的方法（kernel并没有panic），无法启动的概率极大，影响使用体验

机型：戴尔R730XD
RAID卡：PERC H730P Mini（HBA模式）

logs:  
[logs.zip](https://github.com/RROrg/rr/files/15382403/logs.zip)

wjz304, 2024-05-24:
pve vm.conf 看一下

ShCby, 2024-05-26:
> pve vm.conf 看一下

已经破案了lng哥哥，我加了一个virtio的网桥来做虚拟机之间的高速（超10G）通讯，然后在DSM里面设置了静态IP，DSM在启动之后virtio的网卡会和x520那个网卡顺序调换导致这样，加一个sortetif来固定一下网卡顺序就好了（嘻嘻）

closed 2024-05-26 by ShCby