RROrg/rr #1037: 阵列卡有可能实现硬盘正确显示温度吗

gasment, 2024-05-23:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  qemu
 * CPU:  i3-8100T
 * NIC:  MCX4121a

RR version: (You can find it in the update menu.)  
 * RR:  24.5.4
 * addons:  lsiutil
 * modules: 
 * lkms:  

DSM: 
 * model:   SA6400
 * version:  7.2.1

Issue:  
lsi2308 IT模式，smartctl能查到正确温度，网页管理里就是一个固定的（也不是其中某个盘温度），有希望能实现HBA卡的正确温度显示吗
[image]
[image]

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

gasment, 2024-05-25:
> 我是3008 9300-16i的卡也遇到了这个问题，硬盘温度显示29°C，系统版本也是SA6400，之前在3617上2308和3008俩张卡温度显示都正常

应该是DT机型的问题吧，我已经换回来3617xs了能正常显示温度

FireSpell1, 2024-05-25:
我等等看有没有解决方案，温度可以用smartctl设置个脚本，然后通过哪吒探针推送，或是用ipmitool看，机箱上也有温控屏，问题不大

wjz304, 2024-05-26:
DT 型号不在计划内

closed 2024-06-23 by wjz304

ZJ1998888, 2024-09-13:
> > 我是3008 9300-16i的卡也遇到了这个问题，硬盘温度显示29°C，系统版本也是SA6400，之前在3617上2308和3008俩张卡温度显示都正常
> 
> 应该是DT机型的问题吧，我已经换回来3617xs了能正常显示温度
大佬，你是从SA6400  换成  3617xs 硬盘温度就显示了嘛？我使用的是12代的i3 可以使用 3617xs嘛？