RROrg/rr #1043: BUG PhotoFace-Patch

MartinKuhl, 2024-05-24:
Please fill in the following information.  

Install ENV: 
 * CPU:  Intel Atom
 * NIC:  SKY2

RR version: latest 24.5.5

DSM: 
 * model:  3622xs+
 * version:  7.2 Patch 5

Issue:  
The PhotoFace-Patch is not enabling the face recognition in DSM. All tries to recognize the faces ending in timeouts. 
If I directly execute the patch from here: https://github.com/jinlife/Synology_Photos_Face_Patch/blob/main/bin/patch.sh
it is directly working. It there a special order how to install the DSM Photo app and installing the patch from the loader after the  app installation?

wjz304, 2024-05-25:
SynologyPhotos is an application,
Under normal circumstances, the application will not be installed until the system is installed.
The patch in bootloader is run before each system startup, so some modifications to the application do not affect the application at this time.

The corresponding application should be installed after the system installation is completed, and the patch will be re-applied after the next restart.

MartinKuhl, 2024-05-25:
I tried several things, I rebooted DSM after the Synology photos app were installed so that the patch could be applied. Than I  deactivated the addon in the loader rebuild it and than I readded the patch and also rebuild it but nothing changes and the faces were not identified. As soon as I apply the patch directly from the mentioned repository the process of face identification is starting. So maybe the patch from the loader is not executed correctly?

wjz304, 2024-05-25:
Maybe, I'll check if the script is executed correctly,
But I've been busy recently, so it may not be fixed immediately.

wjz304, 2024-05-26:
photosfacepatch addon has bugs

Hesitation: SynologyPhotos is installed in the storage pool, not the system partition, and there are no related operations on the storage pool during the boot stage, so the plug-in does not execute the relevant patch.
Modified to patch through service.

Fixed in next RR version.

MartinKuhl, 2024-05-26:
I will test it as soon as you publish the new version.

wjz304, 2024-05-27:
https://github.com/RROrg/rr/releases/tag/24.5.6

MartinKuhl, 2024-05-27:
I will test it as soon as poosible...

MartinKuhl, 2024-05-27:
> https://github.com/RROrg/rr/releases/tag/24.5.6

I can confirm, now it is working, great job. Thanks a lot

closed 2024-05-27 by MartinKuhl