RROrg/rr #1046: 4个SATA口，12口不识别硬盘，只有34口可以识别

ioioRui, 2024-05-25:
之前用过很老很老的版本可以正常识别，后来一更新就识别不了了！
[image]

wjz304, 2024-05-26:
```
ls -ld /sys/block/s*
lspci -Qnn
```

KSL0ng, 2024-06-03:
我的跟这个老哥的问题一模一样，包括控制芯片，应该是同一款机器，威联通453Bmini，装DS918+ 920+ 7.2.1都是一样的问题，34盘位是原生sata 可以识别， 12盘位不能识别，是另一个sata控制器。
root@ZWL-NAS:~# ls -ld /sys/block/s*
lrwxrwxrwx 1 root root 0 Jun  3 22:47 /sys/block/sata1 -> ../devices/pci0000:00/                                                                                      0000:00:12.0/ata1/host0/target0:0:0/0:0:0:0/block/sata1
lrwxrwxrwx 1 root root 0 Jun  3 22:47 /sys/block/sata2 -> ../devices/pci0000:00/                                                                                      0000:00:12.0/ata2/host1/target1:0:0/1:0:0:0/block/sata2
lrwxrwxrwx 1 root root 0 Jun  3 22:47 /sys/block/synoboot -> ../devices/pci0000:                                                                                      00/0000:00:15.0/usb1/1-4/1-4:1.0/host2/target2:0:0/2:0:0:0/block/synoboot

root@ZWL-NAS:~# lspci -Qnn
0000:00:00.0 Host bridge [0600]: Intel Corporation Celeron N3350/Pentium N4200/A                                                                                      tom E3900 Series Host Bridge [8086:5af0] (rev 0b)
0000:00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 500                                                                                       [8086:5a85] (rev 0b)
0000:00:0e.0 Audio device [0403]: Intel Corporation Celeron N3350/Pentium N4200/                                                                                      Atom E3900 Series Audio Cluster [8086:5a98] (rev 0b)
0000:00:0f.0 Communication controller [0780]: Intel Corporation Celeron N3350/Pe                                                                                      ntium N4200/Atom E3900 Series Trusted Execution Engine [8086:5a9a] (rev 0b)
0000:00:11.0 Unclassified device [0050]: Intel Corporation Celeron N3350/Pentium                                                                                       N4200/Atom E3900 Series Integrated Sensor Hub [8086:5aa2] (rev 0b)
0000:00:12.0 SATA controller [0106]: Intel Corporation Celeron N3350/Pentium N42                                                                                      00/Atom E3900 Series SATA AHCI Controller [8086:5ae3] (rev 0b)
0000:00:13.0 PCI bridge [0604]: Intel Corporation Celeron N3350/Pentium N4200/At                                                                                      om E3900 Series PCI Express Port A #1 [8086:5ad8] (rev fb)
0000:00:13.2 PCI bridge [0604]: Intel Corporation Celeron N3350/Pentium N4200/At                                                                                      om E3900 Series PCI Express Port A #3 [8086:5ada] (rev fb)
0000:00:13.3 PCI bridge [0604]: Intel Corporation Celeron N3350/Pentium N4200/At                                                                                      om E3900 Series PCI Express Port A #4 [8086:5adb] (rev fb)
0000:00:14.0 PCI bridge [0604]: Intel Corporation Celeron N3350/Pentium N4200/At                                                                                      om E3900 Series PCI Express Port B #1 [8086:5ad6] (rev fb)
0000:00:15.0 USB controller [0c03]: Intel Corporation Celeron N3350/Pentium N420                                                                                      0/Atom E3900 Series USB xHCI [8086:5aa8] (rev 0b)
0000:00:1a.0 Serial bus controller [0c80]: Intel Corporation Celeron N3350/Penti                                                                                      um N4200/Atom E3900 Series PWM Pin Controller [8086:5ac8] (rev 0b)
0000:00:1f.0 ISA bridge [0601]: Intel Corporation Celeron N3350/Pentium N4200/At                                                                                      om E3900 Series Low Pin Count Interface [8086:5ae8] (rev 0b)
0000:00:1f.1 SMBus [0c05]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3                                                                                      900 Series SMBus Controller [8086:5ad4] (rev 0b)
0000:02:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network                                                                                       Connection [8086:1539] (rev 03)
0000:03:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network                                                                                       Connection [8086:1539] (rev 03)
0001:01:00.0 SATA controller [0106]: Marvell Technology Group Ltd. 88SE9235 PCIe                                                                                       2.0 x2 4-port SATA 6 Gb/s Controller [1b4b:9235] (rev 11)

原生dts貌似有问题：

KSL0ng, 2024-06-03:
原生dts貌似有问题：
/dts-v1/;

/ {
	compatible = "Synology";
	model = "synology_geminilake_920+";
	version = <0x01>;

	internal_slot@1 {
		protocol_type = "sata";

		ahci {
			pcie_root = "00:12.0";
			ata_port = <0x00>;
		};
	};

	internal_slot@2 {
		protocol_type = "sata";

		ahci {
			pcie_root = "00:12.0";
			ata_port = <0x01>;
		};
	};

	usb_slot@1 {

		usb2 {
			usb_port = "1-1";
		};

		usb3 {
			usb_port = "1-1";
		};
	};

	usb_slot@2 {

		usb2 {
			usb_port = "1-2";
		};

		usb3 {
			usb_port = "1-2";
		};
	};

	usb_slot@3 {

		usb2 {
			usb_port = "1-3";
		};

		usb3 {
			usb_port = "1-3";
		};
	};

	usb_slot@4 {

		usb2 {
			usb_port = "1-4";
		};

		usb3 {
			usb_port = "1-4";
		};
	};

	usb_slot@5 {

		usb2 {
			usb_port = "1-5";
		};

		usb3 {
			usb_port = "1-5";
		};
	};

	usb_slot@6 {

		usb2 {
			usb_port = "1-6";
		};

		usb3 {
			usb_port = "1-6";
		};
	};

	usb_slot@7 {

		usb2 {
			usb_port = "1-7";
		};

		usb3 {
			usb_port = "1-7";
		};
	};

	usb_slot@8 {

		usb2 {
			usb_port = "1-8";
		};

		usb3 {
			usb_port = "1-8";
		};
	};

	usb_slot@9 {

		usb2 {
			usb_port = "2-1";
		};

		usb3 {
			usb_port = "2-1";
		};
	};

	usb_slot@10 {

		usb2 {
			usb_port = "2-2";
		};

		usb3 {
			usb_port = "2-2";
		};
	};

	usb_slot@11 {

		usb2 {
			usb_port = "2-3";
		};

		usb3 {
			usb_port = "2-3";
		};
	};

	usb_slot@12 {

		usb2 {
			usb_port = "2-4";
		};

		usb3 {
			usb_port = "2-4";
		};
	};

	usb_slot@13 {

		usb2 {
			usb_port = "2-5";
		};

		usb3 {
			usb_port = "2-5";
		};
	};

	usb_slot@14 {

		usb2 {
			usb_port = "2-6";
		};

		usb3 {
			usb_port = "2-6";
		};
	};

	usb_slot@15 {

		usb2 {
			usb_port = "2-7";
		};

		usb3 {
			usb_port = "2-7";
		};
	};
};

可能我不会改，改一次也没用

KSL0ng, 2024-06-03:
root@ZWL-NAS:~# cat /sys/block/sata*/device/syno_block_info
pciepath=00:12.0
ata_port_no=0
driver=ahci
pciepath=00:12.0
ata_port_no=1
driver=ahci

dsliboge, 2024-06-04:
一样的机器 威联通453bmini    只有3 4盘能用  貌似有一个 asmedia technology inc. 106x 的控制器识别不了

ioioRui, 2024-06-05:
对的，我就是威联通453bmini，但是以前用的原生的Arpl，旧版本是可以识别1 2口的，但是更新不识别之后再用旧版本Arpl也变成不识别了

dsliboge, 2024-06-16:
GRUB_CMDLINE_LINUX="pcie_port_pm=off"

在GRUB文件里添加一行代码  搞定了

KSL0ng, 2024-06-16:
> GRUB_CMDLINE_LINUX="pcie_port_pm=off"
> 
> 在GRUB文件里添加一行代码 搞定了

具体在哪改呢？我刚才试了以下改\boot\grub\grub.cfg,引导后再RR看磁盘信息还是没有端口啊

dsliboge, 2024-06-16:
我就是在最后面加了一行       开了快速启动

KSL0ng, 2024-06-16:
> 我就是在最后面加了一行 开了快速启动

是U盘里的这个文件吗？重新安装DSM了嘛？

dsliboge, 2024-06-16:
就是引导U盘里的那个文件  你开快速启动试试呢   我换了硬盘 所以DSM也重装了   我没去rr里去看磁盘信息  回家去看看 
https://www.vpslala.com/t/1158  我是参考了这个瞎弄的   居然成功了

KSL0ng, 2024-06-16:
> 就是引导U盘里的那个文件 你开快速启动试试呢 我换了硬盘 所以DSM也重装了 我没去rr里去看磁盘信息 回家去看看 https://www.vpslala.com/t/1158 我是参考了这个瞎弄的 居然成功了

还是搞不定啊。。。

dsliboge, 2024-06-16:
刚刚换了个U盘用最新的24.6.4重新做了一遍，简答说一下我的步骤吧，也很简单

1.用RR编译引导(这里我只添加了DiskIdXMap=0200，其他都没动)
2.RR里高级设置里开启直接启动（上机测试一下，确定下引导没有问题）
3.修改启动U盘的GRUB文件，在最后面添加一行 GRUB_CMDLINE_LINUX="pcie_port_pm=off"  ，添加完就不要进RR了。

现在直接用做好的U盘启动,直接进DSM就能识别4个盘位了（但是RR里磁盘信息里看还是没有端口）

就这样了，祝你好运

KSL0ng, 2024-06-16:
> 刚刚换了个U盘用最新的24.6.4重新做了一遍，简答说一下我的步骤吧，也很简单
> 
> 1.用RR编译引导(这里我只添加了DiskIdXMap=0200，其他都没动) 2.RR里高级设置里开启直接启动（上机测试一下，确定下引导没有问题） 3.修改启动U盘的GRUB文件，在最后面添加一行 GRUB_CMDLINE_LINUX="pcie_port_pm=off" ，添加完就不要进RR了。
> 
> 现在直接用做好的U盘启动,直接进DSM就能识别4个盘位了（但是RR里磁盘信息里看还是没有端口）
> 
> 就这样了，祝你好运

你用的918+  7.2.1吗？我再试试

KSL0ng, 2024-06-16:
> 刚刚换了个U盘用最新的24.6.4重新做了一遍，简答说一下我的步骤吧，也很简单
> 
> 1.用RR编译引导(这里我只添加了DiskIdXMap=0200，其他都没动) 2.RR里高级设置里开启直接启动（上机测试一下，确定下引导没有问题） 3.修改启动U盘的GRUB文件，在最后面添加一行 GRUB_CMDLINE_LINUX="pcie_port_pm=off" ，添加完就不要进RR了。
> 
> 现在直接用做好的U盘启动,直接进DSM就能识别4个盘位了（但是RR里磁盘信息里看还是没有端口）
> 
> 就这样了，祝你好运

解决了，感谢

closed 2025-05-08 by wjz304