RROrg/rr #1051: Fan control sensors BUG

mekler22, 2024-05-26:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i7-8700k 
 * NIC:  1Gbps Wired

RR version: (You can find it in the update menu.)  
 * RR:  24.5.5
 * addons:  24.5.4
 * modules:  24.5.4
 * lkms:  24.5.4

DSM: 
 * model:  DS923+
 * version:  7.2.1-69057 Update 5

Issue:  NCT6775.KO module not loaded and not functioning. Attempt to load it manually results in "unknown symbol" error. sensors-detect command finds the sensor, but does not seem to keep the configuration or to be able to load the said module. Sensors command only shows CPU temperatures rather than the rest of the system.

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-05-26:
请提供 Fancontrol 任务的执行结果

wjz304, 2024-05-27:
modinfo nct6775.ko
depends:        hwmon-vid

mekler22, 2024-05-27:
Got this:
modinfo: ERROR: Module nct6775.ko not found.

wjz304, 2024-05-27:
https://github.com/RROrg/rr/releases/tag/24.5.6

closed 2024-05-29 by wjz304