RROrg/rr #1054: SurveillanceStation组件人脸识别信息添加存在问题

Erignik, 2024-05-26:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  E2136
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  24.5.5
 * addons:  
 * GPU驱动：[NVIDIARuntimeLibrary-x86_64-525.105.17-all_mode-20240308-unsign.spk](https://github.com/pdbear/syno_nvidia_gpu_driver/releases/download/20240310/NVIDIARuntimeLibrary-x86_64-525.105.17-all_mode-20240308-unsign.spk)
 * modules:  
 * lkms:  

DSM: 
 * model:  7.2
 * version:  

Issue:  
DSM的信息如下
[image]
在安装SurveillanceStation后，在新增用户个人资料，点击上传照片时报出错误。试过不同大小的照片，最小的试过72KB的，仍然报出错误。
[image]
[image]

目前看AI的功能也没法使用。

logs:  
日志请回复后我来取

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-05-26:
@pdbear  你编的 兼容 DVA 吗?

wjz304, 2024-05-27:
nvdia 人脸识别 暂未公开

Erignik, 2024-05-27:
> nvdia 人脸识别 暂未公开

既然和引导无关，我把issue关了