RROrg/rr #1066: BUG

marboroman, 2024-05-27:
Install ENV: 
 * DMI:  
 * CPU:  i3 9100
 * NIC:  r8125

RR version: 
 * RR: 24.5.5
 * addons: misc, acpid, reboottoloader, hdddb, storagepanel
 * modules: i915, r8125
 * lkms:  prod

DSM: 
 * model:  423+
 * version:  7.2 69057 u4

Issue:  
I encountered the following problem: periodically, when I turn off DSM via the web interface, the shutdown does not occur and the process freezes somewhere in the middle. I can still connect via ssh, but almost none of the commands work. The only way I can do is a hard reboot with the command "sudo reboot -f". 
Has anyone else encountered this problem? how can it be diagnosed? To exclude the influence of any installed packages or services, I tried removing all disks on the server, installing one unnecessary one for experiments, and making DSM clean install. Next, I performed 10 boot-ups and shutdowns in a row. 2 out of 10 shutdowns are stuck. 
Another important point: for some reason, when you turn it off using the button on the server case, no freezes occurred: 10 out of 10 shutdowns with the button were successful.

wjz304, 2024-05-28:
Need to investigate

wjz304, 2024-05-28:
https://github.com/RROrg/rr/issues/1071

wjz304, 2024-05-31:
plz get /var/log/synopoweroff.log

wjz304, 2024-05-31:
Is has ups?

marboroman, 2024-05-31:
> Is has ups?

no

> plz get /var/log/synopoweroff.log

[synopoweroff.log](https://github.com/user-attachments/files/15510700/synopoweroff.log)

wjz304, 2024-05-31:
[image]
`ASUNAS synoscgi_SYNO.Core.Upgrade_1_start[24265]: system_sys_init.c:68 synopoweroff: Failed to get lock [/tmp/poweroff.lock][0x0400 file_lock_by_file.c:27]`

marboroman, 2024-05-31:
file_lock_by_file - not clear to me. Do you see the exact reason? How can it be solved ?

wjz304, 2024-05-31:
> file_lock_by_file - not clear to me. Do you see the exact reason? How can it be solved ?

Don’t need to pay attention for the time being. I just record the useful information. When I need you to test it, I will @you

bbadm1108, 2024-06-13:
similar issue for me. my PVE is set a timer to poweroff at night but it suspends when shutdown. guess it associates with hardware virtualization and passthrough.
rr: 24.6.3
pve: 8.2
vm: DSM920+ 7.2

wjz304, 2024-06-13:
Probability and complete unavailability are two different issues

wjz304, 2024-06-23:
test latest version ?

bbadm1108, 2024-06-26:
> @bbadm1108 test latest version ?

updated to rr 2.6.7 and up to today it works well without suspending issue in place.

wjz304, 2024-06-26:
@marboroman

marboroman, 2024-06-26:
I experienced freezes only when I was on geminilake. Then I switched to epyc (rr was 24.6.0) and it shutdowns and reboot always were stable.

wjz304, 2024-06-26:
OK

> 我只在双子座湖上经历过冻结。然后我切换到 epyc（rr 是 24.6.0），它关闭和重新启动始终很稳定。