RROrg/rr #1068: Intel ax211无线网卡无法识别

kimyxl-yang, 2024-05-28:
SA6400 7.2安装无线插件之后还是找不到网卡
[image]

wjz304, 2024-05-30:
```DSM shell
lsmod | grep iw
lspci -Qnn | grep 0280
dmesg | grep iwlwifi
```

kimyxl-yang, 2024-06-02:
> DSM shell
> 
> ```
> lsmod | grep iw
> lspci -Qnn | grep 0280
> dmesg | grep iwlwifi
> ```
root@NAS:~# lsmod | grep iw
root@NAS:~# lspci -Qnn | grep 0280
00:14.3 Network controller [0280]: Intel Corporation Raptor Lake-S PCH CNVi WiFi [8086:7a70] (rev 11)
root@NAS:~# dmesg | grep iwlwifi
root@NAS:~# ifconfig
还是找不到网卡

kimyxl-yang, 2024-06-02:
> DSM shell
> 
> ```
> lsmod | grep iw
> lspci -Qnn | grep 0280
> dmesg | grep iwlwifi
> ```
[image]
是不是因为没有AX211的驱动啊

kimyxl-yang, 2024-06-02:
> DSM shell
> 
> ```
> lsmod | grep iw
> lspci -Qnn | grep 0280
> dmesg | grep iwlwifi
> ```
[image]

wjz304, 2024-06-02:
应该不支持

kimyxl-yang, 2024-06-17:
> 应该不支持

[image]
大神你好，我看Intel官网已经有ax211的驱动了，这个驱动要怎么添加呢？

kimyxl-yang, 2024-06-17:
[image]
> 应该不支持

大神你好，我看Intel官网已经有ax211的驱动了，这个驱动要怎么添加呢？

kimyxl-yang, 2024-06-18:
> 没看（5.14） linux kernel 5.14 以上才支持，sa6400 为 5.10， 当然如果你有精力可以移植。

好的，感谢大神，我没那个实力，哈哈，坐等大神跟新吧，实在不行就放弃了

wjz304, 2024-06-19:
目前没有日程安排

closed 2024-06-19 by wjz304