RROrg/rr #1089: 插件打开慢的问题

Erignik, 2024-06-01:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  VMware Inc(Esxi 8.0)
 * CPU:  e2136
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  24.4.5
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DVA3221
 * version:7.2  

Issue:  
安装使用大约一周后，插件打开速度很低（刚安装时速度很快），如file station打开一直在读条，在大约3分钟后弹出界面。synology Photos大约也要等待这么久，同时在套件中心里看到插件，是获取状态中。
[image]
[image]

试过重启DSM虚拟机/重启整机系统也没有改善。DSM虚拟机是安装在Esxi8.0的固态U盘的数据盘中，不知是否有影响。另外，在app
端打开file station/synology Photos等速度很快，不存在等待。

logs:  
有需要等回复我去日志，感谢~

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-01:
今天 都这样

wjz304, 2024-06-01:
[image]
[image]
目测是官方服务器的问题，第三方社群是OK的

congtoumei, 2024-06-01:
折腾了一天，换引导，换版本，换硬盘，能排除的都排除了，看到大家也是这样我就放心了

btou1902, 2024-06-01:
我也有类似问题。12300T+H670主板+rr[24.5.6]。（没用Esxi和pve）

wolfvincent, 2024-06-01:
[image]

看到大家都一样那我就放心了

原本用SSD当主硬盘，刚好昨天换主板+CPU，改M2 SSD，重安装系统后变这样

还以为不相容或是插件有问题，重安装好几次

wjz304, 2024-06-02:
临时解决方案:
DSM shell
```shell
for U in $(synopkg showall | jq -r '.[].thumbnail[0]'); do
  F="$(echo $U | sed 's|https://global.synologydownload.com/download/Package/img|/var/cache/synopkg/lfs/image/AVAIL/SYNO|; s|_72.|_64.|')"
  mkdir -p "$(dirname "${F}")"
  curl -#kL "${U}" -o "${F}"
done
```

congtoumei, 2024-06-02:
大佬，是啥原理呀？

wjz304, 2024-06-03:
官方已经修复 （24.06.03）

closed 2024-06-03 by wjz304