RROrg/rr #1094: 无法识别硬盘，提示无法找到硬盘

purezhang, 2024-06-01:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  Xeon E5606    确实比较老
 * NIC:  i350x4   / 82574L x 2

RR version: (You can find it in the update menu.)  
 * RR:   2.6.0
 * addons:   默认尝试过，增加过hba等相关的
 * modules: 所有 
 * lkms:  

DSM: 
 * model:  6400/3622x2
 * version:   7.2.1

Issue:  
我先在 /dev/sda 安装住了centos-7.9，启动后，用dd把rr.img写在了 /dev/sdg （KIOXIA-32GU盘上），启动都正常。
启动后，选择了所有模块，3622xs+/6400，都尝试过，都是找不到硬盘。

后来尝试安装PVE下再安装rr，都是正常的。在pve下查看了系统硬件状态，请帮忙看下问题，非常感谢。

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined! 

[root@CentOS7-8619 ~]# lsscsi
[0:0:0:0]    disk    ATA      PLEXTOR PX-128M5 1.06  /dev/sda 	PM	ssd-128G	centos-7	86.19
[0:0:1:0]    disk    ATA      ST9500620NS      SN03  /dev/sdb 	PS	hdd-500G-1

[1:0:0:0]    disk    ATA      PLEXTOR PX-128M5 1.05  /dev/sdc 	SM	ssd-128G	rr-2.6.0	??????
[1:0:1:0]    disk    ATA      PLEXTOR PX-128M6 1.02  /dev/sdd 	SS	ssd-128G	pve-8.2		86.31

[2:0:0:0]    disk    ATA      ST9500620NS      AA03  /dev/sde	3M	hdd-500G-2
[3:0:0:0]    disk    ATA      ST91000640NS     AA09  /dev/sdf 	4M	hdd-1t		pve-data
[4:0:0:0]    disk    KIOXIA   TransMemory      PMAP  /dev/sdg 	usb	usb-32G		rr-2.6.0	??????

root@bj-pve-01:~# lspci|grep SATA
00:1f.2 IDE interface: Intel Corporation 82801JI (ICH10 Family) 4 port SATA IDE Controller #1
00:1f.5 IDE interface: Intel Corporation 82801JI (ICH10 Family) 2 port SATA IDE Controller #2
........

root@bj-pve-01:~# udevadm info /dev/sda|grep DEVPATH
E: DEVPATH=/devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0	/block/sda
E: DEVPATH=/devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:1/0:0:1:0	/block/sdb
E: DEVPATH=/devices/pci0000:00/0000:00:1f.2/ata2/host1/target1:0:0/1:0:0:0	/block/sdc
E: DEVPATH=/devices/pci0000:00/0000:00:1f.2/ata2/host1/target1:0:1/1:0:1:0	/block/sdd

E: DEVPATH=/devices/pci0000:00/0000:00:1f.5/ata3/host2/target2:0:0/2:0:0:0	/block/sde
E: DEVPATH=/devices/pci0000:00/0000:00:1f.5/ata4/host3/target3:0:0/3:0:0:0	/block/sdf

E: DEVPATH=/devices/pci0000:00/0000:00:1d.7/usb4/4-1/4-1:1.0/host4/target4:0:0/4:0:0:0	/block/sdg

Getting SMBIOS data from sysfs.
SMBIOS 2.6 present.
70 structures occupying 2853 bytes.
Table at 0x00099C00.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: American Megatrends Inc.
        Version: 2.1a      
        Release Date: 12/30/2011
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 4 MB
......

Base Board Information
        Manufacturer: Supermicro
        Product Name: X8DTL
......

Processor Information
        Socket Designation: CPU 1
        Type: Central Processor
        Family: Xeon
        Manufacturer: Intel            
......
        Version: Intel(R) Xeon(R) CPU           E5606  @ 2.13GHz     
        Voltage: Unknown
        External Clock: 133 MHz
        Max Speed: 2133 MHz
        Current Speed: 2133 MHz
        Status: Populated, Enabled
        Upgrade: Other
        Core Count: 4
        Core Enabled: 4
        Thread Count: 4
......
Processor Information
        Socket Designation: CPU 2
        Type: Central Processor
        Family: Xeon
        Manufacturer: Intel   
......
        Version: Intel(R) Xeon(R) CPU           E5606  @ 2.13GHz     
        Voltage: Unknown
        External Clock: 133 MHz
        Max Speed: 2133 MHz
        Current Speed: 2133 MHz
        Status: Populated, Enabled
        Upgrade: Other
        Core Count: 4
        Core Enabled: 4
        Thread Count: 4
......
Handle 0x002A, DMI type 11, 5 bytes
OEM Strings
        String 1: Intel Nehalem/Tylersburg/ICH10  
        String 2: Supermicro motherboard-X8 Series

Handle 0x002B, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 1
                en|US|iso8859-1
        Currently Installed Language: en|US|iso8859-1
......

wjz304, 2024-06-01:
不支持ide

purezhang, 2024-06-01:
> 不支持ide

感谢大佬回复，是不是bios中选择AHCI就可以了？

purezhang, 2024-06-01:
终于搞定了，这个事情搞了好几个月，断断续续，非常感谢。

closed 2024-06-01 by wjz304