RROrg/rr #1096: BUG quickconnect doesnt work

TTTnas, 2024-06-01:
请填写以下信息.  
Please fill in the following information.  
hi. can you show how to enable quickconnect?. Ths
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-02:
You need genuine SN and MACs to use QC.
Some people on the Internet have shared this information, so find it yourself.

This tool is only used for learning and communication, so I/this organization will not disclose any private property related things.

E-R-Butch, 2024-06-03:
NOT A BUG 
close please

closed 2024-06-03 by wjz304