RROrg/rr #1112: Documentation needed: Proxmox setup

TimKraemer, 2024-06-02:
Hello,

I'm a beginner into the Xpenology / Redpill Recovery topic. I would like to boot my custom .img in proxmox - but I don't have any idea on how to even start with this.
I tried mounting the `rr.img` as a cd drive without success:
[image]
[image]

Any help would be appreciated!
Thank you!

wjz304, 2024-06-02:
[image]

TimKraemer, 2024-06-02:
okay, instead of mounting the .img format directly as cd - instead we can write the image to a disk:
1) upload (custom-) rr.img to proxmox
[image]
2) add new virtual hard-disk to the vm (size should be about the same size as the .img)
[image]
2) go to proxmox host shell, find the path of your newly created disk:
`pvesm path local-lvm:vm-100-disk-0` (replace `vm-100-disk-0` with your disk name)
3) use `dd` to write the `.img` to the new disk based on it's path:
`dd if=/var/lib/vz/template/iso/rr.img of=/dev/pve/vm-100-disk-0 bs=4M status=progress` (replace `rr.img` with your image name and `/dev/pve/vm-100-disk-0` with your disk-path from step 2)
4) after dd is done without errors, setup your vm to boot from this virtual disk by enabling and dragging the disk no the top of boot-order:
[image]
5) start the vm and follow the instructions in console screen
6) consider donating for this awesome project!

With this I got it to work, thank you so much for all your amazing work!

closed 2024-06-02 by TimKraemer

wjz304, 2024-06-02:
qm importdisk 100 /var/lib/vz/template/iso/rr.img local-lvm