RROrg/rr #1132: 群辉 8125 2.5G网卡 无法DHCP

dragontty, 2024-06-05:
昨天折腾黑群晖用PCIE 2.5G网卡.用最新的24.6.0引导后,群辉可以识别8125,驱动之类的也正常,但是无法获取IP,手动指定IP后也ping不通.群辉助手也显示联机失败.
情况是:esxi8.0 直通8125网卡给群辉.主板的网卡可以正常使用,但是8125网卡不行.
帮忙看看呗.
[image]
[image]
[image]
[image]
[image]

wjz304, 2024-06-05:
1. RR 和 DSM 是两个系统。RR下没DHCP 并不代表 DSM 不行，反之也是。
2. get pid vid，

dragontty, 2024-06-05:
按照#173 获取到的信息如下
root@rr:/opt/rr# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX Host bridge [8086:7190] (rev 01)
00:01.0 PCI bridge [0604]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX AGP bridge [8086:7191] (rev 01)
00:07.0 ISA bridge [0601]: Intel Corporation 82371AB/EB/MB PIIX4 ISA [8086:7110] (rev 08)
00:07.1 IDE interface [0101]: Intel Corporation 82371AB/EB/MB PIIX4 IDE [8086:7111] (rev 01)
00:07.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 08)
00:07.7 System peripheral [0880]: VMware Virtual Machine Communication Interface [15ad:0740] (rev 10)
00:0f.0 VGA compatible controller [0300]: VMware SVGA II Adapter [15ad:0405]
00:11.0 PCI bridge [0604]: VMware PCI bridge [15ad:0790] (rev 02)
00:15.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:15.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:16.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:17.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
00:18.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
02:00.0 USB controller [0c03]: VMware USB1.1 UHCI Controller [15ad:0774]
02:01.0 USB controller [0c03]: VMware USB2 EHCI Controller [15ad:0770]
02:02.0 SATA controller [0106]: VMware SATA AHCI controller [15ad:07e0]
03:00.0 SATA controller [0106]: Marvell Technology Group Ltd. 88SE9215 PCIe 2.0 x1 4-port SATA 6 Gb/s Controller [1b4b:9215] (rev 11)
0b:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
13:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 04)
1b:00.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630] [8086:3e92]

是8125中的这个值吗?10ec:8125

dragontty, 2024-06-05:
> 1. RR 和 DSM 是两个系统。RR下没DHCP 并不代表 DSM 不行，反之也是。
> 2. get pid vid，

现在是RR和DSM都没DHCP,刚才不小心点关闭问题了,不好意思

wjz304, 2024-06-09:
https://github.com/RROrg/rr/issues/1137

wjz304, 2024-06-11:
check 24.6.3

dragontty, 2024-06-12:
RR NOT CONNECTED
群辉可以正常识别使用了.
万分感谢.
[image]
[image]

dragontty, 2024-06-12:
删除虚拟网卡后,重启可以分配到IP了.
[image]

sfwwslm, 2024-06-14:
@dragontty 我在半个月前给群晖直通了8125bg网卡，出现了rr 无法dhcp获取ip的情况。最终我把rr引导盘删除，然后新建了一个引导盘后问题解决。新建引导盘不会影响群晖的数据

dragontty, 2024-06-14:
> @dragontty 我在半个月前给群晖直通了8125bg网卡，出现了rr 无法dhcp获取ip的情况。最终我把rr引导盘删除，然后新建了一个引导盘后问题解决。新建引导盘不会影响群晖的数据

我的已经是用最新的RR了,和你的情况不一样.可能是多个驱动有干扰了.在我仅保留R8125的驱动情况下,有所好转,没有那么频繁了,但依然没过几个小时有掉线的情况.
[image]

sfwwslm, 2024-06-14:
@dragontty 自己不要调整设置了，重新引导试试吧。我重新引导后虚拟网卡和物理网卡一切正常。我前面也试了很多种方法都是白费力气

wjz304, 2024-06-21:
https://github.com/RROrg/rr/releases/tag/24.6.6

dragontty, 2024-06-22:
> https://github.com/RROrg/rr/releases/tag/24.6.6

中午更新了,本身系统是静态IP,更新后8125的IP也变成之前同一个IP,导致IP冲突.修改IP后,掉线情况依然存在,后来就IP就变成异常IP了.
[image]

deYangar, 2024-06-27:
补充几点信息：从26号晚上重启之后，我的版本为

deYangar, 2024-06-27:
> https://github.com/RROrg/rr/releases/tag/24.6.6

补充几点信息:
我的主板是华硕z790-p，主板自带的网卡eth0是8125bg，还有个typec拓展坞上的eth1 AX88179A网卡。
首先26号晚上的时候，rr版本24.6.6，一次重启后就检测不到DHCP服务器了，但是在路由器的管理页面上是能看到网卡连上了的。然后用拓展坞网卡连接。用拓展坞AX88179A网卡的时候，会出现两种自动加载驱动的情况，当加载的是ax88179_178a驱动时，也会出现未检测到DHCP服务器的情况。但是如果用的是cdc_ncm驱动，则能正常获取ip。
以上说的情况，手动设置静态IP也无效，也不能进行正常访问。

然后今天看到上面老哥说重建引导盘，就试了一下，顺便更新rr到了24.6.7。并且在编译时，为了避免R8169等驱动的冲突，我也只勾选了8125和下面8125v几这两个驱动，8169，8152，8168等都没有勾选。发现也不能正常获取dhcp，自动获取到的是169.254.187.233这种ip，但也是和之前一样，能在路由器后台看到机器。
[image]
[image]
[image]

对了，还有一点就是ipv4无法获取，但是ipv6是能正常获取的
[image]

下面是log文件
[messages.txt](https://github.com/user-attachments/files/16013016/messages.txt)

sfwwslm, 2024-06-27:
@deYangar  我使用的是引导版本是24.6.0。机型是SA6400，物理和虚拟网卡各一个。编译引导除了选择机型外没有任何其他修改包括勾选或取消驱动。

[image]

wjz304, 2024-06-27:
test https://github.com/RROrg/rr-build/actions/runs/9700319382

deYangar, 2024-06-28:
> test https://github.com/RROrg/rr-build/actions/runs/9700319382

更新到254.6.8后，8125依旧是无法正常获取dhcp，rr和dsm里都是。机型是SA6400，dsm版本是7.2.1-69057 Update 5
[image]
[image]

dawnsummit, 2024-06-28:
I have similar problems. however, 8125 is normal in my station, but I have 2 cards, the other one is a intel 219-v, which can not be found in DSM，although it is really identified when RR launches.

wjz304, 2024-06-28:
dmesg | grep r8125

wjz304, 2024-06-28:
> I have similar problems. however, 8125 is normal in my station, but I have 2 cards, the other one is intel 219-v, which can not be found in DSM，although it is really identified when RR launches.

DSM shell
lspci -Qnnk

deYangar, 2024-06-29:
> > > test https://github.com/RROrg/rr-build/actions/runs/9700319382
> > 
> > 
> > 更新到24.6.8后，8125依旧是无法正常获取dhcp，rr和dsm里都是。机型是SA6400，dsm版本是7.2.1-69057 Update 5 ![1](https://private-user-images.githubusercontent.com/70438628/344130145-a62591e9-a855-4e25-a3a2-e6a9db1f4327.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk2MTAyODAsIm5iZiI6MTcxOTYwOTk4MCwicGF0aCI6Ii83MDQzODYyOC8zNDQxMzAxNDUtYTYyNTkxZTktYTg1NS00ZTI1LWEzYTItZTZhOWRiMWY0MzI3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI4VDIxMjYyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3YmE1MjFiNjMzMjQwN2JlMjUyYWRlN2UzZjM0NjhiN2MxMDVkMjg1NzFmOTNiYTNhY2RhZjNjNTg3ODZlZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ogVpp-q7O6OH8lbdvhEwdfAcd31W3-f5EPlorvG3VqU) ![QQ截图20240628182404](https://private-user-images.githubusercontent.com/70438628/344130160-d17ac8fe-fb8b-4dc7-9589-7b4accaaf5bd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk2MTAyODAsIm5iZiI6MTcxOTYwOTk4MCwicGF0aCI6Ii83MDQzODYyOC8zNDQxMzAxNjAtZDE3YWM4ZmUtZmI4Yi00ZGM3LTk1ODktN2I0YWNjYWFmNWJkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI4VDIxMjYyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEzZGNhNzNkNThhMzQwNWRjMTMxNjlhNjQyMjYwNjllOTJjYjdmMjMyYmM3ZTAyMjU5MTIwOWEyMjExYjUxMDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-mNR8db9K3k7eIrzwe0bofpC8gEWI8m8gRk2QmboAQc)
> 
> dmesg | grep r8125

Yang@Yang-BsNAS:~$ dmesg | grep r8125
[   34.810264] r8125_v9 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded
[   34.821300] r8125_v9 0000:04:00.0: can't disable ASPM; OS doesn't have ASPM control
[   34.849380] r8125_v9: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   34.866106] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
[   69.017729] r8125_v9: eth0: link up
[18998.412604] r8125_v9: eth0: link down

deYangar, 2024-06-29:
> > > > test https://github.com/RROrg/rr-build/actions/runs/9700319382
> > > 
> > > 
> > > 更新到24.6.8后，8125依旧是无法正常获取dhcp，rr和dsm里都是。机型是SA6400，dsm版本是7.2.1-69057 Update 5 ![1](https://private-user-images.githubusercontent.com/70438628/344130145-a62591e9-a855-4e25-a3a2-e6a9db1f4327.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk2MTAyODAsIm5iZiI6MTcxOTYwOTk4MCwicGF0aCI6Ii83MDQzODYyOC8zNDQxMzAxNDUtYTYyNTkxZTktYTg1NS00ZTI1LWEzYTItZTZhOWRiMWY0MzI3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI4VDIxMjYyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3YmE1MjFiNjMzMjQwN2JlMjUyYWRlN2UzZjM0NjhiN2MxMDVkMjg1NzFmOTNiYTNhY2RhZjNjNTg3ODZlZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ogVpp-q7O6OH8lbdvhEwdfAcd31W3-f5EPlorvG3VqU) ![QQ截图20240628182404](https://private-user-images.githubusercontent.com/70438628/344130160-d17ac8fe-fb8b-4dc7-9589-7b4accaaf5bd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk2MTAyODAsIm5iZiI6MTcxOTYwOTk4MCwicGF0aCI6Ii83MDQzODYyOC8zNDQxMzAxNjAtZDE3YWM4ZmUtZmI4Yi00ZGM3LTk1ODktN2I0YWNjYWFmNWJkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI4VDIxMjYyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEzZGNhNzNkNThhMzQwNWRjMTMxNjlhNjQyMjYwNjllOTJjYjdmMjMyYmM3ZTAyMjU5MTIwOWEyMjExYjUxMDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-mNR8db9K3k7eIrzwe0bofpC8gEWI8m8gRk2QmboAQc)
> > 
> > 
> > dmesg | grep r8125
> 
> Yang@Yang-BsNAS:~$ dmesg | grep r8125 [ 34.810264] r8125_v9 2.5Gigabit Ethernet driver 9.009.02-NAPI-RSS loaded [ 34.821300] r8125_v9 0000:04:00.0: can't disable ASPM; OS doesn't have ASPM control [ 34.849380] r8125_v9: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625. [ 34.866106] r8125 Copyright (C) 2022 Realtek NIC software team [nicfae@realtek.com](mailto:nicfae@realtek.com) [ 69.017729] r8125_v9: eth0: link up [18998.412604] r8125_v9: eth0: link down

那个down应该是我凌晨的时候，手动把网线拔了先

wjz304, 2024-06-29:
取消 r8125_v9

dawnsummit, 2024-06-29:
> > I have similar problems. however, 8125 is normal in my station, but I have 2 cards, the other one is intel 219-v, which can not be found in DSM，although it is really identified when RR launches.
> 
> DSM shell
> 
> ```
> lspci -Qnnk
> dmesg | grep e1000e
> ```

00:00.0 Host bridge [0600]: Intel Corporation 12th Gen Core Processor Host Bridge [8086:4650] (rev 05)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] [8086:4692] (rev 0c)
        DeviceName: Onboard - Video
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
        Kernel driver in use: i915
00:06.0 PCI bridge [0604]: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 [8086:464d] (rev 05)
        Kernel driver in use: pcieport
00:08.0 System peripheral [0880]: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator [8086:464f] (rev 05)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:0a.0 Signal processing controller [1180]: Intel Corporation Platform Monitoring Technology [8086:467d] (rev 01)
        DeviceName: Onboard - Other
00:14.0 USB controller [0c03]: Intel Corporation Raptor Lake USB 3.2 Gen 2x2 (20 Gb/s) XHCI Host Controller [8086:7a60] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
        Kernel driver in use: xhci_hcd
00:14.2 RAM memory [0500]: Intel Corporation Raptor Lake-S PCH Shared SRAM [8086:7a27] (rev 11)
        DeviceName: Onboard - Other
00:14.3 Network controller [0280]: Intel Corporation Raptor Lake-S PCH CNVi WiFi [8086:7a70] (rev 11)
        DeviceName: Onboard - Ethernet
        Subsystem: Intel Corporation Device [8086:0244]
00:15.0 Serial bus controller [0c80]: Intel Corporation Raptor Lake Serial IO I2C Host Controller #0 [8086:7a4c] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:15.1 Serial bus controller [0c80]: Intel Corporation Raptor Lake Serial IO I2C Host Controller #1 [8086:7a4d] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:15.2 Serial bus controller [0c80]: Intel Corporation Raptor Lake Serial IO I2C Host Controller #2 [8086:7a4e] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:15.3 Serial bus controller [0c80]: Intel Corporation Device [8086:7a4f] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:16.0 Communication controller [0780]: Intel Corporation Raptor Lake CSME HECI #1 [8086:7a68] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:17.0 SATA controller [0106]: Intel Corporation Raptor Lake SATA AHCI Controller [8086:7a62] (rev 11)
        DeviceName: Onboard - SATA
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
        Kernel driver in use: ahci
00:19.0 Serial bus controller [0c80]: Intel Corporation Device [8086:7a7c] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:19.1 Serial bus controller [0c80]: Intel Corporation Device [8086:7a7d] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:7a3c] (rev 11)
        Kernel driver in use: pcieport
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:7a37] (rev 11)
        Kernel driver in use: pcieport
00:1e.0 Communication controller [0780]: Intel Corporation Device [8086:7a28] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:1e.3 Serial bus controller [0c80]: Intel Corporation Device [8086:7a2b] (rev 11)
        DeviceName: Onboard - Other
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:7a06] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:1f.3 Audio device [0403]: Intel Corporation Raptor Lake High Definition Audio Controller [8086:7a50] (rev 11)
        DeviceName: Onboard - Sound
        Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0897]
00:1f.4 SMBus [0c05]: Intel Corporation Raptor Lake-S PCH SMBus Controller [8086:7a23] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:1f.5 Serial bus controller [0c80]: Intel Corporation Raptor Lake SPI (flash) Controller [8086:7a24] (rev 11)
        DeviceName: Onboard - Other
        Subsystem: Zycoo Co., Ltd Device [6688:0760]
00:1f.6 Ethernet controller [0200]: Intel Corporation Device [8086:0dc8] (rev 11)
        DeviceName: Onboard - Ethernet
        Subsystem: Intel Corporation Device [8086:0000]
01:00.0 SATA controller [0106]: JMicron Technology Corp. JMB58x AHCI SATA controller [197b:0585]
        Subsystem: JMicron Technology Corp. Device [197b:0000]
        Kernel driver in use: ahci
02:00.0 Non-Volatile memory controller [0108]: Phison Electronics Corporation PS5013-E13 PCIe3 NVMe Controller (DRAM-less) [1987:5013] (rev 01)
        Subsystem: Phison Electronics Corporation PS5013-E13 PCIe3 NVMe Controller (DRAM-less) [1987:5013]
        Kernel driver in use: nvme
03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
        Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0123]
        Kernel driver in use: r8125_v9

dmesg |grep e1000e:  no output, nothing found.

dmesg |grep 8125:   2.5gb found.

dawnsummit, 2024-06-30:
> 取消 r8125_v9

why? it's really good for me.

wjz304, 2024-06-30:
[8086:0dc8] is I219_V22, It is too new and is not supported by the current e1000e driver.

wjz304, 2024-06-30:
你们单独创建issue吧，乱糟糟的

wjz304, 2024-06-30:
@dawnsummit 
[e1000e.zip](https://github.com/user-attachments/files/16044222/e1000e.zip) (only SA6400)
[image]
and rebuild

deYangar, 2024-06-30:
> 取消 r8125_v9
rr24.6.9，取消r8125_v9重新编译后依旧不行，rr和dsm里都无法正常dhcp

Yang@Yang-BsNAS:~$ dmesg | grep r8125
[   20.867735] r8125 Ethernet controller driver 9.013.02-NAPI-RSS loaded
[   20.874595] r8125 0000:04:00.0: can't disable ASPM; OS doesn't have ASPM control
[   20.900803] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   20.914262] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[   54.389940] r8125: eth0: link up
[   62.112278] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.181726] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.186826] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.266714] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.346514] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.359937] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.439315] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.508078] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.513192] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.644110] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.713738] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.718840] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.853909] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.922640] r8125 0000:04:00.0 eth0: rss get rxnfc
[   62.927757] r8125 0000:04:00.0 eth0: rss get rxnfc
[   91.370083] r8125 0000:04:00.0 eth0: rss get rxnfc
[   91.439455] r8125 0000:04:00.0 eth0: rss get rxnfc
[   91.444551] r8125 0000:04:00.0 eth0: rss get rxnfc

[dsm内核日志.txt](https://github.com/user-attachments/files/16044479/dsm.txt)
[image]
[image]

wjz304, 2024-06-30:
> > 取消 r8125_v9
> 
> rr24.6.9，取消r8125_v9重新编译后依旧不行，rr和dsm里都无法正常dhcp
> 
> Yang@Yang-BsNAS:~$ dmesg | grep r8125 [ 20.867735] r8125 Ethernet controller driver 9.013.02-NAPI-RSS loaded [ 20.874595] r8125 0000:04:00.0: can't disable ASPM; OS doesn't have ASPM control [ 20.900803] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625. [ 20.914262] r8125 Copyright (C) 2024 Realtek NIC software team [nicfae@realtek.com](mailto:nicfae@realtek.com) [ 54.389940] r8125: eth0: link up [ 62.112278] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.181726] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.186826] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.266714] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.346514] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.359937] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.439315] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.508078] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.513192] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.644110] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.713738] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.718840] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.853909] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.922640] r8125 0000:04:00.0 eth0: rss get rxnfc [ 62.927757] r8125 0000:04:00.0 eth0: rss get rxnfc [ 91.370083] r8125 0000:04:00.0 eth0: rss get rxnfc [ 91.439455] r8125 0000:04:00.0 eth0: rss get rxnfc [ 91.444551] r8125 0000:04:00.0 eth0: rss get rxnfc
> 
> [dsm内核日志.txt](https://github.com/user-attachments/files/16044479/dsm.txt) ![IMG_4934](https://private-user-images.githubusercontent.com/70438628/344463754-4a40c8df-2cc8-46ff-bf06-41cd7738145c.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3NTE0MzgsIm5iZiI6MTcxOTc1MTEzOCwicGF0aCI6Ii83MDQzODYyOC8zNDQ0NjM3NTQtNGE0MGM4ZGYtMmNjOC00NmZmLWJmMDYtNDFjZDc3MzgxNDVjLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjMwVDEyMzg1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmMTc0YjA1YmFkNDllMDBkZmZkOTdhMjY2OGRiNDA0ODc2YmM0NzkyYmFkZWRlODQxOGY0YWNhNGU0YTg1YTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hnE6TRc0BQTRZ-ECNyxBa4bqgV15gfniue9IJcFQgUs) ![QQ截图20240630200452](https://private-user-images.githubusercontent.com/70438628/344463859-92b326d5-aa9d-462b-81bb-0a38a20e9b33.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3NTE0MzgsIm5iZiI6MTcxOTc1MTEzOCwicGF0aCI6Ii83MDQzODYyOC8zNDQ0NjM4NTktOTJiMzI2ZDUtYWE5ZC00NjJiLTgxYmItMGEzOGEyMGU5YjMzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjMwVDEyMzg1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2MDMwYTdlYjVlZjFmYWI2YWIxY2M3YWFiNjcwOGJlYTM0NWE5ZjFhY2NlYjQ2ZTExMTI5Mjg2ODE0MzI0MDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3ixO92DH67Q1qJaWYpNgIWWulDFDygJdFmg9NNWZRJs)

看着没什么问题，而且 ipv6 都是 分配了的，可能ipv4 的 dhcp 模式 不兼容吧

ShmilyStar, 2024-07-01:
woy

> 看着没什么问题，而且 ipv6 都是 分配了的，可能ipv4 的 dhcp 模式 不兼容吧

我也是同样的问题，8125网卡，用最新的24.6.9也是不能获取DHCP

dawnsummit, 2024-07-01:
> @dawnsummit [e1000e.zip](https://github.com/user-attachments/files/16044222/e1000e.zip) (only SA6400) [...]

OK， I try it.

dawnsummit, 2024-07-01:
> OK， I try it.

It works now. Great. thanks.

wjz304, 2024-07-05:
> It works now. Great. thanks.

https://github.com/RROrg/rr/releases/tag/24.7.0

wjz304, 2024-07-19:
> woy
>
>
> 我也是同样的问题，8125网卡，用最新的24.6.9也是不能获取DHCP

@deYangar 
r8125 test
https://github.com/RROrg/rr-build/actions/runs/10013261482

deYangar, 2024-07-21:
> @deYangar r8125 test https://github.com/RROrg/rr-build/actions/runs/10013261482

rr24.7.3,重新刷写的u盘引导，目前试过10次左右，唯一一次能在rr界面dhcp成功，但是进到dsm里依旧是169开头的ip。其他都是直接在rr引导阶段都无法获取ip。网卡在路由器端是检测到了的。由于是实体机安装的，所以只能拍屏，抱歉
[image]
[image]

wjz304, 2024-07-21:
进到 rr 下
```
cat /tmp/messages* | grep dhcpcd
```

deYangar, 2024-07-22:
> 进到 rr 下
> 
> ```
> ethtool eth0
> cat /tmp/messages* | grep dhcpcd
> ```

[image]
[image]

deYangar, 2024-08-02:
rr 24.8.0，目前在rr下，能正常获取到ipv4地址，路由器中也正常分配了。但是进入到dsm中就比较奇怪
[image]
[image]
在dsm中，设置-网络-网络界面 中，显示的依旧为169开头ip，但是用192开头ip可以正常进入web，然后打开套件中心，也是正常联网了的。
[image]
但是Virtual Machine Manager中虚拟机获取到的是169开头的ip，无法正常上网
[image]

HASS-yang, 2024-09-19:
> rr 24.8.0，目前在rr下，能正常获取到ipv4地址，路由器中也正常分配了。但是进入到dsm中就比较奇怪 ![QQ图片20240802184143](https://private-user-images.githubusercontent.com/70438628/354568155-e843a970-92ef-4a94-9e90-5ec128d9d8d4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3NjYxMDgsIm5iZiI6MTcyNjc2NTgwOCwicGF0aCI6Ii83MDQzODYyOC8zNTQ1NjgxNTUtZTg0M2E5NzAtOTJlZi00YTk0LTllOTAtNWVjMTI4ZDlkOGQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE5VDE3MTAwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmODE1NmQwOGFhMWZkYWExMjlmMGY1NTQ2NGVlOTU0NGQxZmZlYjhkY2RiNWIzYTNkY2U0YzE5ODczYzIyMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.EKrlM57QbuYiWcieXWHcFVr5U0bGCAAoEv2LO8lxwQ4) ![image](https://private-user-images.githubusercontent.com/70438628/354568236-e62b557e-45fd-4754-94ff-10e4494365da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3NjYxMDgsIm5iZiI6MTcyNjc2NTgwOCwicGF0aCI6Ii83MDQzODYyOC8zNTQ1NjgyMzYtZTYyYjU1N2UtNDVmZC00NzU0LTk0ZmYtMTBlNDQ5NDM2NWRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE5VDE3MTAwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNlZDFlMzMxZDUyZDdjNzVlYzgxNmQ1Y2ViZGI4OTVlOTMzZDY3ZjkxMDVjYWQ1YTI2ZmQ4MGM1YzVmOWJjMGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Axx3ktLv0-6n7kocW34nTdDt9jeoTyWTO5QEj-exrn4) 在dsm中，设置-网络-网络界面 中，显示的依旧为169开头ip，但是用192开头ip可以正常进入web，然后打开套件中心，也是正常联网了的。 ![QQ截图20240802184312](https://private-user-images.githubusercontent.com/70438628/354568802-126db356-9712-43d0-9be8-7686eb498f64.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3NjYxMDgsIm5iZiI6MTcyNjc2NTgwOCwicGF0aCI6Ii83MDQzODYyOC8zNTQ1Njg4MDItMTI2ZGIzNTYtOTcxMi00M2QwLTliZTgtNzY4NmViNDk4ZjY0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE5VDE3MTAwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhZDYyNGRiMTRlYjZiYzQ0OGZkYTIzYzFhOThmZWE5Y2U3NGMxNTJkYjdhYWU3ZTQzYjBjNjgyN2EwNTFkNzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Z5Gl5Wwt1qqjFug76Q7YR6t9m2_NwF6E92bNQJVHLpU) 但是Virtual Machine Manager中虚拟机获取到的是169开头的ip，无法正常上网 ![image](https://private-user-images.githubusercontent.com/70438628/354569216-3bdb7030-8195-4eda-bef3-d97aff5389d8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3NjYxMDgsIm5iZiI6MTcyNjc2NTgwOCwicGF0aCI6Ii83MDQzODYyOC8zNTQ1NjkyMTYtM2JkYjcwMzAtODE5NS00ZWRhLWJlZjMtZDk3YWZmNTM4OWQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE5VDE3MTAwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkZjJkYjA3YzBhZTg5ZmQzYTU2ZGZlMWI0NDVjYzc0MWZjYzc5N2FiODU3ODhmMmQyNzg2NmQ1ZWRkNTg4YzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DBhQVMag5VU83ZRYzYQ6F6Cr6bbKzW4N5-TtPKJXevo)

请问您是否更新了新的引导？问题解决了吗？

HASS-yang, 2024-09-20:
@deYangar  我也是8125B网卡，请问现在的情况跟你差不多，请问你解决了吗

wjz304, 2024-12-05:
如有其他问题 请以最新版测试反馈

closed 2024-12-05 by wjz304