RROrg/rr #1149: BUG：SA6400搭配MCX311A需要重新加载驱动才能使用

smc326, 2024-06-07:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Default string Default string/Default string, BIOS 5.27 04/02/2024
 * CPU:  Intel(R) N100
 * NIC:  (pid & vid) Mellanox Technologies MT27500 Family [ConnectX-3] [15b3:1003]

RR version: (You can find it in the update menu.)  
 * RR:  24.6.0
 * addons:  acpid cpuinfo mountloader nvmesystem nvmevolume reboottoloader sensors storagepanel
 * modules:  all
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1-69057 Update 5

Issue:  
SA6400 MCX311A网卡不能加载
需要卸载驱动之后，重新加载并重启网络，就能看到这个网卡
logs:  
[image]
[image]

wjz304, 2024-06-07:
[image]
set this to true

smc326, 2024-06-08:
效果一样

smc326, 2024-06-08:
[image]

smc326, 2024-06-08:
还是做个脚本，临时应付吧
rmmod rmmod mlx4_core.ko
insmod /lib/modules/mlx4_core.ko
不知道重启网络命令是什么~

wjz304, 2024-06-08:
应该不需要重启网络， `ifconfig eth2 up` 就可以了

smc326, 2024-06-08:
是可以起来，但是之前指定的IP无法生效，只能改成dhcp再改回来才能生效
[image]

wjz304, 2024-06-11:
check 24.6.3

smc326, 2024-06-15:
24.6.4  网络界面可以显示，但是仍然需要卸载驱动，重新加载，然后刷新网络

smc326, 2024-06-15:
24.6.4  效果一样，同样不好使，有可能是这个主板或者网卡的兼容性或者固件问题吗？

wjz304, 2024-06-15:
你也试试 直接启动吧

JasonMa0012, 2024-06-16:
我今天遇到了同样的问题, 配置基本相同:

----------------------------------------------------------------------------
Install ENV: (You can find it in the boot interface.)

DMI: 未知
CPU: Intel(R) I7 11700T
NIC: (pid & vid) Mellanox Technologies MT27500 Family [ConnectX-3] 
RR version: (You can find it in the update menu.)

RR: 24.6.4
addons: acpid cpuinfo mountloader nvmesystem nvmevolume reboottoloader sensors storagepanel
modules: all
lkms:
DSM:

model: SA6400
version: DSM 7.2.1-69057 Update 5
----------------------------------------------------------------------------

我已开启直接启动, 但DSM仍然无法加载ConnectX-3.

按照[smc326](https://github.com/smc326)的方法也得到了相同的结果, 重新加载驱动后正常:
[image]
[image]
[image]

我的ConnectX-3是连接到PC的, 我还发现无论PC中怎么设置网络连接, 在RR中ConnectX-3始终无法获取IP:
[image]

JasonMa0012, 2024-06-16:
> 还是做个脚本，临时应付吧 rmmod rmmod mlx4_core.ko insmod /lib/modules/mlx4_core.ko 不知道重启网络命令是什么~

想问下大佬计划任务的完整脚本是啥, 我开机后手动执行是可以的, 但自动执行无论sleep 1还是sleep 300都不生效, 可能有啥看不到的报错

wjz304, 2024-06-21:
https://github.com/RROrg/rr/releases/tag/24.6.6

wjz304, 2024-06-25:
https://github.com/RROrg/rr/issues/1379

smc326, 2024-06-29:
新买了4121网卡，mlx5_core的驱动就没有问题，看来mlx4_core这个驱动太老了，可能群晖支持得不是很好了

closed 2024-07-09 by smc326