RROrg/rr #1157: BUG

alekc70, 2024-06-09:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  H610-HDV/M.2
 * CPU:  i5-12400
 * NIC:  (pid & vid) 0x1656 0x0951

RR version: (You can find it in the update menu.)  
 * RR:  24.4.3
 * addons:  acpid, reboottoloader
 * modules:  i915, etc...
 * lkms:  prod

DSM: 
 * model:  SA6400
 * version:  7.2(69057u4)

Issue:  
Hardware transcoding in VideoStation is not working.

logs:  
i915 drive seems to me work well:
sh-4.4# lsmod | grep i915
i915                 2650112  0
i2c_algo_bit           16384  1 i915
drm_buddy              20480  1 i915
intel_gtt              20480  1 i915
drm_display_helper    110592  1 i915
drm_kms_helper         98304  2 drm_display_helper,i915
ttm                    53248  1 i915
drm                   393216  6 drm_kms_helper,drm_display_helper,drm_buddy,i915                                                                                             ,ttm
drm_mipi_dsi           20480  2 i915,drm
i915_compat            24576  3 drm_display_helper,i915,drm
dmabuf                 61440  8 drm_kms_helper,drm_mipi_dsi,drm_display_helper,d                                                                                             rm_buddy,i915,ttm,i915_compat,drm
sh-4.4# dmesg | grep i915
[   27.892412] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   27.899190] i915 0000:00:02.0: [drm] Transparent Hugepage support is recommen                                                                                             ded for optimal performance on this platform!
[   27.911897] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem                                                                                             ,decodes=none:owns=io+mem
[   27.922637] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_                                                                                             dmc_ver2_01.bin (v2.1)
[   27.943054] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin ve                                                                                             rsion 70.20.0
[   27.952124] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin versi                                                                                             on 7.9.3
[   27.963084] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workload                                                                                             s
[   27.971316] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   27.978012] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   27.984418] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[   28.006983] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[   28.014935] i915 display info: display version: 12
[   28.020202] i915 display info: cursor_needs_physical: no
[   28.026039] i915 display info: has_cdclk_crawl: no
[   28.031313] i915 display info: has_cdclk_squash: no
[   28.043829] i915 display info: has_ddi: yes
[   28.048432] i915 display info: has_dp_mst: yes
[   28.053304] i915 display info: has_dsb: yes
[   28.057906] i915 display info: has_fpga_dbg: yes
[   28.062971] i915 display info: has_gmch: no
[   28.067568] i915 display info: has_hotplug: yes
[   28.072554] i915 display info: has_hti: yes
[   28.077161] i915 display info: has_ipc: yes
[   28.081756] i915 display info: has_overlay: no
[   28.086620] i915 display info: has_psr: yes
[   28.091230] i915 display info: has_psr_hw_tracking: no
[   28.096881] i915 display info: overlay_needs_physical: no
[   28.102829] i915 display info: supports_tv: no
[   28.107701] i915 display info: has_hdcp: yes
[   28.112422] i915 display info: has_dmc: yes
[   28.112423] i915 display info: has_dsc: yes
sh-4.4# ls /dev/dri/
by-path  card0  renderD128
(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-09:
不支持 VideoStation  吧，@MoetaYuko

moetayuko, 2024-06-09:
log is okay and I don't use VideoStation

alekc70, 2024-06-09:
@wjz304 @moetayuko Thank you for the quick reply. It seems to me that the DSM should be able to detect the presence of an iGPU, like the one in the DS920+, through certain parameters in the loader or elsewhere. Currently, it functions like dedicated NVIDIA cards, providing GPU acceleration only in applications that support this configuration.

alekc70, 2024-06-09:
suddenly, but hardware transcoding isn't working in syno photos as well

alekc70, 2024-06-30:
There is solution, may be could be included as dedicated patch.
https://xpenology.com/forum/topic/70520-video-station-使用vaapi-硬解-的方法/

wjz304, 2024-10-29:
VideoStation 在 7.2.2 中已被移除，该问题不再跟进

closed 2024-10-29 by wjz304