RROrg/rr #1179: The rr latest versions don't conduct the internet connection correctly for NIC sriov vf iavf driver

bbadm1108, 2024-06-12:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  PVE 8.2
 * CPU:  intel i9-10900t
 * NIC:  (pid & vid) intel x710 DA2

RR version: (You can find it in the update menu.)  
 * RR:  24.4.x-24.6.x
 * addons:  
 * modules:   iavf 
 * lkms:  

DSM: 
 * model:  DS920plus and SA6400 etc
 * version:  7.2.1

Issue:  
PVE has set up the intel x710DA2 nic card in sr-iov mode and generated 6 vfs for each pf,  but vf don't work at all in DSM 920+/SA6400 with the booting message "eth1(iavf)" " "can't find ip" "NO DHCP SERVER DETECTED"...I checked with "ip link show" in PVE console, and found the vf mac was changed to 00-00-00-00-00-00 from original "54-00-00-00-01".

1) All the vfs work fine in windows/openwrt VMs by passthrough,
2) If I pass through the pf directly to rr-DSM, it boots perfectly with drive 'i40e' automatically, only vf with drive iavf lost connection.

my gut feeling is something wrong with driver 'iavf' or DSM MAC rewrite, but it's just a guess.

call for help to fix the problem, thanks

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

bbadm1108, 2024-06-12:
补充，关于mac重写问题。
直接passthru x710进入群晖。
1）ssh到boot查看user-config.yml，发现文件中SN/MAC数据是设定的数值，证明rr配置菜单记录正确。
2）此时系统信息中心，发现SN号正确，但mac地址是另外一个陌生的数值。

wjz304, 2024-06-24:
dmesg | grep iavf

bbadm1108, 2024-06-24:
> dmesg | grep iavf

pve@DS920plus:~$ dmesg | grep iavf
[    5.073885] iavf: Intel(R) Ethernet Adaptive Virtual Function Network Driver - version 4.5.3
[    5.141056] iavf 0000:04:00.0: Invalid MAC address 00:00:00:00:00:00, using random
[    5.141926] iavf 0000:04:00.0: Multiqueue Enabled: Queue pair count = 4
[    5.142788] iavf 0000:04:00.0: MAC address: 72:2c:f1:62:c9:30
[    5.143198] iavf 0000:04:00.0: GRO is enabled
[   26.343270] Modules linked in: i915(+) drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops video backlight i2c_algo_bit dm_crypt sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psnap p8022 llc hfsplus zram md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel kvm_intel kvm irqbypass loop btrfs raid6_pq xor zstd_compress zstd_decompress ecryptfs xxhash sha256_generic synorbd(O) synofsbd(O) hid_generic usbhid hid virtio_net(OE) etxhci_hcd virtio_pci(OE) virtio(OE) virtio_ring(OE) drm drm_panel_orientation_quirks iosf_mbi cfbfillrect cfbimgblt cfbcopyarea iavf(OE) auxiliary(OE) fb fbdev(OE)
[   27.236586] iavf 0000:04:00.0: Failed to add MAC filter, error Unknown status code (update virtchnl.h)
[   27.241050] iavf 0000:04:00.0 eth0: changing MTU from 1500 to 9000
[   27.518259] iavf 0000:04:00.0 eth0: NIC Link is Up Speed is 10 Gbps Full Duplex

wjz304, 2024-06-24:
virtchnl.h 不兼容， 即使 RR 兼容了，DSM也不一定兼容

bbadm1108, 2024-06-24:
> virtchnl.h 不兼容， 即使 RR 兼容了，DSM也不一定兼容

rr引导启动时，vf是可以获取dhcp自动分配一个ip地址，只是passthrough到dsm后，vf mac清零为00:00:00:00:00:00，从此失联。

Niko-1118, 2024-07-07:
https://www.geekxw.top/638/
试试这个，不知道是不是需要下面那个脚本的原因，只能vf网卡之间互通
我的82599也是用的sriov给群晖分配的虚拟网卡是没问题的

bbadm1108, 2024-07-09:
谢谢！

这个脚本曾经试过，未解决vf直通群晖后失联问题，我再研究一下。

bbadm1108, 2024-10-08:
DS920+ / DSM7.2 / rr 24.9.1
今天换了mellanox cx4421a网卡，打开sriov创建vf后直通群晖，问题依旧。
1) 直通的vf mac地址清零，群晖能发现vf网卡但不能联网。
2) 直通pf给群晖可以正常驱动联网。

bbadm1108, 2024-10-25:
有了进展。
1）x710DA2和cx4421a的两个网口，其中网口0可以生成sriov vf，用了最新的rr 24.7.x引导后，直通DSM7.2.2-72806可以连通，mac地址也不再清零了。
2）网口2生成vf失败，只能pf使用。
目前不清楚是rr更新还是DSM更新修复导致的，抑或是b460m主板本身只能生成网口0的vf。此问题暂时解决。

closed 2024-10-25 by bbadm1108