RROrg/rr #1182: IP address issue

azchoi, 2024-06-13:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Native
 * CPU:  Intel Xeon E-2236
 * NIC:  (pid & vid) : Intel X520 (10G)

RR version: (You can find it in the update menu.)  
 * RR:  24.6.0, 24.6.1, 24.6.2, 24.6.3
 * addons:  All addon without NVME, codec patch & wireless
 * modules:  select loaded modules
 * lkms:  

DSM: 
 * model:  DS3622xs+
 * version:  7.2.1-U5

Issue:  

Please check my IP address TIMEOUT issue.
This symptom is only apear in version 24.6 or later.

1. IP address assign failed (timeout)
2. ping 1.1.1.1 failed @ rr console
3. ifconfig shows no IP address @ rr console
4. router cannot find rr
5. Test Result
  1) 23.5.6  : OK
  2) 24.6.0 ~ 24.6.3 : NG

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-13:
rr console
```
cat /tmp/resolv.conf
ip addr show

/etc/init.d/S41dhcpcd restart
cat /tmp/resolv.conf
ip addr show
`

azchoi, 2024-06-15:
> rr console
> 
> ```
> cat /tmp/resolv.conf
> ip addr show
> 
> /etc/init.d/S41dhcpcd restart
> cat /tmp/resolv.conf
> ip addr show
> ```

Works fine...
Well noted... Thank you.

wjz304, 2024-06-19:
the root cause of HDCP failure is not yet clear

azchoi, 2024-06-19:
> the root cause of HDCP failure is not yet clear

I believe this issue can be clear at the soonest time possible. I always thanks for your greate support.

wjz304, 2024-06-23:
test latest version

azchoi, 2024-06-23:
> test latest version

I tried to update 24.6.6 to 24.6.7.
Unfortunately same symptoms appeared.
Does it need clean install instead of update?

[image]

wjz304, 2024-06-24:
可以抓取 RR 下的 /tmp/messages 的文件看看

azchoi, 2024-06-24:
> 可以抓取 RR 下的 /tmp/messages 的文件看看

Please see below log.

[image]

azchoi, 2024-06-24:
> > 可以抓取 RR 下的 /tmp/messages 的文件看看
> 
> Please see below log.
> 
> ![log1](https://private-user-images.githubusercontent.com/111041447/342169579-48ef3a1d-7c22-470e-98c2-23a780c2e015.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkxODkxMzEsIm5iZiI6MTcxOTE4ODgzMSwicGF0aCI6Ii8xMTEwNDE0NDcvMzQyMTY5NTc5LTQ4ZWYzYTFkLTdjMjItNDcwZS05OGMyLTIzYTc4MGMyZTAxNS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyNFQwMDI3MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MmZiZGZkYWNhNzVjODE0NDg5YWExNjFkZDE4MzAyNmI2MGU5YzljNzVlZjYxZTIyMDJkMjg5M2M2NWE4YjljJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2qbAPsx7HAyQlVGIcDQkxeWKAQLrD8QsTTR2B43KBWk)

and only 23.5.6 (or before) works fine. After 24.6.0 has this problem.

wjz304, 2024-06-24:
dmesg | grep ixgbe

azchoi, 2024-06-24:
> dmesg | grep ixgbe

please check below.

[image]

wjz304, 2024-06-24:
Try to remove the sortnetif addon

azchoi, 2024-06-24:
> Try to remove the sortnetif addon

Well, it works like below.
ethernet name has been changed eth4 to eth1.

[image]

wjz304, 2024-06-24:
Got it, it seems some network cards will behave abnormally after rename

azchoi, 2024-06-24:
> Got it, it seems some network cards will behave abnormally after rename

但是使用起来没有异常。
感谢您一直以来的帮助。

[image]

azchoi, 2024-06-27:
> test https://github.com/RROrg/rr-build/actions/runs/9700319382 with enable sortnetif addon

Working good, the issue has been gone and cleared.

I always thanks for your support.

[image]

wjz304, 2024-06-28:
https://github.com/RROrg/rr/releases/tag/24.6.8

This problem is related to dhcpcd. The upper layer can only tolerate and alleviate the symptoms, but cannot eliminate the problem. Wait for the official update.

closed 2024-06-28 by wjz304