RROrg/rr #1212: Activating Spectre & Meltdown protection (DSM 7.2) does not work

nillebor, 2024-06-15:
* CPU:  INtel I5 8th & 9th Generation
 * RR:  24.6.4

DSM: 
 * model:  DS720+/ DS920+
 * version:  7.2 Update 5

Issue:  
Unfortunately, it is not possible to activate the security device for protection against Spectre & Meltdown (DSM 7.2). After activating and restarting, the option is deactivated again.

How can I activate the option? I have tried several different systems with loader, without success. Activation on a DS220 (device not loader) works without problems.

[image]

nillebor, 2024-06-17:
Thanks for your reply. 
I will translate the answer into English for all user:

```
# Mikrocode version
cat /proc/cpuinfo | grep microcode
# Hole information:
curl -kL https://raw.githubusercontent.com/speed47/spectre-meltdown-checker/master/spectre-meltdown-checker.sh | bash
# DSM tool (not available)
/usr/syno/bin/syno_spectre_meltdown_tool -seton ALL

According to the review of system-related tools, the system will modify the boot information to patch related vulnerabilities, but the unofficial boot information is inconsistent with the official boot information, resulting in the system tools are unavailable and the information obtained is inaccurate.，

You can update the microcode by yourself to fix its vulnerabilities.
```
Can you give me some more information?
1. Command displays the microcode.
2. Command is the script for checking 
3. Command activates the syno_spectre_meltdown_tool 

The commands are sent in DSM, not in the loader?
What do you mean by "DSM tool (not available)"?
Is the option not activatable via the DSM GUI, only via ssh?
How can I update the microcode or do you mean to activate of the tool?

wjz304, 2024-06-19:
Because the boot partition structure of RR is different from the official boot partition structure, the official tool modification is not effective.
After a few days of testing, it can be turned on through cmdline, cmdline add SpectreAll_on, this function will be turned on.

[image]

nillebor, 2024-06-19:
Thanks for your test and work.

Maybe you can include this in the command line as an example or better the option to enable / disable it in the advance options?
It is a security feature

wjz304, 2024-06-19:
> Thanks for your test and work.
> 
> Maybe you can include this in the command line as an example or better the option to enable / disable it in the advance options? It is a security feature

I have added it to the cmdline example and will release a new version soon. The advanced options still handle some complex functions.

closed 2024-06-19 by nillebor