RROrg/rr #1214: BUG

StatCombo, 2024-06-16:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

yunzhongci, 2024-06-16:
我也是退回24.6.3才正常

wjz304, 2024-06-16:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
* DMI:  
* CPU:  
* NIC:  (pid & vid)  

RR version: (You can find it in the update menu.)  
* RR:  
* addons:  
* modules:  
* lkms:  

DSM: 
* model:  
* version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

StatCombo, 2024-06-16:
Install ENV: Baremetal UEFi x64

DMI: Filled by O.E.M ASRock Z370M-itx/ac
CPU: Intel core i3 8100
NIC: (can't view pid/vid right Now, but in my mobo built in 2x Ethernet i211AT and i219V)
RR version: (24.6.4)

RR: 26.6.4
addons: latest (after update all)
modules: latest (after update all)
lkms: latest (after update all)

DSM
model: DS918+
version: 7.2.1 u5

wjz304, 2024-06-16:
[rp-lkms-24.5.0.zip](https://github.com/user-attachments/files/15858868/rp-lkms-24.5.0.zip)

```
curl -kL https://github.com/user-attachments/files/15858868/rp-lkms-24.5.0.zip -O /tmp
export DEBUG=1
menu.sh
```
Update menu ->Update LKMs

StatCombo, 2024-06-16:
Ok I try this later

StatCombo, 2024-06-17:
> [rp-lkms-24.5.0.zip](https://github.com/user-attachments/files/15858868/rp-lkms-24.5.0.zip)
> 
> Used RR v24.6.4
> 
> ```
> curl -kL https://github.com/user-attachments/files/15858868/rp-lkms-24.5.0.zip -o /tmp/rp-lkms-24.5.0.zip
> export DEBUG=1
> menu.sh
> ```
> 
> Update menu ->Update LKMs
> 
> rebuild reboot

In my system LKMs updated to 24.5.0 with 24.6.4 rr automatically... But it's not working... Rr works fine, but can't boot to DSM!, or try again update to 24.6.4rr and update lkms manually?

nillebor, 2024-06-17:
> In my system LKMs updated to 24.5.0 with 24.6.4 rr automatically... But it's not working... Rr works fine, but can't boot to DSM!, or try again update to 24.6.4rr and update lkms manually?

You can activate "**direct boot**" in **advanced Settings**. I also had the problem since **24.5.2+**.  Your problem looks very similar.

https://github.com/RROrg/rr/issues/963#issuecomment-2155874296

StatCombo, 2024-06-17:
> > In my system LKMs updated to 24.5.0 with 24.6.4 rr automatically... But it's not working... Rr works fine, but can't boot to DSM!, or try again update to 24.6.4rr and update lkms manually?
> 
> You can activate "**direct boot**" in **advanced Settings**. I also had the problem since **24.5.2+**. Your problem looks very similar.
> 
> [#963 (comment)](https://github.com/RROrg/rr/issues/963#issuecomment-2155874296)

It's stranger, become 24.6.3 RR works perfectly without direct boot...

wjz304, 2024-06-19:
try 24.6.5

StatCombo, 2024-06-19:
I think something wrong, 6.3 works perfectly without any directboot options!

wjz304, 2024-06-19:
> I think something wrong, 6.3 works perfectly without any directboot options!

Get the logs (get the logs that can't be boot)

StatCombo, 2024-06-19:
> > I think something wrong, 6.3 works perfectly without any directboot options!
> 
> Get the logs (get the logs that can't be boot)

Ok I try

wjz304, 2024-06-21:
try https://github.com/RROrg/rr-build/actions/runs/9606242954

yunzhongci, 2024-06-21:
> try https://github.com/RROrg/rr/releases/tag/24.6.6

还是需要在高级设置中激活 直接启动才行

wjz304, 2024-06-21:
> > try https://github.com/RROrg/rr/releases/tag/24.6.6
> 
> 还是需要在高级设置中激活 直接启动才行

没有任何日志，能猜到可能的原因点 都回退了，

yunzhongci, 2024-06-22:
> > > try https://github.com/RROrg/rr/releases/tag/24.6.6
> > 
> > 
> > 还是需要在高级设置中激活 直接启动才行
> 
> 没有任何日志，能猜到可能的原因点 都回退了，

如何查看日志记录？
编译之后，启动，启动不了，需要重启用键盘选项进入arpl，这种情况，日志在哪里查看？
我可以等晚上再测试找一下日志

wjz304, 2024-06-22:
(请先看一下https://github.com/RROrg/rr/issues/173、https://github.com/RROrg/rr/issues/175、#226的内容)
(Plz review the content of https://github.com/RROrg/rr/issues/173, https://github.com/RROrg/rr/issues/175, https://github.com/RROrg/rr/issues/226 first)

StatCombo, 2024-06-22:
> (请先看一下[https://github.com/RROrg/rr/issues/173、https://github.com/RROrg/rr/issues/175、#226的内容](https://github.com/RROrg/rr/issues/173%E3%80%81https://github.com/RROrg/rr/issues/175%E3%80%81#226%E7%9A%84%E5%86%85%E5%AE%B9)) (Plz review the content of #173, #175, #226 first)

So i try to get logs.... added addon dbgutil and rebuild RR. And i am wait 10min

wjz304, 2024-06-22:
If the kernel crashes or the boot disk cannot be recognized, the logs will not be get.

StatCombo, 2024-06-22:
So? No way to understand why?
Direct boot works perfectly...
And 24.6.3 RR without Direct boot works fine. only 24.6.4 and upper not boot without direct mode...

wjz304, 2024-06-22:
The root cause should be Kernel crash.

If you want to solve it, you can only use the elimination method.
But there are 20 or 30 changes, and it is impossible to exclude them one by one.
If all are excluded, it will be the same as 24.6.3, so there is no need to upgrade.

StatCombo, 2024-06-22:
But we need to find why the kernel is crashing in boot process... May be we can test your build of RR one by one with exclude changes? To find out one reason why the kernel is crash?

wjz304, 2024-06-22:
Yes, but it is a long process. The entire compilation process of modifying the kernel takes several hours. And it needs to be tested many times. 

I will modify a few that may be related to startup and compile one.

StatCombo, 2024-06-22:
> Yes, but it is a long process. The entire compilation process of modifying the kernel takes several hours. And it needs to be tested many times.
> 
> I will modify a few that may be related to startup and compile one.

I am ready, I think we find it together.. I can test the build

wjz304, 2024-06-22:
https://github.com/RROrg/rr-build/actions/runs/9626756901
disable: CONFIG_LXT_PHY, CONFIG_INTEL_XWAY_PHY, CONFIG_INTEL_IOMMU_DEFAULT_ON

StatCombo, 2024-06-22:
> https://github.com/RROrg/rr-build/actions/runs/9626756901 disable: CONFIG_LXT_PHY, CONFIG_INTEL_XWAY_PHY, CONFIG_INTEL_IOMMU_DEFAULT_ON

same not boot... same problem! i checked it ...

wjz304, 2024-06-22:
https://github.com/RROrg/rr-build/actions/runs/9627637229
disable CONFIG_EFI_STUB, IOMMU_SVA

wjz304, 2024-06-22:
It takes about 3 hours to make a version. I will annotate them in batches. You can wait for more and test them together.

StatCombo, 2024-06-22:
> It takes about 3 hours to make a version. I will annotate them in batches. You can wait for more and test them together.

it's nevermind. i am download disable CONFIG_EFI_STUB, IOMMU_SVA version!

wjz304, 2024-06-22:
> > It takes about 3 hours to make a version. I will annotate them in batches. You can wait for more and test them together.
> 
> it's nevermind. i am download @disable CONFIG_EFI_STUB, IOMMU_SVA@ version! i am Very want to find ERROR! thnak you very much for supporting! but download speed is SLOW.... and i need only IMG variant!, may be build only one for testing? it's faster for downloading and checking... Best regards!

next

StatCombo, 2024-06-22:
> disable CONFIG_EFI_STUB, IOMMU_SVA

Not boot!....

wjz304, 2024-06-22:
24.6.3 - 24.6.4 All options for kernel modification
```
BR2_PACKAGE_QEMU=y
BR2_PACKAGE_QEMU_TOOLS=y
BR2_PACKAGE_QEMU_GUEST_AGENT=y

CONFIG_VMWARE_BALLOON=m
CONFIG_ATL2=m
CONFIG_IAVF=m
CONFIG_LXT_PHY=m
CONFIG_INTEL_XWAY_PHY=m
CONFIG_USB_NET_RNDIS_WLAN=m
CONFIG_USB4_NET=m
CONFIG_USB_OTG=y
CONFIG_USB_XHCI_HCD=m
CONFIG_USB_XHCI_PCI_RENESAS=m
CONFIG_USB_XHCI_PLATFORM=m
CONFIG_USB_EHCI_HCD=m
CONFIG_USB_EHCI_ROOT_HUB_TT=y
CONFIG_USB_EHCI_PCI=m
CONFIG_USB_EHCI_FSL=m
CONFIG_USB_EHCI_HCD_PLATFORM=m
CONFIG_USB_OHCI_HCD=m
CONFIG_USB_UHCI_HCD=m
CONFIG_USB_SERIAL_CH341=m
CONFIG_USB_SERIAL_CP210X=m
CONFIG_USB_SERIAL_FTDI_SIO=m
CONFIG_USB_SERIAL_PL2303=m
CONFIG_VIRTIO_MMIO_CMDLINE_DEVICES=y
CONFIG_IOMMU_SVA=y
CONFIG_INTEL_IOMMU_SVM=y
CONFIG_INTEL_IOMMU_DEFAULT_ON=y
CONFIG_BCM_KONA_USB2_PHY=m

CONFIG_FORCEDETH=m
CONFIG_SFC=m
CONFIG_SFC_MCDI_MON=y
CONFIG_SFC_SRIOV=y
CONFIG_SFC_MCDI_LOGGING=y
CONFIG_SFC_SIENA=m
CONFIG_SFC_SIENA_MCDI_MON=y
CONFIG_SFC_SIENA_SRIOV=y
CONFIG_SFC_SIENA_MCDI_LOGGING=y
CONFIG_VIA_RHINE=m
CONFIG_VIA_RHINE_MMIO=y
```

StatCombo, 2024-06-22:
try to use 6.3 USB params

wjz304, 2024-06-22:
> try to use 6.3 USB params

Is it easier to try booting using sata? That should eliminate the USB-related options.

StatCombo, 2024-06-22:
> Is it easier to try booting using sata? That should eliminate the USB-related options.

sorry but i can't use the SATA.... my NAS is fully assembled, i can use only USB boot method

StatCombo, 2024-06-22:
> > Is it easier to try booting using sata? That should eliminate the USB-related options.
> 
> sorry but i can't use the SATA as boot drive.... my NAS is fully assembled, i can use only USB boot method

StatCombo, 2024-06-22:
> > > Is it easier to try booting using sata? That should eliminate the USB-related options.

 sorry but i can't use the SATA as boot drive.... my NAS is fully assembled, i can use only USB boot method

StatCombo, 2024-06-22:
> CONFIG_USB_XHCI_HCD=m
> CONFIG_USB_XHCI_PCI_RENESAS=m
> CONFIG_USB_XHCI_PLATFORM=m
> CONFIG_USB_EHCI_HCD=m
> CONFIG_USB_EHCI_ROOT_HUB_TT=y
> CONFIG_USB_EHCI_PCI=m
> CONFIG_USB_EHCI_FSL=m
> CONFIG_USB_EHCI_HCD_PLATFORM=m

somthing here i think...

StatCombo, 2024-06-22:
> > CONFIG_USB_EHCI_HCD=m
> > CONFIG_USB_EHCI_ROOT_HUB_TT=y
> > CONFIG_USB_EHCI_PCI=m
> > CONFIG_USB_EHCI_FSL=m
> > CONFIG_USB_EHCI_HCD_PLATFORM=m
> 
> somthing here i think...

nillebor, 2024-06-22:
> 24.6.3 - 24.6.4 All options for kernel modification

The same [problem](https://github.com/RROrg/rr/issues/963) started with me from version 24.5.3. The last working version was 24.5.1. However, I have not tested the pre-release 24.5.2.  You may still consider the changes from 24.5.1

wjz304, 2024-06-22:
USB is basically an extension of special models. In theory, non-specific devices will not be called. It is mainly virtualization and memory related options.
compiling one version now, the next next version will try to remove USB*

StatCombo, 2024-06-22:
> same [problem](https://github.com/RROrg/rr/issues/963)

i think it's not some problem... first i think we need to find differnce beetwen 6.3 and 6.4 and after find differnce beetwen 5.1 and 5.3

wjz304, 2024-06-22:
> > 24.6.3 - 24.6.4 All options for kernel modification
> 
> The same [problem](https://github.com/RROrg/rr/issues/963) started with me from version 24.5.3. The last working version was 24.5.1. However, I have not tested the pre-release 24.5.2. You may still consider the changes from 24.5.1

5.1 - 5.3 only delete DMRAID，What is your motherboard? Is raid enabled?

nillebor, 2024-06-22:
@wjz304 https://github.com/RROrg/rr/issues/963#issuecomment-2146186355

RAID ist disabled. I have tested all versions before and after. 
Direct boot solves the problem as well as @StatCombo. I can live with it if necessary. But the problem is deeper.

wjz304, 2024-06-22:
The hardest thing is that it only happens in certain circumstances.... :(

StatCombo, 2024-06-22:
> The hardest thing is that it only happens in certain circumstances.... :( The hardware I have is very limited.

we FIND the way together. Now i go to sleep, and tomorrow i will check this topic for your messages!

wjz304, 2024-06-22:
+8   now 
[image]

wjz304, 2024-06-22:
https://github.com/RROrg/rr-build/actions/runs/9628479717/job/26556768743
disable BCM_KONA_USB2_PHY, INTEL_IOMMU_SVM, VIRTIO_MMIO_CMDLINE_DEVICES, USB_NET_RNDIS_WLAN

wjz304, 2024-06-23:
https://github.com/RROrg/rr-build/actions/runs/9629339161
CONFIG_USB4_NET
CONFIG_USB_OTG
CONFIG_USB_XHCI_HCD=m
CONFIG_USB_XHCI_PCI_RENESAS=m
CONFIG_USB_XHCI_PLATFORM=m
CONFIG_USB_EHCI_HCD=m
CONFIG_USB_EHCI_ROOT_HUB_TT=y
CONFIG_USB_EHCI_PCI=m
CONFIG_USB_EHCI_FSL=m
CONFIG_USB_EHCI_HCD_PLATFORM=m
CONFIG_USB_OHCI_HCD=m
CONFIG_USB_UHCI_HCD=m

yunzhongci, 2024-06-23:
> https://github.com/RROrg/rr-build/actions/runs/9629339161 CONFIG_USB4_NET CONFIG_USB_OTG CONFIG_USB_XHCI_HCD=m CONFIG_USB_XHCI_PCI_RENESAS=m CONFIG_USB_XHCI_PLATFORM=m CONFIG_USB_EHCI_HCD=m [...]

今天这两个，也都需要在高级设置中开启直接启动才可以
不开启直接启动：
[image]

开启直接启动重新编译之后：
[image]

实际情况是，不开启直接启动，eth0超时，eth1显示有ip，ip为192.168.1.50，但是实际上无论在路由器还是用群晖的搜索都搜不到
而开启直接启动之后，eth0和eth1都有ip而且可以在路由器和搜索中找到。

yunzhongci, 2024-06-23:
我还怀疑了一下是不是我将两个网口设置了链路聚合的原因，刚才特别测试了一些
跟链路聚合设置无关

wjz304, 2024-06-23:
https://github.com/RROrg/rr-build/actions/runs/9632096082
disable VMWARE_BALLOON,ATL2,IAVF,USB_SERIAL*

StatCombo, 2024-06-23:
hello, i go test all builds...

wjz304, 2024-06-23:
```
curl -kL https://github.com/RROrg/public/releases/download/public/buildroot-24.6.2.zip -o /tmp/buildroot.zip
echo A | unzip /tmp/buildroot.zip -d /mnt/p3
```
Then "menu.sh -> Advanced menu -> Save modifications of '/opt/rr'"
Then reboot
Then test

StatCombo, 2024-06-23:
> buildroot-24.6.2.zip is RR 24.6.3‘s kernel

yep, it;s works. the system are booted up!

wjz304, 2024-06-23:
Could it be qemu?

StatCombo, 2024-06-23:
> Could it be qemu?

You need to test it on QEMU?, i am use baremetal .... for qemu need to stup in pc. but it have another hardware....

wjz304, 2024-06-23:
These three options only increase the qemu function. Of course, the qemu-ga service is enabled by default. In theory, the hardware device (non-qemu virtual machine) will fail and skip this service, which has no effect.

wjz304, 2024-06-23:
CONFIG_FORCEDETH=m
CONFIG_SFC=m
CONFIG_SFC_MCDI_MON=y
CONFIG_SFC_SRIOV=y
CONFIG_SFC_MCDI_LOGGING=y
CONFIG_SFC_SIENA=m
CONFIG_SFC_SIENA_MCDI_MON=y
CONFIG_SFC_SIENA_SRIOV=y
CONFIG_SFC_SIENA_MCDI_LOGGING=y
CONFIG_VIA_RHINE=m
CONFIG_VIA_RHINE_MMIO=y

These are the network card drivers.

StatCombo, 2024-06-23:
> These are the network card drivers.

lets try....
Disable quemu flags... if it's not work may be disable networks flags) may be just hardware conflict?

[15 more replies]

closed 2024-06-23 by wjz304