RROrg/rr #1222: BUG UHD770核显 SR-IOV安装SA6400使用Jellyfin OpenCL色调映射问题

NJcoxx, 2024-06-17:
请填写以下信息.
Please fill in the following information.

Install ENV: (You can find it in the boot interface.)

DMI: ProxmoxVE 8.2.2 QEMU Standard PC (Q35 + ICH9, 2009)
CPU: i3-13100(ES) Q0WL UHD770 Graphics
NIC: R8125B
RR version: (You can find it in the update menu.)

RR: 24.6.3
addons: 24.6.0
modules: 24.6.1
lkms: 24.6.1
DSM:

model: SA6400
version: DSM 7.2.1-69057 Update 5
Issue:
UHD770核显SR-IOV安装SA6400使用Jellyfin OpenCL色调映射问题
虚拟机安装RR+DSM，CPU类型为host，VF拆分为7个，0000:00:02.2分配给虚拟机
在DSM环境下ls /dev/dri 可显示render128、card0

在Docker套件内使用nyanmisaka/jellyfin:latest及nyanmisaka/jellyfin:latest-legacy均出现以下问题：
在启用QSV硬件加速并开启OpenCL色调映射后，视频一直黑屏转圈无法播放(启用VPP映射无此问题)，进入终端后手动运行ffmpeg播放命令出现以下错误:

[Parsed_tonemap_opencl_3 @ 0x55c2f096ad80] Failed to wait for event completion: -14.
Segmentation fault (core dumped)

logs:  
[ffmpelog.txt](https://github.com/user-attachments/files/15782867/ffmpelog.txt)

[image]

@nyanmisaka 劳烦大佬帮忙看看

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-19:
try 24.6.5

NJcoxx, 2024-06-21:
> 尝试 24.6.5

问题依旧
[image]

NJcoxx, 2024-06-21:
> 尝试 24.6.5

还是老问题
[image]

goukey, 2024-06-27:
我也是同样的问题,vpp正常,opencl直接失败.都用6.7最新rr了

NJcoxx, 2024-06-27:
> 我也是同样的问题,vpp正常,opencl直接失败.都用6.7最新rr了

我只是杜比片源的OpenCL色调映射会报错，正常片源没问题

goukey, 2024-06-27:
我环境跟你都差不多,只是我用的12300t

NJcoxx, 2024-06-27:
> 我环境跟你都差不多,只是我用的12300t

你也是SR-IOV拆分核显的吗

goukey, 2024-06-27:
一模一样
pve拆sr-iov
现在我试试回退rr看看行不行

NJcoxx, 2024-06-27:
> 一模一样 pve拆sr-iov 现在我试试回退rr看看行不行

试着在虚拟机配置内加入
args: -set device.hostpci0.addr=02.0 -set device.hostpci0.x-igd-gms=0x2
我在公司暂时没片源 还没试能不能解决

goukey, 2024-06-27:
我在公司远程pve 正在回退rr 重新刷引导

goukey, 2024-06-27:
如果直通整个核显,也就是.0直通,我实验转码正常,看来还是sr-iov驱动问题

goukey, 2024-06-27:
回退到rr三月版本也无效

goukey, 2024-06-27:
> > 一模一样 pve拆sr-iov 现在我试试回退rr看看行不行
> 
> 试着在虚拟机配置内加入 args: -set device.hostpci0.addr=02.0 -set device.hostpci0.x-igd-gms=0x2 我在公司暂时没片源 还没试能不能解决

这样设置后,直接找不到/dev/dri了

NJcoxx, 2024-06-27:
> > > 一模一样 pve拆sr-iov 现在我试试回退rr看看行不行
> > 
> > 
> > 试着在虚拟机配置内加入 args: -set device.hostpci0.addr=02.0 -set device.hostpci0.x-igd-gms=0x2 我在公司暂时没片源 还没试能不能解决
> 
> 这样设置后,直接找不到/dev/dri了

应该是你设置有问题

NJcoxx, 2024-07-02:
补充下 dmesg |grep i915
DSM内的：
cs@JLtest:/$ dmesg |grep i915
[    2.737948] i915 0000:00:02.0: Running in SR-IOV VF mode
[    2.739855] i915 0000:00:02.0: [drm] GT0: GUC: interface version 0.1.9.0
[    2.740545] i915 0000:00:02.0: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
[    2.742641] i915 0000:00:02.0: [drm] GT0: GUC: interface version 0.1.9.0
[    2.744687] i915 0000:00:02.0: [drm] GT0: GUC: interface version 0.1.9.0
[    2.747377] i915 0000:00:02.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    2.748277] i915 0000:00:02.0: HuC firmware PRELOADED
[    2.750285] i915 0000:00:02.0: [drm] PMU not supported for this GPU.
[    2.751093] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[  192.539146] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [13700]
[  192.539736] i915 0000:00:02.0: [drm] ffmpeg[13700] context reset due to GPU hang
[ 1872.325427] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:00000000, in OpenCL-Benchmar [22743]
[ 1872.326403] i915 0000:00:02.0: [drm] OpenCL-Benchmar[22743] context reset due to GPU hang
[ 1898.309536] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:00000000, in OpenCL-Benchmar [22743]
[ 1898.310498] i915 0000:00:02.0: [drm] OpenCL-Benchmar[22743] context reset due to GPU hang
[ 1917.373690] i915 0000:00:02.0: [drm] GT0: GUC: Ignoring context reset notification of exiting context 0x0005 on rcs0

DSM的串口输出报错
[409822.249987] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [9624]
[409822.250511] i915 0000:00:02.0: [drm] ffmpeg[9624] context reset due to GPU hang
[409822.747499] ffmpeg[9624]: segfault at 178 ip 00007ff66b38ca14 sp 00007ffd3beb7c28 error 4 in libOpenCL.so.1.0.0[7ff66b389000+6000]
[409822.748236] Code: 66 0f 1f 44 00 00 ff a0 78 01 00 00 b8 dc ff ff ff c3 0f 1f 40 00 48 8b 05 f1 7b 00 00 48 85 c0 75 14 48 85 ff 74 15 48 8b 07 <ff> a0 78 01 00 00 66 0f 1f 44 00 00 ff a0 80 01 00 00 b8 dc ff ff
[409835.910776] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [9735]
[409835.911276] i915 0000:00:02.0: [drm] ffmpeg[9735] context reset due to GPU hang
[409836.408122] ffmpeg[9735]: segfault at 178 ip 00007f2879545a14 sp 00007ffc2aba1558 error 4 in libOpenCL.so.1.0.0[7f2879542000+6000]
[409836.408836] Code: 66 0f 1f 44 00 00 ff a0 78 01 00 00 b8 dc ff ff ff c3 0f 1f 40 00 48 8b 05 f1 7b 00 00 48 85 c0 75 14 48 85 ff 74 15 48 8b 07 <ff> a0 78 01 00 00 66 0f 1f 44 00 00 ff a0 80 01 00 00 b8 dc ff ff
[409851.771842] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [9912]
[409851.772340] i915 0000:00:02.0: [drm] ffmpeg[9912] context reset due to GPU hang
[409852.269097] ffmpeg[9912]: segfault at 178 ip 00007f7aa10afa14 sp 00007ffcb34a47b8 error 4 in libOpenCL.so.1.0.0[7f7aa10ac000+6000]
[409852.269809] Code: 66 0f 1f 44 00 00 ff a0 78 01 00 00 b8 dc ff ff ff c3 0f 1f 40 00 48 8b 05 f1 7b 00 00 48 85 c0 75 14 48 85 ff 74 15 48 8b 07 <ff> a0 78 01 00 00 66 0f 1f 44 00 00 ff a0 80 01 00 00 b8 dc ff ff

NJcoxx, 2024-07-02:
> try 24.6.5

DSM内的/sys/class/drm/card0/error
GPU HANG: ecode 12:1:00000000, in ffmpeg [13700]
Kernel: 5.10.55+ x86_64
Driver: 20201103
Time: 1719475091 s 980206 us
Boottime: 192 s 542984 us
Uptime: 189 s 801973 us
Capture: 4294859759 jiffies; 1280376 ms ago
Active process (on ring rcs0): ffmpeg [13700]
Reset count: 0
Suspend count: 0
Platform: ALDERLAKE_S
Subplatform: 0x1
PCI ID: 0xa780
PCI Revision: 0x0c
PCI Subsystem: 1462:7d25
IOMMU enabled?: 0
RPM wakelock: yes
PM suspended: no
IER: 0x00000000
DERRMR: 0x00000000
GT awake: yes
  Missing GuC capture node for rcs0
  hung: 1
  engine reset count: 0
  Active context: ffmpeg[13700] prio 0, guilty 0 active 0, runtime total 69218ns, avg 69218ns
  context timeline seqno 1
rcs0 --- HW Status = 0x00000000 fed0e000
:_<mb;!!!H/pRoA'#7kZI!!!!5!"a&;!!!$"
rcs0 --- ring = 0x00000000 fecca000
:cL%-H0Z":2MpMG8paC\b-n,j$!J5k>:r>.%YTpu^!+_Q#i'Jat!cQ""IgYqGX67drrMR#$Vo6-GNiC1$ot)#RBbr/f4dOk]j&OH8=8+3X2CVBeYF[`&S>Z4EhYQF'e8C,TY^I!Ua"j"LI=0mdhJJEUY@jn0T41"V%K#D8zzzqu?]sA31a"!!(@Y
rcs0 --- HW context = 0x00000000 fec33000
:g$P;Sk8S4lp[O"V<ME)6F?2V>6qf[/'I$6*"$t'2UT=n0QITh<K7`3-$W%]F[TXC\$5Xi44<,Qf_4^:B*Xr@,Kn-Zi%#.$Y5XD+o#Cra8ToheR8:!<#Y(7)aZ_$^E,-fIj!h;QqI^$lm43*D08YaAtHH#7(\+)\XO5ID/-E:B*nL&RQO2;7IrpS87:u1'DPbfg$V(##7-?PU(c6a:MMCdkY*G-e@gub]I[[kuggin6ok02<mr108%H,EdK,A:S7CobQt[C/_uRtT#M7a>5Mb8/j)rdUL-s)&5DcHI)?B3h##`Rnt>'&tSSFus.cn'X2'I@#)bGa5tFi&R4G/!^G1+qMY/b5uk2eR1CDDO9]Cf@\]IpPBo!XnRk=jpg.brUFs*H..goQF'Z&IK(o]2Z<N"58h/TVqcJSZ$1Ja="Rq$-Z:fOH&"+b-&Af7Ws)ua=JLI-j/auQ!cU%WRgO-_S1T^[_jjlG2S`RC/e[.rgfNg>A,A2p'L?HbL/,q2,Y`lYLG<GnLp7A6nb&#fT&TaB%m3A^'N(IX=F4F^YZ<NrbZT[XE(:sXL,`r_#0.^+k8)uN_;'I\p%dj-F1V9MB@i'^]P)@%_jIaD`)qaVL9O\kEQ;0kSdjUoM7aSD/"a=1_`nG%c8k[?fQ;N152A/8hQir[GibGilQkH6V\<K5mG!bCm[rK<1APMZB*N=`ET[gOG3R`^j)5ZPD8G'IGR'k4%cQjmnINQ%[<_D>ThYF@hFheZ):Y;RMOcCk=t[RqJ$:5<\"s^d*re05/=UPM'&Vu$rBENao@U"rkHc]He^X<kLp0t!T%3A/F`TV^FttuEhd;l(f>kK^qt%(Z?hKhQ2=oW1q/9hXZmU4J/kTE$H&"+e-&Af7-Ndj0Ybr+Q_V!=o(LZSI9D.8I+Q)&%+,0*,s6<Lg=rTZ?@""C4G\\V9a71)Vc?"E6*8^:VS'c;'p0?HI=#Z$!S&BTck.//q>i&u8DfOq6K_'G+o6O1k]f.m7>FX>lc/UqCk<.V-]\`8)Df*bW-0Fi+k4W,"?1Hdslcf&a4i_t=bDpiQCVG-+S8rj=HbW5U,'m\kSj37O`#@!OL-!9F0])%]j,VMMl%qM/;nuIk-`.7G'ms`T`:UrSnRmJ:e5&b\*)5o]dIIXakG._&M*?*sOF92T%V!^g&RlUK$H?1@7g*g4=C[)ji,!DR']Mjj>8hK>34"5sJ@P3+-,a&*Fq2Ms=UD[e6QT[[b6_W/N[hkIE7QGqN]?>58\gi*:+!iJE+]Pg*cU9XMiFqWfZSrXi8-*&:#N+_]""$<1Y04]%HD>`2hJe^iiX'e42Y<?iU7S-D@uZ!F`&Pa5592_!W99Ar'TE1s+UIe)>H_<ER2Wgq0%1)YUad_^$-MD%X`Nt)OJLI>HKK,QG+I0D.X,2DfJq"^$B0!^:S^_c%!Bt%qQidd?gh)bI9R+75'M415YR!ikMdVs"#\ChXW0d?r$fFI`?$"9m=;1lN-/@XZc/<\Xs)P%L0^tfjIOeB:md=GMg18L\>5IAs^8"f*h@j4$R=?XG/Y*g!@TWn,6hTMLQ@Lp#!bRYJ'92YWb)FY9pG3hYk?Yhns@`%5ZnQp"'/`o=5aQ@Mj.dK7#0]Uu/&L=E``]\l-4OA!`K%O&SttBVJRp(;Cm&on$kPT+q2#aig6M?E@FT`k/BK08"('2?m:^$pNj]jd&VGZ;-[TmM5;ol/.&%iSV=\2hpJgj*A:]?ebJAk!adt.,q9*CUrj:mm[0RD/Ib@3nmAXI\/gr.8!SL8lCnM;O"",Pbfg$V(##7.8!L&8lCnM;O"",Pbfg$V(##7.8!L&8lCnM;O"",Pbfg$V(##7.8!L&8lCnM;O"",Pbfg$V(##7.8!L&8lCnM;O"",Pbfg$V(##7.8!L&8lCnMqNRg*Xg+0m?'l(q)n^Q::Z[/RhcfWlHKiWP?g]QX)eZdd?D"qB0'"IJD=s@;A`>'?QKF_-Ah%s?.7a+Ahbo]8KPMHQG;k;<=Qh<.mpl9FVR[i?6/8=:8lCnM&,kmD_P/0e
rcs0 --- batch = 0x00007f0f c00ac000
:cL%-H5m'id^X*,qC]Xd<LB7\5k^or_7S%Mk)HXlD,m$mh@ESm\TRgnudA6MgO63V35/-IYA_U960$1*DS#>CO;Z#S:?E)Q1X)sTO\TF$DHa<$64SQUlc:_uLqf_Q#fh^83J$I]$lWP1oZ)W3_*ka>:C&M$-qGZE<rI;M=SRIbp!6fn3zzzzzzzzzzzzzzzYODMl*#q?W!!!#P
GuC firmware: PRELOADED
HuC firmware: PRELOADED
GuC timestamp: 0x00000000
GuC CTB fence: 95
GuC Send CTB: raw: 0x00000000, 0x0000022C/0000022C, cached: 0x00000000/0000022C, desc = 0x00000000, buf = 0x00001000 x 0x00000400
GuC Recv CTB: raw: 0x00000000, 0x000000DF/000000DF, cached: 0x000000DF/00000000, desc = 0x00000800, buf = 0x00002000 x 0x00001000
global --- GuC CT buffer = 0x00000000 ccdd7000
:R-gDg73lcm%UVX=o#HBJ*aP'oh:0]:fF7pJ_<,X^Q5_X(%u<h/`4?W?!adc#&&T]F+Okrb/@VF[1,k=,>Y;b;^<6]E^""T&D_D]*,9qqR*/bAVKHUl9*H"1TKKlSf\HkM5OsNYAOnqK,7#hl.`3CUD+B\W`8@Pufs3LQ1,_b,T^6RA,A+Z7/5Q+*"mQ9]th"Wf-DRE3hlIni&%c'J^l=3<KX/5R5\D:kgm^>-YURS2Z[p%R)lQNjdoTap9f2VZcBfAm@FeLn!eG5;,h*T`)>><U=Eiu;)7I$l*n(gdq)LkTHU[8llR8YsU5,>I"Z[?4_lQ$'OiH0pCn8n84A>efdWR/[l_]=cDA(EkG3Z,Y7btuDi>o#hj3i::#S'DYupgoJ#p_EXcC9;PNR*JC.f-*Xo/S/r@FceT6f/1Y2CR`TR6MO98e@lHA/"0[IkiO[f>j<Dj\!>%h=!R"m%`k':3MAO[eL0=!BhuS6<R#rpr,Was)g05*\ZHTe:p*<+E''1<EQr4SqMb>"2.tjZX.%kr7fG3.H`f\1c"sGBV-JOnk81_lqi5FYhpgiI`s(k6H#_1PIpSfJ4R(2\cFj/H/i^[[lg/X_F:e#^F>3XoF9qgTg(A`;O[o/lEgPX"?\9EQr&F^Qn4WVPJ=l+",l^nPVtu;K+4XHN5H1j's&WQZqM+o)of(fAmtP-%hhF@pAUJjQlNn#2Gk\3(c\&;HrZSGQZ8Xuh]^j!qH"654N>LfpQfuP?kf&"1ng@F=Xg5qbmt!Q1IN<p=`qm_GdNlEFo*nCeFk4h5DjRZL\#%Y5c!B1GT)9)gpYg>m,CA$"Y%pr$\^fU/QX+Od)@ruu,DX_5I&R@#n6`IXd>sR=j[`o/.I@(u&;224j[oLCE[Sf:&*A>Kmrp&H[K"O7\)Y4Sp?T9SEO5hK^UJG,k]LM8@='sP**t_jV=nQu*Q-d.h7rEiqW6jdp:elII)!8]h(,m;e.:aa#1)20%PO;ON&b$>i.#)2)]])<8qXla2li5!\,LXHd>mG?23[9m(ImK^:SsIJ@5I-(;phE5kg<2'N:^sC:StJG2:LdB:=64j5-j>iiN2fm301dl6$i=/7d!5WCYM<kK(NtC01LlIoPD2Y7]XpoDS6pkS'cu_hMh1ugE2huhMq8!]-!GT]c$9.B`-#&`SfW[rQHhB64een&;224KHUUf+UCCG#XT)U64een&;224KHUUf+UCCG^VGqaJD.gF
available engines: 41403
slice total: 1, mask=0001
subslice total: 2
slice0: 2 subslices, mask=00000003
EU total: 32
EU per subslice: 16
has slice power gating: yes
has subslice power gating: no
has EU power gating: no
slice0: 2 subslice(s) (0x00000003):
        subslice0: 16 EUs (0xffff)
        subslice1: 16 EUs (0xffff)
        subslice2: 0 EUs (0x0)
        subslice3: 0 EUs (0x0)
        subslice4: 0 EUs (0x0)
        subslice5: 0 EUs (0x0)
graphics version: 12
media version: 12
graphics stepping: D0
media stepping: D0
[... cut ...]
i915.mmio_debug=0
i915.edp_vswing=0
i915.reset=3
i915.inject_probe_failure=0
i915.fastboot=-1
i915.enable_dpcd_backlight=-1
i915.force_probe=*
i915.request_timeout_ms=0
i915.lmem_size=0
i915.lmem_bar_size=0
i915.max_vfs=0
i915.force_disable_ccs=0
i915.enable_mtl_rcs_ccs_wa=yes
i915.enable_hangcheck=yes
i915.load_detect_test=no
i915.force_reset_modeset_test=no
i915.error_capture=yes
i915.disable_display=no
i915.verbose_state_checks=yes
i915.nuclear_pageflip=no
i915.enable_dp_mst=yes
i915.enable_gvt=no

uhbjnm, 2024-07-12:
+1
我的也有这个问题
pve8.2 i7-1365U的CPU 开启SR-IOV 
在虚拟机安装群辉（SA6400） 直通虚拟显卡。
然后在群辉Docker安装jellyfin（nyanmisaka/jellyfin和linuxserver/jellyfin）。
一样的问题。。。不能使用openCL的色调映射
jellyfin ffmpeg日志:
/usr/lib/jellyfin-ffmpeg/ffmpeg -analyzeduration 200M -probesize 1G -init_hw_device vaapi=va:,kernel_driver=i915,driver=iHD -init_hw_device qsv=qs@va -init_hw_device opencl=ocl@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -i file:"/media/link/TV/日韩剧/寄生兽：灰色部队 (2024)/Season 1/寄生兽：灰色部队 - S01E01 - 第 1 集.mkv" -noautoscale -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 h264_qsv -preset veryfast -look_ahead 0 -b:v 31067268 -maxrate 31067268 -bufsize 62134536 -g:v:0 72 -keyint_min:v:0 72 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,hwmap=derive_device=opencl,tonemap_opencl=format=nv12:p=bt709:t=bt709:m=bt709:tonemap=bt2390:peak=100:desat=0,hwmap=derive_device=qsv:reverse=1:extra_hw_frames=16,format=qsv" -codec:a:0 libfdk_aac -ac 2 -ab 256000 -af "volume=2" -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type fmp4 -hls_fmp4_init_filename "e24da20a5c5c9fb418755e7bf1f111da-1.mp4" -start_number 0 -hls_segment_filename "/config/cache/transcodes/e24da20a5c5c9fb418755e7bf1f111da%d.mp4" -hls_playlist_type vod -hls_list_size 0 -y "/config/cache/transcodes/e24da20a5c5c9fb418755e7bf1f111da.m3u8"
......

libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
[Parsed_tonemap_opencl_2 @ 0x55d63a641c80] Failed to wait for event completion: -14.

uhbjnm, 2024-07-12:
> +1 我的也有这个问题 pve8.2 i7-1365U的CPU 开启SR-IOV 在虚拟机安装群辉（SA6400） 直通虚拟显卡。 然后在群辉Docker安装jellyfin（nyanmisaka/jellyfin和linuxserver/jellyfin）。 一样的问题。。。不能使用openCL的色调映射 jellyfin ffmpeg日志: [...]

。。没办法。只能直通编号为0的物理显卡。才解决了！

AustinRayn, 2024-09-16:
我的也有这个问题 pve8.2 i3-12300T 开启SR-IOV 在虚拟机安装群辉2.8.6（SA6400）直通虚拟显卡
硬解有问题

root@DSM-12300T:~# ls -l /dev/dri
total 0
drw-rw-rw- 2 root root              80 Sep 16 00:05 by-path
crw-rw-rw- 1 root root        226,   0 Sep 16 00:06 card0
crw-rw-rw- 1 root videodriver 226, 128 Sep 16 00:06 renderD128
root@DSM-12300T:~# lspci | grep VGA
00:01.0 VGA compatible controller: Device 1af4:1050 (rev 01)
06:10.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
root@DSM-12300T:~# dmesg | grep i915
[    3.678402] i915 0000:06:10.0: Running in SR-IOV VF mode
[    3.690466] i915 0000:06:10.0: [drm] GT0: GUC: interface version 0.1.9.0
[    3.691427] i915 0000:06:10.0: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
[    3.695740] i915 0000:06:10.0: [drm] GT0: GUC: interface version 0.1.9.0
[    3.696445] i915 0000:06:10.0: [drm] GT0: GUC: interface version 0.1.9.0
[    3.697520] i915 0000:06:10.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    3.698036] i915 0000:06:10.0: HuC firmware PRELOADED
[    3.699806] i915 0000:06:10.0: [drm] PMU not supported for this GPU.
[    3.700352] [drm] Initialized i915 1.6.0 20201103 for 0000:06:10.0 on minor 0
[   90.319137] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16022]
[   90.319512] i915 0000:06:10.0: [drm] ffmpeg[16022] context reset due to GPU hang
[  106.319546] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16173]
[  106.319917] i915 0000:06:10.0: [drm] ffmpeg[16173] context reset due to GPU hang
[  123.343193] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16287]
[  123.343573] i915 0000:06:10.0: [drm] ffmpeg[16287] context reset due to GPU hang
[  140.366475] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16375]
[  140.366849] i915 0000:06:10.0: [drm] ffmpeg[16375] context reset due to GPU hang
[  157.326587] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16461]
[  157.327082] i915 0000:06:10.0: [drm] ffmpeg[16461] context reset due to GPU hang
[  174.350779] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16566]
[  174.351272] i915 0000:06:10.0: [drm] ffmpeg[16566] context reset due to GPU hang
[  190.350361] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16653]
[  190.350743] i915 0000:06:10.0: [drm] ffmpeg[16653] context reset due to GPU hang
[  207.310695] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16780]
[  207.311193] i915 0000:06:10.0: [drm] ffmpeg[16780] context reset due to GPU hang
[  224.334482] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16875]
[  224.334854] i915 0000:06:10.0: [drm] ffmpeg[16875] context reset due to GPU hang
[  241.359076] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [16987]
[  241.359448] i915 0000:06:10.0: [drm] ffmpeg[16987] context reset due to GPU hang
[  258.319020] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [17092]
[  258.319388] i915 0000:06:10.0: [drm] ffmpeg[17092] context reset due to GPU hang
[  275.343653] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [17194]
[  275.344235] i915 0000:06:10.0: [drm] ffmpeg[17194] context reset due to GPU hang
[  286.243707] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [17292]
[  286.244195] i915 0000:06:10.0: [drm] ffmpeg[17292] context reset due to GPU hang
[  296.892638] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [17638]
[  296.893134] i915 0000:06:10.0: [drm] ffmpeg[17638] context reset due to GPU hang
[  367.907555] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [17716]
[  367.908039] i915 0000:06:10.0: [drm] ffmpeg[17716] context reset due to GPU hang
[  487.356202] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [17961]
[  487.356583] i915 0000:06:10.0: [drm] ffmpeg[17961] context reset due to GPU hang
[  498.281002] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [18519]
[  498.281392] i915 0000:06:10.0: [drm] ffmpeg[18519] context reset due to GPU hang
[  509.510722] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [18593]
[  509.511114] i915 0000:06:10.0: [drm] ffmpeg[18593] context reset due to GPU hang
[  526.351172] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [18662]
[  526.351546] i915 0000:06:10.0: [drm] ffmpeg[18662] context reset due to GPU hang
[  543.310928] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [18808]
[  543.311477] i915 0000:06:10.0: [drm] ffmpeg[18808] context reset due to GPU hang
[  555.865556] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [18904]
[  555.865946] i915 0000:06:10.0: [drm] ffmpeg[18904] context reset due to GPU hang
[  566.884628] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [19015]
[  566.885061] i915 0000:06:10.0: [drm] ffmpeg[19015] context reset due to GPU hang
[  704.336622] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [19512]
[  704.337108] i915 0000:06:10.0: [drm] ffmpeg[19512] context reset due to GPU hang
[  718.351641] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [19575]
[  718.352018] i915 0000:06:10.0: [drm] ffmpeg[19575] context reset due to GPU hang
[  732.305608] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [19675]
[... cut ...]
[  824.336913] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [20036]
[  824.337277] i915 0000:06:10.0: [drm] ffmpeg[20036] context reset due to GPU hang
[  838.353727] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [20138]
[  838.354232] i915 0000:06:10.0: [drm] ffmpeg[20138] context reset due to GPU hang
[  852.367967] i915 0000:06:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [21021]
[  852.368472] i915 0000:06:10.0: [drm] ffmpeg[21021] context reset due to GPU hang