RROrg/rr #1225: Question on 3622xs and igpu

kills, 2024-06-17:
Wondering if it's possible to enable igpu on 3622xs?
I know previously this was not possible, but I know alot has been improved on in RR, if it might be possible now on later versions?

Thank you.

wjz304, 2024-06-17:
Because except for sa6400, the Linux kernel of other models is version 4.4, and version 4.4 supports up to 10th generation igpu, and now it is 14th generation, no one will spend a lot of effort to port an outdated thing.

wjz304, 2024-06-17:
Off topic: 
I haven't done much research on gpu. Currently, the igpu in RR is mainly done by MoetaYuko, and jim3ma is also working on it. This project is for learning and communication. I can't ask others to do anything, but you can ask them for their ideas.

kills, 2024-06-17:
no problems, I totally understand, thank you.

closed 2024-06-20 by wjz304