RROrg/rr #1228: Any know fixes for mpt3sas raid cards and using SA6400?

kills, 2024-06-18:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Gigabyte H97M-DH3
 * CPU:  Intel 4770k
 * NIC:  RTL8168evl

RR version: (You can find it in the update menu.)  
 * RR:  24.6.4
 * addons:  
 * modules:  mpt3sas
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  7.2.1-69057 Update 5

Issue:  
Using an Avago technologies mpt sas3 bios raid card with disks connected
Disks appear to be functional in DSM, and operating:
[image]

However in the /var/log/messages logs, its cluttered with errors:
```
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8645]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata1/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8648]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern
```

I was hoping there was a way to resolve these types of issues. I have seen posts previously about this: https://github.com/RROrg/rr/issues/174

I red Issue #226, however I was lost in the commands, and I could not run the last command in part 1:
`"curl -sSL --insecure -w '\n' --upload-file ~/model.dts https://transfer.sh"`
I received the following error: 
`curl: (28) Failed to connect to transfer.sh port 443 after 129388 ms: Error`

Are the steps in #266 required to resolve this above issue, or is there something else I need to do to allow the mpt3sas card and attached disks to operate without the above errors appearing in the logs.

Thank you,

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-18:
transfer.sh is just a temporary file transfer tool. You can export model.dts in any other way.

kills, 2024-06-18:
I see :) Silly me :)

I've just copied model.dts to SMB share and I can see the file now:

example:
````
cat ~/model.dts
/dts-v1/;

/ {
        compatible = "Synology";
        model = "synology_epyc7002_sa6400";
        version = <0x01>;

        internal_slot@1 {
                protocol_type = "sata";

                ahci {
                        pcie_root = "0000:00:01.0,00.0";
                        ata_port = <0x00>;
                };
        };

        internal_slot@2 {
                protocol_type = "sata";

                ahci {
                        pcie_root = "0000:00:01.0,00.0";
                        ata_port = <0x01>;
```

so I need to take the details from:
`cat /sys/block/sata*/device/syno_block_info`

```
pciepath=0000:00:01.0,00.0
ata_port_no=0
driver=ahci
pciepath=0000:00:01.0,00.0
ata_port_no=1
driver=ahci
pciepath=0000:00:01.0,00.0
ata_port_no=2
driver=ahci
pciepath=0000:00:01.0,00.0
ata_port_no=3
driver=ahci
pciepath=0000:00:01.0,00.0
ata_port_no=4
driver=ahci
pciepath=0000:00:01.0,00.0
ata_port_no=5
driver=ahci

```

And add this to my ~/model.dts file as per the screenshot in 2.2 of #226 

Once I update model.dts. Do I just reboot and it takes effect? or do I need to rebuilt DSM again and add the custom DTS file?

Thank you for your help @wjz304 <3

wjz304, 2024-06-18:
need add the custom DTS file and rebuilt

kills, 2024-06-18:
I appear to be missing something from the files or they are not displaying correctly for me. as I don't see anything on my end related to updating a file like below (taken from #266)
[image]

my cat /sys/block/sata*/device/syno_block_info is basically empty as shown above

I'm not sure why though.

kills, 2024-06-18:
> need add the custom DTS file and rebuilt

Do you recommend what I could do / try? Or is there more to that guide in #226

kills, 2024-06-18:
Does completing these steps, stop DSM from erroring with these messages in the /var/log/messages file?

```
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8645]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata1/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8648]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern
```

wjz304, 2024-06-18:
> 完成这些步骤，是否可以阻止 DSM 在 /var/log/messages 文件中出现这些消息错误？
> 
> ```
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8645]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata1/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8648]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern
> ```

dts has nothing to do with "Can't get sata chip name"

wjz304, 2024-06-18:
> I appear to be missing something from the files or they are not displaying correctly for me. as I don't see anything on my end related to updating a file like below (taken from #226 ) [...]

This screenshot was taken a long time ago. Later, the LED was removed because it didn’t play a big role.

wjz304, 2024-06-18:
> 完成这些步骤，是否可以阻止 DSM 在 /var/log/messages 文件中出现这些消息错误？
>
> ```
> 2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8645]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern [...]

This log doesn't have much impact. but it is also troublesome to fix it

kills, 2024-12-16:
`2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8645]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata1/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8657]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2024-06-18T20:59:04+10:00 testdsm synostgd-disk[8654]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name`

Wondering if over the past 6 months if there's been any progress in resolving these errors that are reported for mpt3sas cards on sa6400 builds?

I know its only in the logs, but it pollutes the logs so much. Wondering if there are any ways around it.

Has anyone else been building systems with an mpt3sas controller and seeing these messages all through their logs?

wjz304, 2025-02-16:
https://github.com/RROrg/rr/releases/tag/25.2.0

closed 2025-02-16 by wjz304