RROrg/rr #1229: sa6400 , 请问13代CPU支持硬解吗？

yangdong123, 2024-06-18:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i5 13500t
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

NJcoxx, 2024-06-19:
支持，可以去看我的issue，但是我用Sr-iov拆分核显OpenCL色调映射会有问题，你可以试试物理机
[https://github.com/RROrg/rr/issues/1222](url)

yangdong123, 2024-06-19:
> 支持，可以去看我的issue，但是我用Sr-iov拆分核显OpenCL色调映射会有问题，你可以试试物理机 [#1222](https://github.com/RROrg/rr/issues/1222)

感谢！

closed 2024-06-19 by wjz304