RROrg/rr #1235: BUG

cjwkof, 2024-06-19:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  未知
 * CPU:  N100
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.6.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model: SA6400
 * version:  7.2

Issue: 从24.6.0升级到24.6.4， i915驱动初始化失败。降到24.6.1正常。

logs:  
[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

aat002, 2024-06-19:
26.4.3升到26.4.4后，jellyfin也无法硬解了，n300的U

ywxkv5, 2024-06-19:
同样是n100，物理机直接安装，从24.3.几升级到24.6.4后docker里的emby无法硬解

cjwkof, 2024-06-19:
> 24.6.3升到24.6.4后，jellyfin也无法硬解了，n300的U

我降级引导是可以了

wjz304, 2024-06-19:
24.6.5

aat002, 2024-06-20:
> 24.6.5

升级24.6.5可以硬解了

cjwkof, 2024-06-20:
已经解决