RROrg/rr #1294: 重新安装群晖系统一直提示文件可能损毁

yelijun, 2024-06-20:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  celeron 2957u
 * NIC:  (pid & vid) vid: 0x054c; pid: 0x02a5

RR version: (You can find it in the update menu.)   **default options**
 * RR:  v24.6.3
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  **7.0、7.2**

Issue:  
[image]
全新安装，一直提示这个，换了satat、msata 都这样。

之前安装7.2成功过，格式化引导盘、数据盘后，这个设备就一直提示这个错误。换优盘、sata硬盘仍然这样。

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

leehomkey, 2024-06-23:
我也有这个问题，怎么搞

leehomkey, 2024-06-23:
试了换到这个版本就好了。[24.5.3](https://github.com/RROrg/rr/releases/tag/24.5.3)

vzrs, 2024-06-23:
24.6.5就行，安装成功了。

yelijun, 2024-06-24:
我今天晚点试试新版本。不行我就一个一个版本尝试😄。成功后我再来和大家同步。

liuqq58, 2024-09-18:
同样的问题，重新安装群晖系统，重新编译下载，在RR界面是正常的，编译完成启动后，打开群晖的界面，然后还需要下载一次系统，不论选择从群晖下载自动安装还是上传安装的系统文件，都提示文件损坏，无法安装

liuqq58, 2024-09-20:
> 同样的问题，重新安装群晖系统，重新编译下载，在RR界面是正常的，编译完成启动后，打开群晖的界面，然后还需要下载一次系统，不论选择从群晖下载自动安装还是上传安装的系统文件，都提示文件损坏，无法安装

更正：全新安装24.9.1安装系统不会报错群晖系统损坏正常安装，但是从之前版本升级到24.9.1后重新安装群晖系统会报错群晖系统文件损坏，不确定原因，未测试过其他版本。

wjz304, 2024-12-05:
如有其他问题 请以最新版测试反馈

closed 2024-12-05 by wjz304