RROrg/rr #1378: BUG AME 即使全白 mac+SN 也无法激活

tearfulDalvik, 2024-06-23:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.6.6
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1-69057 Update 5

Issue:  
类似 https://github.com/RROrg/rr/issues/393，synoboot 分区挂载不上。RR 默认设置，只改了 SN 和 Mac

logs:  

2024-06-22T20:25:21-07:00 synoscgi_SYNO.AME.License_1_check[24791]: boot/boot_lock.c(259): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:13)
2024-06-22T20:25:21-07:00 synoscgi_SYNO.AME.License_1_check[24791]: utils.cpp:18 init fail
2024-06-22T20:25:21-07:00 synoscgi_SYNO.AME.License_1_check[24791]: boot/boot_lock.c(259): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:13)
2024-06-22T20:25:21-07:00  synoscgi_SYNO.AME.License_1_check[24791]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
2024-06-22T20:25:21-07:00 synoscgi_SYNO.AME.License_1_check[24791]: license.cpp:39 Failed to create LicenseManger

如果试图手动 mount：
sudo mount /dev/synoboot2 /tmp/bootmnt
mount: /tmp/bootmnt: wrong fs type, bad option, bad superblock on /dev/synoboot2, missing codepage or helper program, or other error.

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

tearfulDalvik, 2024-06-23:
如果 sudo /usr/syno/etc.defaults/rc.sysv/apparmor.sh stop 关掉 apparmor，挂载失败的消息消失，错误变成：

2024-06-22T20:53:52-07:00 synoscgi_SYNO.AME.License_1_check[28459]: api.cpp:561 API failed: POST /v1/user/login, response: {"data":{},"errno":256,"lineno":159,"status":false}
2024-06-22T20:53:52-07:00 synoscgi_SYNO.AME.License_1_check[28459]: user.cpp:178 SLSendAPI fail
2024-06-22T20:53:52-07:00 synoscgi_SYNO.AME.License_1_check[28459]: utils.cpp:21 not login yet
2024-06-22T20:53:52-07:00 synoscgi_SYNO.AME.License_1_check[28459]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
2024-06-22T20:53:52-07:00 synoscgi_SYNO.AME.License_1_check[28459]: license.cpp:39 Failed to create LicenseManger

wjz304, 2024-06-23:
/usr/syno/etc/rc.sysv/apparmor.sh remove_packages_profile 0 CodecPack
mv /var/packages/CodecPack/target/apparmor /var/packages/CodecPack/target/apparmor.bak

就行了， 关掉可能会产生大量日志

cz132023cz, 2024-06-24:
确保启动盘是sata的，不应该是其它。目前sata启动最稳定，AME能激活。（SCSI不行）

tearfulDalvik, 2024-06-26:
> 确保启动盘是sata的，不应该是其它。目前sata启动最稳定，AME能激活。（SCSI不行）

启动盘是 SATA 的

> /usr/syno/etc/rc.sysv/apparmor.sh remove_packages_profile 0 CodecPack mv /var/packages/CodecPack/target/apparmor /var/packages/CodecPack/target/apparmor.bak
> 
> 就行了， 关掉可能会产生大量日志
好的，谢谢。我回头试试

tearfulDalvik, 2024-07-03:
> /usr/syno/etc/rc.sysv/apparmor.sh remove_packages_profile 0 CodecPack mv /var/packages/CodecPack/target/apparmor /var/packages/CodecPack/target/apparmor.bak
> 
> 就行了， 关掉可能会产生大量日志

我能确定是全白的，quickconnect 都能用

还是不行，登录后提示无法完成操作：
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: api.cpp:561 API failed: POST /v1/user/login, response: {"data":{},"errno":256,"lineno":159,"status":false}
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: user.cpp:178 SLSendAPI fail
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: utils.cpp:21 not login yet
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: license.cpp:101 Failed to create "LicenseManager"

再次打开，日志如下。提示无法完成操作，需要重新登录，重新登录后，又是上面的情况
2024-07-03T09:30:34+08:00 synoscgi_SYNO.Core.UserSettings_1_apply[7333]: volume_offline_op_check.c:15 Fail to get space info of [/volume2]
2024-07-03T09:30:35+08:00 synoscgi_SYNO.Core.UserSettings_1_apply[7321]: volume_offline_op_check.c:15 Fail to get space info of [/volume2]
2024-07-03T09:30:37+08:00 synoscgi_SYNO.Core.UserSettings_1_apply[7370]: volume_offline_op_check.c:15 Fail to get space info of [/volume2]
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: api.cpp:561 API failed: POST /v1/user/login, response: {"data":{},"errno":256,"lineno":159,"status":false}
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: user.cpp:178 SLSendAPI fail
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: utils.cpp:21 not login yet
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: license.cpp:39 Failed to create LicenseManger

wjz304, 2024-07-03:
Fail to get space info of [/volume2]  ?

not login yet ?

tearfulDalvik, 2024-07-04:
> Fail to get space info of [/volume2] ?
> 
> not login yet ?

**before login**:
2024-07-03T09:30:34+08:00 synoscgi_SYNO.Core.UserSettings_1_apply[7333]: volume_offline_op_check.c:15 Fail to get space info of [/volume2]
2024-07-03T09:30:35+08:00 synoscgi_SYNO.Core.UserSettings_1_apply[7321]: volume_offline_op_check.c:15 Fail to get space info of [/volume2]
2024-07-03T09:30:37+08:00 synoscgi_SYNO.Core.UserSettings_1_apply[7370]: volume_offline_op_check.c:15 Fail to get space info of [/volume2]
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: api.cpp:561 API failed: POST /v1/user/login, response: {"data":{},"errno":256,"lineno":159,"status":false}
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: user.cpp:178 SLSendAPI fail
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: utils.cpp:21 not login yet
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
2024-07-03T09:30:38+08:00 synoscgi_SYNO.AME.License_1_check[7394]: license.cpp:39 Failed to create LicenseManger

**after login**:
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: api.cpp:561 API failed: POST /v1/user/login, response: {"data":{},"errno":256,"lineno":159,"status":false}
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: user.cpp:178 SLSendAPI fail
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: utils.cpp:21 not login yet
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
2024-07-03T09:31:34+08:00 synoscgi_SYNO.AME.License_1_activate_free_license[7795]: license.cpp:101 Failed to create "LicenseManager"

AprEcho, 2024-07-04:
我也遇到了这个，全白无法激活AME，log是同样的

mhlywxl, 2024-07-09:
同样全白遇到问题，执行这个命令后可以，不确定是不是同个错误：
curl http://code.imnks.com/ame3patch/ame72-3005.py | python

xigongyue, 2024-07-18:
同样遇到该问题，全白sn+mac1，设置netif_num=1，依旧打开AME显示无法执行此操作

lkxlzx, 2024-08-10:
一样，以前没问题，升级系统到最新就出现这问题了，引导也是最新的

Fog-Forest, 2024-08-14:
一样，全白 sn+mac1，设置netif_num=1，打开AME显示无法执行此操作

superHyqi, 2024-08-14:
SA6400，也是无法执行此操作，全白码

brianhuang2014, 2024-10-03:
請問如何更改mac? cmdline加入chage mac 並無效果

closed 2025-05-12 by wjz304