RROrg/rr #1453: BUG DVA3219/3221 don't boot with intel i219v

alekc70, 2024-06-24:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
DMI: H610-HDV/M.2 (lan intel i219v)
CPU: i5-12400
NIC: (pid & vid) 0x1656 0x0951

RR version: (You can find it in the update menu.)  
 * RR:  24.4.3
 * addons:  default
 * modules:  default
 * lkms:  prod

DSM: 
 * model:  DVA3221 / DVA3219
 * version:  no

Issue:  
DVA3219/3221 don't boot
other models DS920 / SA6400 are working fine on this system

logs:  
No logs

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

nillebor, 2024-06-25:
Update to the last Release [24.5.7](https://github.com/RROrg/rr/releases).

alekc70, 2024-06-25:
there are no loader with version 24.5.7. Where did you find it?
[image]

nillebor, 2024-06-25:
sorry  24.6.7 not 24.5.7

alekc70, 2024-06-25:
The same( don't boot.
How to activate direct boot option in loader?

nillebor, 2024-06-25:
https://github.com/RROrg/rr/issues/963#issuecomment-2155874296

alekc70, 2024-06-25:
Thank you for your help! But results is quite the same.
Only boot screen has changed.
[image]

alekc70, 2024-06-30:
Closed because there is no problem in network adapter side.

closed 2024-06-30 by alekc70