RROrg/rr #1534: modules list reset when updating RR

marboroman, 2024-06-25:
Modulesbecome all selected after updating RR. Need to inspect them all again to deselect unnecessary. The correct attitude is to preserve previous selections and add new modules unselected.

wjz304, 2024-06-26:
Because some changes in drivers are difficult to handle,
For example
1. 24.6.7 A driver depends on B, you select A and B, but A in 24.6.8 depends on B and C, there is no good way to only add C selection
2. X driver exists in 24.6.7, but X driver is removed in 24.6.8,
3. N1 driver exists in 24.6.7, but it is upgraded to N2 driver in 24.6.8, and the name changes
So currently select all to avoid some problems

marboroman, 2024-06-26:
ok, please let us know also these things:

1) if there are few versions of one drivers exist and all selected (i915, r8125) which one is used? are there any priorities or everything is just determined in alphabetical order?

2) what is affected by the number of connected modules? loading time? free space on some system partitions? Does it make sense to disable those modules that you know for sure will never be needed (virtio on baremetal, wifi on wired configurations and so on)?

wjz304, 2024-06-26:
1. Yes,  just determined in alphabetical order.
2. It has no impact. The driver loading relies on eudev to automatically identify the hardware id. When you don't have device A, even if driver A exists, it will not be loaded (no loading failure, but no loading process at all). 
As for the space, in fact, all the current drivers add up to less than 10M.
When there is no conflict, disabling does not make much sense.

wjz304, 2024-06-26:
In other words, the number of drivers that exist but you don't use under your win/ubuntu system is far greater than that under RR.