RROrg/rr #1555: Xen Virtual Block Device无法加载bootloader

Wszhdg, 2024-06-27:
我的vps厂商给我的硬盘是基于Xen Virtual Block Device的命名为xvda这在启动阶段会导致找不到bootloader这就会很尴尬，也就是说根本无法安装，我想知道我能 自己添加驱动然后让该类型硬盘可以正确被识别吗

wjz304, 2024-06-27:
可以

Wszhdg, 2024-06-27:
万分感谢大大的回答，除了需要添加xen_blkfront以外，还需要把xvda改成sda还是只需要添加对应xen_blkfront驱动一切就会完美运行

Wszhdg, 2024-06-27:
> 可以

万分感谢大大的回答，除了需要添加xen_blkfront以外，还需要把xvda改成sda还是只需要添加对应xen_blkfront驱动一切就会完美运行

wjz304, 2024-06-27:
引导盘能驱动就行，名字无所谓，存储只支持 sata

Wszhdg, 2024-06-27:
> 引导盘能驱动就行，名字无所谓，存储只支持 sata

好的，谢谢大佬，我想办法打个驱动

Wszhdg, 2024-06-27:
> 引导盘能驱动就行，名字无所谓，存储只支持 sata

> 引导盘能驱动就行，名字无所谓，存储只支持 sata

如果可能能否在之后的驱动中加入xen-blkfront.ko我尝试投机取巧用debian上的xen-blkfront.ko导入到系统然后insmod然后不兼容。。。，包括群辉的驱动。我个人尝试实在是能力有限这个vps还不给dhcp每次启动找不到ip然后手动ip add这样弄实在是，大佬看吧有兴趣就加一个。不论如何先谢过，万分感谢，一直在用您的项目，在家里的pve和甲骨文云小鸡dd确实都很好用没有问题，这次实在是emm，总之谢谢大佬一直的付出~

wjz304, 2024-06-27:
先去一些log看看
``
lspci -nnk
ls -l /sys/class/scsi_host
ls /sys/block/ 
ls -ld /sys/class/net/*/device/driver
```

Wszhdg, 2024-06-27:
> ls -ld /sys/class/net/*/device/driver

辛苦您看看
这是我重装了debian12之后测试的

命令1：

root@ca:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation 440FX - 82441FX PMC [Natoma] [8086:1237] (rev 02)
	Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
00:01.0 ISA bridge [0601]: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II] [8086:7000]
	Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
00:01.1 IDE interface [0101]: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II] [8086:7010]
	Subsystem: XenSource, Inc. 82371SB PIIX3 IDE [Natoma/Triton II] [5853:0001]
	Kernel driver in use: ata_piix
	Kernel modules: ata_piix, ata_generic
00:01.2 USB controller [0c03]: Intel Corporation 82371SB PIIX3 USB [Natoma/Triton II] [8086:7020] (rev 01)
	Subsystem: XenSource, Inc. 82371SB PIIX3 USB [Natoma/Triton II] [5853:0001]
	Kernel driver in use: uhci_hcd
	Kernel modules: uhci_hcd
00:01.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 01)
	Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
	Kernel modules: i2c_piix4
00:02.0 VGA compatible controller [0300]: Cirrus Logic GD 5446 [1013:00b8]
	Subsystem: XenSource, Inc. GD 5446 [5853:0001]
	Kernel driver in use: cirrus
	Kernel modules: cirrusfb, cirrus
00:03.0 SCSI storage controller [0100]: XenSource, Inc. Xen Platform Device [5853:0001] (rev 01)
	Subsystem: XenSource, Inc. Xen Platform Device [5853:0001]
	Kernel driver in use: xen-platform-pci

命令2：

root@ca:~# ls -l /sys/class/scsi_host
总计 0
lrwxrwxrwx 1 root root 0  6月27日 22:51 host0 -> ../../devices/pci0000:00/0000:00:01.1/ata1/host0/scsi_host/host0
lrwxrwxrwx 1 root root 0  6月27日 22:51 host1 -> ../../devices/pci0000:00/0000:00:01.1/ata2/host1/scsi_host/host1

命令3：

root@ca:~# ls /sys/block/
loop0  loop1  loop2  loop3  loop4  loop5  loop6  loop7	sr0  xvda  xvdb  xvdc

命令4：

root@ca:~# ls -ld /sys/class/net/*/device/driver
lrwxrwxrwx 1 root root 0  6月27日 22:50 /sys/class/net/enX0/device/driver -> ../../bus/xen/drivers/vif

说明：网络是能正常使用的，但是需要使用ip固定ip和网关，一共有三块硬盘xvda  xvdb  xvdc猜测一个ide两个scsi，xvda是启动盘，然后进入引导后显示can not found bootloader用lsblk显示为空明显是没驱动都找不到盘

wjz304, 2024-06-27:
xen-platform-pci

wjz304, 2024-06-27:
vif

Wszhdg, 2024-06-27:
> vif

不好意思大大，我没理解这两个是什么意思，一个是虚拟化的平台，另一个是虚拟网络接口，还需要我提供更详细的内容吗

比如系统信息
 CPU Model          : Intel(R) Xeon(R) CPU E5-2650 v2 @ 2.60GHz
 CPU Cores          : 4 @ 2593.756 MHz
 CPU Cache          : 20480 KB
 AES-NI             : ✓ Enabled
 VM-x/AMD-V         : ✓ Enabled
 Total Disk         : 34.2 GB (1.7 GB Used)
 Total Mem          : 3.9 GB (274.0 MB Used)
 System uptime      : 0 days, 0 hour 29 min
 Load average       : 0.08, 0.02, 0.01
 OS                 : Debian GNU/Linux 12
 Arch               : x86_64 (64 Bit)
 Kernel             : 6.1.0-21-amd64
 TCP CC             : 
 Virtualization     : Xen-DomU

多补充一句本来是不支持VM-x/AMD-V的厂家应要求给我开了，但是吧，经过测试才知道他们的主机内核不兼容嵌套虚拟化，所以看似开了但是一旦用kvm加速就会重启，所以没用，所以emm我本来就是想直接装，然后想虚拟机，然后才到您这，这一套折腾从上周五干到现在。。。。。。

wjz304, 2024-06-27:
一时半会搞不了， 我也没有 xen，很多配置也不知道效果是啥，

Wszhdg, 2024-06-27:
> 一时半会搞不了， 我也没有 xen，很多配置也不知道效果是啥，

好的，麻烦您啦，用了这么多vps，xen我确实也是第一次见，您要是方便看看能不能先加个硬盘驱动，我可以试试给您反馈，或者什么时候您正好碰到了，实在不方便也没关系，要说完全用不了也不至于，pve不开kvm加速qemu纯模拟也能用，就是卡一点反正，哈哈

wjz304, 2024-06-29:
https://github.com/RROrg/rr-build/actions/runs/9723091013

Wszhdg, 2024-06-29:
> https://github.com/RROrg/rr-build/actions/runs/9723091013

感谢大佬驱动立竿见影，只不过好像格式有问题，但是根据lspci -nnk来看，只有ide才对，就挺奇怪的

[image]

Wszhdg, 2024-06-29:
> https://github.com/RROrg/rr-build/actions/runs/9723091013

1.编译过程没有问题很顺利，就是即便设置了静态ip，也没网，确认是dns问题，重新设置一下静态即可解决

2.然后是启动：

2.1.都会有提示*** Please insert at least one sata/scsi disk for system installation, except for the bootloader disk. ***
2.2.已经测过的型号（编译后启动被关机之后，会直接编译下一个，还在接着测试）：
Model:    DS3622xs+(broadwellnk)（差不多200s后关机,prod）
Model:    SA6400(epyc7002)（差不多150s后关机,dev）
Model:    DS3617xs(broadwell)（10秒钟左右后就会关机,test）
Model:    SA3400D(broadwellntbap)（90秒钟左右后就会关机,prod）
ps：关机的意思就是真的关机了，没有任何提示的情况下，甚至不是重启，得去vps面板开启

3.启动的信息如下（只会更改lkm和Model别的不动）：

                                                                           BOOTING: [BIOS] [/DEV/XVDA BLOCK DoM]
Model:    SA3400D(broadwellntbap)
Version:  7.2(69057)
Kernel:   official
LKM:      prod
DMI:      Xen HVM domU, BIOS 4.13 01/26/2021
CPU:      Intel(R) Xeon(R) CPU E5-2650 v2 @ 2.60GHz
MEM:      2963 MB
*** Please insert at least one sata/scsi disk for system installation, except for the bootloader disk. ***
Cmdline:
dom_szmax=35850 earlyprintk syno_hw_version=SA3400D console=ttyS0,115200n8 consoleblank=600 netif_num=2 noefi pid=0x0001 earlycon=uart8250,io,0x3f8,115200n8 synoboot_satadom=2 mac2=* mac1=* sn=* SMBusHddDynamicPower=1 vid=0x46f4 elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=5 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0 network.02ad567799c7=*
Detected 1 network cards.
eth0 connected.                  
Waiting IP.
eth0(vif): Access http://* to connect the DSM via web.

wjz304, 2024-06-29:
方便远程吗？ QQ 304403268

wjz304, 2024-07-05:
https://github.com/RROrg/rr/releases/tag/24.7.0

wjz304, 2024-07-05:
test sa6400 custom

jesusSerranoCarballido, 2025-04-11:
I'm using XEN. After loading the bootloader, when the DSM (sa6400 custom) installation starts, it tells me it can't find the hard drive.
I've checked and it's in XVD format, and I can't change it to SATA/SCSI.
What's the next step, or am I missing something?

Wszhdg, 2025-04-11:
> I'm using XEN. After loading the bootloader, when the DSM (sa6400 custom) installation starts, it tells me it can't find the hard drive. I've checked and it's in XVD format, and I can't change it to SATA/SCSI. What's the next step, or am I missing something?

no you do nit miss anything. the xvda what is not support by 2 reason. first the drive of xvda need to add to custom. which was finished by wjz304. second the dsm not support boot from xvd* if you own the xen services you can change the hard drive to common sd* or vd*

jesusSerranoCarballido, 2025-04-11:
Thanks.
I'll look into how to change from xvd* to sd* or vd* and if I find anything, I'll post it here so anyone else finds themselves in the same situation in the future.

wjz304, 2025-10-15:
https://github.com/RROrg/rr/issues/18166