RROrg/rr #1561: BUG p4800x通过m2转us连接，进行Secure Erase的时候，卡99%，然后直接整个盘不见了。正常当作存储池使用没问题

deYangar, 2024-06-27:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  ASUS System Product Name/PRIME Z790-P D4,BIOS 1645 03/15/2024
 * CPU:  13900k es2 q0pu
 * NIC:  (pid & vid)  rtl8125bg

RR version: (You can find it in the update menu.)  
 * RR:  24.6.7
 * addons:  acpid  mountloader reboottoloader codecpatch console cpuinfo dbgutils hdddb expands lsiutil nvmecache nvmevolume nvmesystem sensors storagepanel updatenotify 
 * modules:  all
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1-69057 Update 1

Issue:  BUG p4800x通过m2转us连接，进行Secure Erase的时候，卡99%。这时候通过ssh sudo fdisk -l，发现没有这块盘的信息了。正常当作存储池使用没问题。
[image]

[fdisk.txt](https://github.com/user-attachments/files/16014540/fdisk.txt)

logs:  
[messages.txt](https://github.com/user-attachments/files/16014534/messages.txt)

nillebor, 2024-06-27:
It's best if you don't use Secure Erase. Some Synology users have already broken the disk. Some were still able to save them. Others had broken discs that could be disposed of. Prefer to use external tools. Synology Secure Erase is not good.

deYangar, 2024-06-27:
> It's best if you don't use Secure Erase. Some Synology users have already broken the disk. Some were still able to save them. Others had broken discs that could be disposed of. Prefer to use external tools. Synology Secure Erase is not good.

是的，但是如果要将ssd加入raid缓存，系统会自动进行一次Secure Erase，然后也会卡在这，卡在99%，导致raid加入不了ssd缓存。主要是因为这个。
Yes, but if you want to add the SSD to the RAID cache, the system will automatically perform a Secure Erase, and then it will also get stuck here, stuck at 99%, resulting in the RAID not joining the SSD cache. Mainly because of this.
[image]

deYangar, 2024-06-28:
另外吗，冒昧的问一句，大佬的Q群还能加人吗，五群我申请了两次好像也没验证通过

wjz304, 2024-06-28:
RR 的 log 没有意义，要抓 DSM 掉盘时的 日志，  dmesg  &  /var/log/messages

wjz304, 2024-06-28:
> 另外吗，冒昧的问一句，大佬的Q群还能加人吗，五群我申请了两次好像也没验证通过

加 6 群吧， 管理员都设置了 小号自动屏蔽，我刚把6群放开了

deYangar, 2024-06-30:
> RR 的 log 没有意义，要抓 DSM 掉盘时的 日志， dmesg & /var/log/messages

这是掉盘后的dsm日志
[messages.txt](https://github.com/user-attachments/files/16044572/messages.txt)

leochenmeng, 2024-07-19:
解决了吗？我也有这个问题，似乎对傲腾都有这种情况

wjz304, 2024-07-19:
> > RR 的 log 没有意义，要抓 DSM 掉盘时的 日志， dmesg & /var/log/messages
> 
> 这是掉盘后的dsm日志 [messages.txt](https://github.com/user-attachments/files/16044572/messages.txt)

我是说取这两个log

```
dmesg
cat /var/log/messages
```

nvme nvme0: I/O 14 QID 0 timeout, reset controller

deYangar, 2024-07-19:
> > > RR 的 log 没有意义，要抓 DSM 掉盘时的 日志， dmesg & /var/log/messages
> > 
> > 
> > 这是掉盘后的dsm日志 [messages.txt](https://github.com/user-attachments/files/16044572/messages.txt)
> 
> 我是说取这两个log
> 
> ```
> dmesg
> cat /var/log/messages
> ```
> 
> nvme nvme0: I/O 14 QID 0 timeout, reset controller

nvme nvme0: I/O 14 QID 0 timeout, reset controller之后就没有产生新的内容了，我这是卡住后等了5分钟才读取的log。还是说需要补充什么操作

DrDREAM233, 2025-02-16:
我是960G的905p，也是m2转u2，相同的问题，期待解决