RROrg/rr #1566: 当前无法下载群晖的pat

HivenZhu, 2024-06-28:
选择型号及版本后，自动指向群晖的pat下载无法完成。

HivenZhu, 2024-06-28:
Install ENV: (You can find it in the boot interface.)

DMI: VMWare virtual platform（ESXi 6.7）/440BX Desktop reference platform，BIOS 6.00 12/12/2018
CPU: Intel Pentiem Gold G5400 3.70GHz
NIC: (pid & vid)  pid=0x0001 vid=0x46f4
RR version: (You can find it in the update menu.)  try it for v24.6.4 and v24.6.7, v24.6.4 worked before

RR:
addons: No
modules: No
lkms: No
DSM: v7.2 DAV3221/DAV3219/DS1019+/......

model: DAV3221/DAV3219/DS1019+/......
version: v7.2/v7.1/v7.0
Issue:  unable to connect to synology website,  please check the network and try again, or use 'Parse Pat'!

logs: "unable to connect to synology website,  please check the network and try again, or use 'Parse Pat'!"

wjz304, 2024-06-28:
please use 'Parse Pat'!

HivenZhu, 2024-06-28:
Dear,

If it's a new install, where to get the 'parse pat'? and upload to system?

| |
朱华峰
|
|
+86-13501195469
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年6月28日 13:37 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***>,
***@***.***> |
| 主题 | Re: [RROrg/rr] 当前无法下载群晖的pat (Issue #1566) |

please use 'Parse Pat'!

wjz304, 2024-06-28:
menu

wjz304, 2024-06-28:
https://archive.synology.cn/download/Os/DSM/7.2.1-69057-1-NanoPacked

016, 2024-06-28:
@wjz304 any document or link can learn how to "Parse pat"? thx, I got same problem

016, 2024-06-28:
what is DUFS?

016, 2024-06-28:
found it , http://192.168.1.3:7304/

016, 2024-06-28:
it's a net problem, just try many times, it will working

chenyuchenluo, 2024-06-28:
我昨晚安装也遇到了这个问题，安装进度到57左右，提示pat损坏。重启机器重新进入rr配置编译启动，然后就好了。如果失败多试几次。。。

wjz304, 2024-06-28:
> @wjz304 any document or link can learn how to "Parse pat"? thx, I got same problem

https://rrorg.cn/archives/Upload%20PAT

closed 2024-08-03 by wjz304