RROrg/rr #1580: BUG

tony80437, 2024-06-29:
Manufacturer: Supermicro
Product Name: X7SPA-H 
SATA 128G硬盘启动

请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  [dmi.txt](https://github.com/user-attachments/files/16039501/dmi.txt)
 * CPU:  Intel Atom D525 @ 1.80GHz 
 * NIC:  (pid & vid)  SATA 硬盘启动？？

RR version: (You can find it in the update menu.)  
 * RR:  24.6.8
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  启动到40多秒提示 Finding for bootloader disk .............................................................
Bootloader disk not found.
后面就进入命令行窗口。

查看日志可以看到已经有IP地址，通过SSH登录后也是提示 Finding for bootloader disk .............................................................
Bootloader disk not found.

logs:  
[messages.log](https://github.com/user-attachments/files/16039508/messages.log)

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

tony80437, 2024-06-29:
补充: 这个机器如果用U盘启动，只能进入到grub的命令行界面，无法启动rr

wjz304, 2024-07-19:
磁盘控制器无法识别，

`lspci -nnk`

wjz304, 2024-07-19:
DMI: Supermicro X7SPA-H /X7SPA-H , BIOS 1.2        09/14/11

wjz304, 2024-08-03:
无进一步反馈