RROrg/rr #1624: BUG DVA3221 doesn't boot

alekc70, 2024-06-30:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  H610-HDV/M.2
 * CPU:  i5-12400
 * NIC:  (pid & vid) i219-v or r8152

RR version: (You can find it in the update menu.)  
 * RR:  different (from latest 24.6.9 to 23.12.23)
 * addons:  default
 * modules:  default
 * lkms:  prod

DSM: 
 * model:  DVA3219 / 3221
 * version:  any (7.0 / 7.1 / 7.2)

Issue:  DSM doesn't boot, and the network doesn't start.

logs:  
[logs.tar.gz](https://github.com/user-attachments/files/16043429/logs.tar.gz)

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-06-30:
[image]
?

alekc70, 2024-06-30:
Seems to me it isn't mine, because I had neve been run ds920 with usb flash.
I assure it is located inside loader by default

wjz304, 2024-06-30:
ds920 from your hard drive system

Unselect cdc_ether in modules menu and rebuild, retest

alekc70, 2024-06-30:
thank you!
I remove all hdd from system. after that I've got ping but still no connection to dsm
also I unselect cdc_ether in modules

Could you please review log file
[logs.tar.gz](https://github.com/user-attachments/files/16045182/logs.tar.gz)

wjz304, 2024-07-01:
i219 seem to not have dhcp to ip

alekc70, 2024-07-01:
@wjz304 thank you!
I've checked two additional external usb adapters on rtl 8152(1g) and 8156 (2.5g) with the same results - no boot :(
Could you please review one more log where I ran it with external usb network 
[logs.tar.gz](https://github.com/user-attachments/files/16058346/logs.tar.gz)

wjz304, 2024-07-01:
try enable "Switch direct boot" form  advanced menu

alekc70, 2024-07-01:
I've tried direct boot, but the result is the same—it didn't boot
log file please see below
[logs.tar.gz](https://github.com/user-attachments/files/16059008/logs.tar.gz)

wjz304, 2024-07-01:
Have you checked the routing?

All logs have a weird problem, messages are empty, (I don't know why)
From dmesg.log, I can see that the network cards are recognized normally and there is no error message. 
However, whether DHCP to IP is normal can only be known through messages.

You can try other models, such as sa6400

alekc70, 2024-07-02:
What do you mean the routing? 

SA6400 and DS920 works very well on this hardware

snailium, 2024-08-05:
Maybe try to switch "Priority use official driver" to false?

wjz304, 2024-12-05:
test 24.12.1

closed 2025-01-03 by alekc70