RROrg/rr #1716: 识别固态超时 nvme timeout,completion polled

xw1369, 2024-07-04:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  VMware, Inc. VMware20 , 1/440BX Desktop Reference P latform, BIOS UMW201 ，00U .23553139. B64.240326094003/ 26/ 2024

 * CPU:  i3-8100t
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.6.9
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+
 * version:  7.2(69057u4)

Issue:  
在esxi8里分别直通了威刚,固德佳M2盘,都无法开机. 试过三星铠侠固态是正常
logs:  
[image]

[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

tianiue, 2024-07-06:
升级7.0还是同样的问题，6.8版可以启动

wjz304, 2024-07-19:
试下 7.2 吧，加了 rootwait 不知道能不能改善

wjz304, 2024-08-03:
无进一步反馈

closed 2024-08-03 by wjz304

wjz304, 2024-08-09:
nvme_core.io_timeout=4294967295 nvme_core.admin_timeout=4294967295

iron8706, 2025-07-06:
> nvme_core.io_timeout=4294967295 nvme_core.admin_timeout=4294967295

用esxi安装7.2.1出现同样的问题，表象为卡在booting界面超过二十分钟，进去到dsm后，可以识别到nvme盘，但是无法添加为硬盘或缓存，长时间无响应，想知道这行代码添加到哪里？grub.cfg?