RROrg/rr #1790: BUG

category-9431, 2024-07-06:
vm loader on proxmox 8.2.4

rr-24.6.9.img - boot
rr-24.7.0.img - no boot

something wrong!!

请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-07-06:
what‘s wrong??

icebrew, 2024-07-06:
[image]

Works but when you run the console it shows this - Proxmox 8.2.4

icebrew, 2024-07-06:
yeah you right happens after you have completed your upgrade at least that was for me.

wjz304, 2024-07-06:
https://github.com/RROrg/rr/issues/1791

wjz304, 2024-07-06:
"Power off display after boot: false" will be the same as before

closed 2024-07-06 by wjz304