RROrg/rr #1796: r8125双网口主板其中一个网口无法dhcp。仅仅一个网口正常

lingshengjiang898, 2024-07-06:
请填写以下信息.  
Please fill in the following information.  
星云b365主板自带r8125双网口其中一个网口无法dhcp，确认网线和路由器均正常，设置静态ip无效。群晖显示已经连接，交换网线网无效。更换交换机网口无效

[image]
[image]

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)
00:00.0 Host bridge [0600]: Intel Corporation 8th Gen Core 4-core Desktop Processor Host Bridge/DRAM Registers [Coffee Lake S] [8086:3e1f] (rev 08)
00:01.0 PCI bridge [0604]: Intel Corporation 6th-10th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 08)
00:02.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630] [8086:3e91]
00:14.0 USB controller [0c03]: Intel Corporation 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller [8086:a2af]
00:16.0 Communication controller [0780]: Intel Corporation 200 Series PCH CSME HECI #1 [8086:a2ba]
00:17.0 SATA controller [0106]: Intel Corporation 200 Series PCH SATA controller [AHCI mode] [8086:a282]
00:1b.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #21 [8086:a2eb] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #5 [8086:a294] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #11 [8086:a29a] (rev f0)
00:1d.3 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #12 [8086:a29b] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2cc]
00:1f.2 Memory controller [0580]: Intel Corporation 200 Series/Z370 Chipset Family Power Management Controller [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation 200 Series PCH HD Audio [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation 200 Series/Z370 Chipset Family SMBus Controller [8086:a2a3]
01:00.0 Ethernet controller [0200]: Emulex Corporation OneConnect 10Gb NIC (be3) [19a2:0710] (rev 02)
01:00.1 Ethernet controller [0200]: Emulex Corporation OneConnect 10Gb NIC (be3) [19a2:0710] (rev 02)
02:00.0 Non-Volatile memory controller [0108]: Sandisk Corp Device [15b7:5007] (rev 01)
03:00.0 Non-Volatile memory controller [0108]: Sandisk Corp WD PC SN810 / Black SN850 NVMe SSD [15b7:5011] (rev 01)
04:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
05:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)

RR version: (You can find it in the update menu.)  
 * RR:  24.7.0
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  
[logs.zip](https://github.com/user-attachments/files/16116873/logs.zip)
(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

lingshengjiang898, 2024-07-06:
群晖7.2.1
高级设置-直接启动
[image]
日志在附件[logs.zip](https://github.com/user-attachments/files/16116873/logs.zip)

ljfzx, 2024-07-14:
我也有同样的问题

wjz304, 2024-07-19:
https://github.com/RROrg/rr/issues/1829

milkman-wang, 2024-07-20:
我也有这个问题

wjz304, 2024-07-20:
test https://github.com/RROrg/rr-build/actions/runs/10013261482

wjz304, 2024-07-20:
https://github.com/RROrg/rr/issues/1132

milkman-wang, 2024-07-20:
现在有新问题，全部重装以后，引导装群晖系统的界面连不上网，但路由器后台能看到。我是物理机

我现在装这个action的镜像试一下

milkman-wang, 2024-07-20:
202407200321版本重新制作引导以后问题依旧存在，设置跟之前都一样，不知道是什么问题

milkman-wang, 2024-07-21:
说一下我的解决方法吧，硬盘全部格式化后装了Windows，把最新的驱动打上，然后再装rr，这时候dhcp问题解决，但rr7.2版本装群晖时可联网，重启后进系统依旧连不了网，后台可看到设备在线但一直进不去后台，退回rr7.1版本解决

lingshengjiang898, 2024-07-21:
我这边尝试下面步骤解决了，没有升级RR: 24.7.0，没有重新做引导盘
1、把所有硬盘拔出来调整顺序插回去
2、先不插r8125双网口，先用10gsfp网口保证拿到dhcp
3、保证10g网口先不拔出，插入r8125双网口网线
4、r8125双网口均能正常dhcp
PS：星云b365用rr引导装完都进不了系统，需要切换直接启动，其他设置不变之后才可以进入，给同志参考：RR: 24.7.0

rongcenter, 2024-07-27:
你试试，把自定sn/mac，sn不动，把mac改成网卡真实的mac，然后看看获取到ip不，如果获取到ip后再改回虚拟mac

wjz304, 2024-08-14:
无人反馈，本issues关闭

closed 2024-08-14 by wjz304