RROrg/rr #1806: 9p not loading

Fentanest, 2024-07-09:
请填写以下信息.  
Please fill in the following information.  

**It's under Proxmox environment**

Install ENV: (You can find it in the boot interface.)  
 * DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 4.2023-08-4 02/15/2024
 * CPU:  Host(EPYC 7352)
 * NIC:  SATA

RR version: (You can find it in the update menu.)  
 * RR:  24.7.0
 * addons:  24.7.0
 * modules:  24.7.0
 * lkms:  24.6.1

DSM: 
 * model:  DS920+ ( I tried SA6400 both)
 * version:  DSM 7.2.1-69057 U5 (DS920+)

Issue:  
I checked all modules when i build the buildloader(I assume I installed 9p modules-9p, 9pnet, 9pnet_virtio)
and I set hostshare(9p) path in *.conf file in Proxmox.

args: -fsdev 'local,security_model=passthrough,id=fsdev0,path=/mnt/pve/test' -device 'virtio-9p-pci,id=fs0,fsdev=fsdev0,mount_tag=hostshare'
[image]

but when I open SSH shell and let the DSM mount the 9p share path after the DSM boot finished,
It said
'Unknown filesystem type: 9p'
[image]

I tried this way in Ubuntu VM.
it works.
so I think the way I trying is not wrong.

How can I mount 9p in DSM?

additional information #1
root@DS920:/mnt# lsmod | grep 9pnet_virtio
9pnet_virtio            6897  0
9pnet                  60560  1 9pnet_virtio
virtio                  5442  4 virtio_net,virtio_pci,9pnet_virtio,virtio_console
virtio_ring             6903  4 virtio_net,virtio_pci,9pnet_virtio,virtio_console
[image]

additional information #2
root@DS920:/mnt# grep 9pnet_virtio /proc/modules
9pnet_virtio 6897 0 - Live 0xffffffffa031f000 (OE)
9pnet 60560 1 9pnet_virtio, Live 0xffffffffa02ef000 (OE)
virtio 5442 4 9pnet_virtio,virtio_console,virtio_net,virtio_pci, Live 0xffffffffa00cb000 (OE)
virtio_ring 6903 4 9pnet_virtio,virtio_console,virtio_net,virtio_pci, Live 0xffffffffa00c5000 (OE)
[image]

logs:  nothing appeared in console.

thank you!

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-07-09:
only SA6400 custom kernel

Fentanest, 2024-07-09:
I got it. it works. really appreciate.
I really love this function.

closed 2024-07-10 by wjz304