RROrg/rr #1810: netif_num 无法生效，并且没有办法检测down状态的网口

Pramy, 2024-07-09:
## 环境
1. pve8.0 安装dsm 7.2，直通网卡pcie，网卡是hp 544+flr双网口
2. rr版本 v24.7.0，其他设置默认，只调整过cmdline 的 mac 地址 和 netif_num
3. 一个网口接着交换机，另外一个网口接通另外一个nas设备。
[image]

## 问题
当将连接nas的网口的线拔掉，启动的时候只能检测到一个网口，进到dsm里面网络面板也只有一个接口，我设置了netif_num=3进去一样只有一个网络配置
[image]
[image]
这个时候如果将这条线接上，dsm就无法访问了，只能重启

重启的时候我接着两条线，启动命令如下
[image]
[image]
即使设置netif_num=3，进去之后也只有两个网络接口，这时候插拔网线是没有问题的。
[image]

## 问题
这里是不是只会扫描up的网口过滤了down的网口？这里是rr的设置导致的还是dsm导致的呢？

wjz304, 2024-07-09:
netif_num 和 网卡没有半毛钱的关系， netif_num 等于 cmdline 中的 macN  个数，否则 SYNO 相关服务不可用。

你几个网卡就几个网卡，你就2个网卡 引导也变不出来3个，

down 的网卡， webUI 不显示，DSM 的设计

wjz304, 2024-07-09:
多说一句 cmdline 中的 macN  和网卡也没半毛钱关系，你设置几个 也不会影响 物理环境，只用于激活服务

Pramy, 2024-07-09:
> 多说一句 cmdline 中的 macN 和网卡也没半毛钱关系，你设置几个 也不会影响 物理环境，只用于激活服务

感谢指导哈，我再请教一下，我记得以前是可以设置 maxlanport 来启用多个网口，即使是 down 的网口也会显示，现在我看到 maxlanport 已经设置了 8 了，但是没有显示 down 的网口，这里还有其他的办法可以让他显示么

wjz304, 2024-07-09:
ifconfig ethX up

Pramy, 2024-07-10:
> ifconfig ethX up

在 dsm 里面只显示一个网口，没有办法主动 up
[image]

这是 dmesg 信息 
[image]

下面是 lspci 的信息

```
01:00.0 Class 0280: Device 15b3:1007
	Subsystem: Device 103c:22f4
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 32 bytes
	Interrupt: pin A routed to IRQ 16
	Region 0: Memory at fde00000 (64-bit, non-prefetchable) [size=1M]
	Region 2: Memory at f0000000 (64-bit, prefetchable) [size=32M]
	Expansion ROM at fdf00000 [disabled] [size=1M]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [48] Vital Product Data
		Product Name: HP InfiniBand FDR/Ethernet 10Gb/40Gb 2-port 544+FLR-QSFP Adapter
		Read-only fields:
			[PN] Part number: 764285-B21
			[EC] Engineering changes: A3
			[SN] Serial number: IL252603PF
			[V0] Vendor specific: Alom FDR x8 13W
			[V2] Vendor specific: 5526
			[V4] Vendor specific: 5065F3887D81
			[V5] Vendor specific: A3
			[VA] Vendor specific: HP:V2=MFG:V4=MAC:V5=PCAR
			[VB] Vendor specific: HP ConnectX-3Pro QSFP
			[RV] Reserved: checksum good, 0 byte(s) reserved
		Read/write fields:
			[V1] Vendor specific:
			[YA] Asset tag: N/A
			[V3] Vendor specific:
			[RW] Read-write area: 243 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 253 byte(s) free
			[RW] Read-write area: 252 byte(s) free
		End
	Capabilities: [9c] MSI-X: Enable+ Count=128 Masked-
		Vector table: BAR=0 offset=0007c000
		PBA: BAR=0 offset=0007d000
	Capabilities: [60] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <64ns, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 116.000W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #8, Speed 8GT/s, Width x8, ASPM L0s, Exit Latency L0s unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (ok), Width x8 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR-, OBFF Not Supported
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
			 AtomicOpsCtl: ReqEn-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [c0] Vendor Specific Information: Len=18 <?>
	Capabilities: [100 v0] Null
	Capabilities: [148 v1] Device Serial Number 50-65-f3-ff-ff-88-7d-80
	Capabilities: [154 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: mlx4_core
```
目前这个卡是 pve 直通给群晖的

Pramy, 2024-07-10:
貌似是识别到了两个网口，不过没有接线的那个网口默认识别成 ib 模式了，接了线的网口识别成 eth
[image]
这里 mlx4_core 驱动是可以支持默认设置成eth 模式的，需要在 /etc/modprobe.d/mlx4.conf 配置文件上面增加 options mlx4_core port_type_array=2,2 参数，这里我在 rr 中如何设置呢，是不是可以直接加参数，还是说我需要改grub.cfg配置呢

wjz304, 2024-07-10:
都可以

Pramy, 2024-07-15:
cmdline 添加 `mlx4_core.port_type_array=2,2` 启动可以设置网卡模式为 eth，感谢大佬耐心的回答

> 都可以