RROrg/rr #1830: BUG SA6400 Power off display after boot

wahabe, 2024-07-13:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  ROG MAXIMUS X CODE Z370
 * CPU:  i7-8700K
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.7.1 / 24.7.0
 * addons:  default
 * modules:  default
 * lkms:  default

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1

Issue: {Power off display after boot: true} Freeze in Control Panel + Graphics card crashes (GTX 1650)
{Power off display after boot: false} The system status returns to normal

logs:  
[Power off display after boot true.txt](https://github.com/user-attachments/files/16201429/Power.off.display.after.boot.true.txt)
[image]
[Power off display after boot false.txt](https://github.com/user-attachments/files/16201433/Power.off.display.after.boot.false.txt)
[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-07-13:
I really didn't consider the situation of nvdia,
I will test it

wjz304, 2024-07-19:
24.7.2

closed 2024-07-19 by wjz304