RROrg/rr #1834: BUG

aarpro, 2024-07-13:
请填写以下信息.  
I can not update from version 23.11.10 to the latest

RR version: (You can find it in the update menu.)  
23.11.10
DSM: 
DS3622xs+ 
Issue:  

logs:  
https://github.com/user-attachments/assets/472c812d-c9b2-402b-8b33-ee6800e338dd

... 如果你提供不了详细信息，那就等有缘人吧!
I can't rewrite the flash card - now access to the server only through remote KWM IMPI

wjz304, 2024-07-14:
The upgrade has changed a lot during the GitHub shutdown period, I can't remember them all, but you can use the local upload method to update.

In addition, some changes during this period will only take effect if you rewrite the boot disk

wjz304, 2024-07-14:
a long time last saw you.
[image]

aarpro, 2024-07-14:
> a long time last saw you.
You haven't been the same for a long time
Glad you returned to the project

aarpro, 2024-07-14:
> In addition, some changes during this period will only take effect if you rewrite the boot disk

I guess about it
but i heve access now only via KWM IMPI
[image]

version 23.11.10 I worked for a long time and did not require attention
A few days ago, access to the server disappeared
I decided to see what was the problem and found that no any bild on flash, as if I had inserted a new flash.

aarpro, 2024-07-14:
> but you can use the local upload method to update.

Now I will try to do it
[image]

aarpro, 2024-07-14:
> but you can use the local upload method to update.

Now I will try to do it
[image]

What file should I use for this?
[image]

wjz304, 2024-07-14:
[image]
After downloading, rename it to update.zip

aarpro, 2024-07-14:
something happens .. I'm waiting
Video below
https://github.com/user-attachments/assets/b05aeee6-caee-478e-851a-a919e07bd338

aarpro, 2024-07-14:
> something happens .. I'm waiting Video below https://github.com/user-attachments/assets/b05aeee6-caee-478e-851a-a919e07bd338

@wjz304 How long can it last?

wjz304, 2024-07-14:
Web compatibility issues, when the file is too large, it will be garbled, please use ssh connection to update,
This page can be closed directly.

Subsequent versions have disabled RR updates via the web

wjz304, 2024-07-14:
You can use xshell

aarpro, 2024-07-14:
> use ssh connection

Please remind the passport/login - I can't find it quickly on your blog

aarpro, 2024-07-14:
> Please remind the passport/login

root / Redp1lL-1s-4weSomE

aarpro, 2024-07-14:
I just rebooted system and server is loaded into the DSM
[image]
 :)

wjz304, 2024-07-14:
username root
password rr

aarpro, 2024-07-14:
> username root
> password rr

ok - I got access to my DSM - it booted ...
Now I am overloaded to the bootloader and see what is there
I haven't connected to SSH yet

aarpro, 2024-07-14:
@wjz304 via ssh the same video below

https://github.com/user-attachments/assets/0cfdad83-13e8-4540-bd4e-bc4dbb84ac31

wjz304, 2024-07-14:
There is also a probability that xshell will be incompatible and garbled characters will appear

wjz304, 2024-07-14:
So the later versions abandoned sz rz,

aarpro, 2024-07-14:
@wjz304 need to rewrite the flash? there are no other options ?

wjz304, 2024-07-14:
The garbled code is a probability problem, you can try it a few more times.
There are other ways, but it is still recommended to rewrite

wjz304, 2024-07-14:
[rr-update.sh.gz](https://github.com/user-attachments/files/16228125/rr-update.sh.gz)

aarpro, 2024-07-14:
@wjz304 
So far - like this
[image]
I expect

wjz304, 2024-07-14:
stuck ？

aarpro, 2024-07-14:
> stuck ？

yes
After hard reset, server но loading wia flash :)
so i need to rewrite flash...  It will not be soon... The server is far from me

aarpro, 2024-07-14:
> server но loading via flash :)

video below
Uploading Video_2024-07-14_204454 (1).mp4…

aarpro, 2024-07-14:
> server но loading via flash :)

video below

https://github.com/user-attachments/assets/477b5dd6-2217-48ee-ba94-525d27659da5

wjz304, 2024-07-14:
After what operation is this? The boot seems to be lost?

aarpro, 2024-07-14:
> After what operation is this? The boot seems to be lost?

After what: https://github.com/RROrg/rr/issues/1834#issuecomment-2227409849 and  reboot

aarpro, 2024-07-18:
@wjz304 
now i rewrite flash...
all is OK
Thanks again for you returned

I think you can close issue

wjz304, 2024-07-18:
Now what is the situation of this issue https://github.com/RROrg/rr/issues/214

aarpro, 2024-07-18:
> Now what is the situation of this issue #214

nohow :)
Remove the excess - now only 32 gb memory

wjz304, 2024-07-18:
[image]

closed 2024-07-19 by wjz304