RROrg/rr #1835: 虚拟机环境 24.7.1  无限重启

exeron21, 2024-07-14:
虚拟机环境 24.7.1 无限重启，尝试过918,1615，923,1617，都选择7.2版本，都没有修改高级选项，一路回车下一步

其中只有DS3617xs_7.21-69057能进入pat文件的选择安装界面，其他全是无限重启

CPU：E5-1650 v2
主板：华南X79 ITX-4L
内存：三星DDR 16*2
硬盘：Fanxiang S500Q 2TB

Windows 11 专业版
VMware® Workstation 17 Pro

exeron21, 2024-07-14:
看了一下日志：
[    0.568936] Kernel command line: BOOT_IMAGE=/bzImage-rr earlyprintk earlycon=ttyS0,115200 console=ttyS0,115200 root=/dev/ram rootwait pcie_aspm=off net.ifnames=0 IWANTTOCHANGETHECONFIG
[    0.568987] PCIe ASPM is disabled
[    0.569005] Unknown kernel command line parameters "IWANTTOCHANGETHECONFIG BOOT_IMAGE=/bzImage-rr", will be passed to user space.

--------------------------------------------------------------------------------------

    3.547855] pci 0000:00:10.0: BAR 6: assigned [mem 0xc0000000-0xc0003fff pref]
[    3.551117] pci 0000:00:15.3: BAR 13: no space for [io  size 0x1000]
[    3.554016] pci 0000:00:15.3: BAR 13: failed to assign [io  size 0x1000]
[    3.557023] pci 0000:00:15.4: BAR 13: no space for [io  size 0x1000]
[    3.559890] pci 0000:00:15.4: BAR 13: failed to assign [io  size 0x1000]
[    3.562931] pci 0000:00:15.5: BAR 13: no space for [io  size 0x1000]
[    3.565796] pci 0000:00:15.5: BAR 13: failed to assign [io  size 0x1000]
.....

--------------------------------------------------------------------------------------

Starting iucode-tool: /usr/sbin/iucode_tool: /dev/cpu/microcode: cannot open for writing: No such file or directory
FAIL

--------------------------------------------------------------------------------------

14 Jul 07:43:45 ntpd[268]: INIT: Command line: /usr/sbin/ntpd -u ntp:ntp -g -p /var/run/ntpd.pid
OK
Starting ntpd: FAIL

完整的日志：
[s.log](https://github.com/user-attachments/files/16216032/s.log)

exeron21, 2024-07-14:
update : rr23.6.3 + ds3617xs 7.2 安装成功

wjz304, 2024-07-14:
24.7.1  918 首次 是灰主动重启一次的，有 918 一直重启的log么

wjz304, 2024-07-14:
[    0.066018] x2apic: IRQ remapping doesn't support X2APIC mode
[    0.067987] x2apic disabled
[    0.068987] general protection fault: 0000 [#1] SMP 
[    0.069987] Modules linked in:
[    0.070003] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.4.302+ #69057
[    0.070987] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 11/12/2020
[    0.072987] task: ffff880138de0000 ti: ffff880138de8000 task.ti: ffff880138de8000
[    0.074988] RIP: 0010:[<ffffffff8103c2a1>]  [<ffffffff8103c2a1>] native_apic_msr_read+0x11/0x20

需要 nox2apic